ghzserg/zmod #3: AUTO_FULL_BED_LEVEL сбрасывает температуру

Dorithur, 2025-02-20:
Из мануала:
> [AUTO_FULL_BED_LEVEL](https://github.com/ghzserg/zmod/wiki/Macros#auto_full_bed_level) - снятие карты стола с очисткой сопла при заданной температуре стола и экструдера. Выключает нагрев после снятия карты.

В моем случае поддержание указанной температуры происходит только до окончания этапа чистки сопла. Дальше выставляетя температура 120/60 и снимается карта. Это баг или так и должно быть? Судя по мануалу - карта должна сниматься на указанных параметрах. Запускаю через Fluidd по кнопке `КАЛИБРОВКА СТОЛА`.

seligor, 2025-02-20:
Пришлите ваши start G-code и end G-code
Например:

```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
```
```
END_PRINT
```

А так же параметры температур пластика, которые используются

[image]

Так же учитывайте, что перед построением карты стола производится очистка сопла. Сопло и стол нагреваются до рабочей температуры, происходит очистка, далее сопло остывает до 120 чтобы не тёк пластик и производится снятие карты стола. После этого сопло снова нагревается до рабочей температуры

ghzserg, 2025-02-20:
Пришлите  ссылку на TAR_CONFIG. Остывания стола до 60 не должно быть. Очистка, потом остывпание сопла до 120, потом снятие карты стола на вашей температуре.

seligor, 2025-02-20:
> Остывания стола до 60 не должно быть

А если печать PLA? или 80\120 это хардкод?

Dorithur, 2025-02-20:
> Пришлите ваши start G-code и end G-code

Макрос я вызываю отдельно, через Fluidd. Провожу выравнивание стола. G-code не правил еще.

>  далее сопло остывает до 120 чтобы не тёк пластик и производится снятие карты стола

Вот в этом и был вопрос. Я посчитал, что карта должна сниматься на рабочей температуре **и** стола **и** сопла, поскольку расширяются оба тела. Получается, что это нормальное поведение, не баг.

ghzserg, 2025-02-20:
Сопло остывает, стол не остывает. Если у вас стол остывает, то пришлите скрин температурного графика из fluidd и TAR_CONFIG

ghzserg, 2025-02-20:
AUTO_FULL_BED_LEVEL - Калибровка стола с очисткой сопла

Параметры, в скобках значение по умолчанию:
EXTRUDER_TEMP - температура экструдера (230)
BED_TEMP - температура стола (80)
PROFILE - для какого профиля (auto)

ghzserg, 2025-02-20:
Не ошибка.

closed 2025-02-20 by ghzserg