ghzserg/zmod #4: Шаблон оформления тикета

seligor, 2025-02-20:
- [ ] Я работаю с экраном
- [ ] Я работаю без экрана
- [x ] Я работаю с Guppyscreen
(установите x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/Macros#tar_config) в чат [zmod](https://t.me/FF_5M_5M_Pro/408369)
прикрепите ссылку на сообщение с файлом

Я написал стартовый код:

```

```

Я написал завершающий код:

```

```

Вот мои параметры SAVE_ZMOD_DATA:

```

```

Подробное описание:

ghzserg, 2025-02-20:
На основании вашего шаблона сделан общий шаблон, посмотрите  всего ли хватает.
Большое  спасибо за помощь

seligor, 2025-02-20:
Мне кажется нет смысла делать баг шаблон. Просто кастомный. Далее уже классифицировать по факту того баг это, запрос фичи или ещё что. 
В шаблоне "я прочитал..." лучше заменить на рекомендацию ознакомиться с документацией перед тем как отправить тикет. Люди в общей массе такие, что не думая отметят все галочки )

closed 2025-02-20 by seligor