ghzserg/zmod #6: Движение по осям не происходит

R9YAG, 2025-02-21:
Ни через гуппи, ни через флюид любая кнопка движения осей не работает. После сегодняшнего обновления, возможно, правили баг что ниже уровня ось Z ездила...

20:07:12 
!! Malformed command '_CLIENT_LINEAR_MOVE Y-50 F1200'
20:07:17 
$ _CLIENT_LINEAR_MOVE Y=-1 F=7800
20:07:17 
!! Move out of range: 110.025 219.050 220.000 [0.000]

R9YAG, 2025-02-21:
[image]

R9YAG, 2025-02-21:
Уточнение:
После перезагрузки из fluidd и штатного экрана оси двигаются.
Когда прогружается гуппи, из него не двигается и fluidd перестает двигаться.
При нажатии кнопки движения в гуппи ошибка - !! Malformed command '_CLIENT_LINEAR_MOVE X-10 F1200'
При нажатии кнопки движения в fluidd ошибка - !! Move out of range: 220.037 110.037 220.000 [0.000]
[image]

ghzserg, 2025-02-21:
На текущий момент есть ошибка с граничными точками

ghzserg, 2025-02-22:
Окончательно исправлено в версии 1.3.1-70.
Работают граничные точки и перемещение.

closed 2025-02-22 by R9YAG