ghzserg/zmod #7: Проблема с отображением эскиза на родном экране

vodyanov, 2025-02-23:
Перед тем, как задавать вопрос прочитайте:
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod 
https://disk.yandex.ru/d/7VLD1MIeyO_opA
(По скрепке не дает привязать файл)

Я использую стартовый код:

```
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
G90
M83
G1 Z5 F6000
G1 E-0.2 F800
G1 X110 Y-110 F6000
G1 E2 F800
G1 Y-110 X55 Z0.25 F4800
G1 X-55 E8 F2400
G1 Y-109.6 F2400
G1 X55 E5 F2400
G1 Y-110 X55 Z0.45 F4800
G1 X-55 E8 F2400
G1 Y-109.6 F2400
G1 X55 E5 F2400
G92 E0
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
05:16:24 
$ GET_ZMOD_DATA
05:16:24 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
05:16:24 
info Вы работаете с родным экраном
05:16:24 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
05:16:24 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
05:16:24 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
05:16:24 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
05:16:24 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
05:16:24 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
05:16:24 
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
05:16:24 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
05:16:24 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
05:16:24 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
05:16:24 
info Параметры используемые при окончании и отмене печати [END_PRINT]
05:16:24 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
05:16:24 
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
05:16:24 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
05:16:24 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
05:16:24 
info Общесистемные параметры
05:16:24 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
05:16:24 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
05:16:24 
// Не исправлять ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=0
05:16:24 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
05:16:24 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
05:16:24 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
05:16:25 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
05:16:25 
// zversion:

```

Подробное описание ошибки:
После последнего обновления ZMOD на экране теперь вместо миниатюры при печати отображается ярлык в виде листа с буковкой G на синем фоне. В орка настройки эскизов не трогал "32x32/PNG, 300x300/PNG, 140x110/PNG"

ghzserg, 2025-02-23:
Нужна фотография, того что вы видите на экране и файл, его gcod. Приложите эти данные.

ghzserg, 2025-02-23:
Провел эксперимент. Отправил на печать новый файл. Его изображение появилось на экране принтера.

vodyanov, 2025-02-23:
Добрый день! Да, конечно. Сейчас пришлю! Так со всеми gcode, после последнего обновления ZMOD
https://disk.yandex.ru/d/Z9gUUIJOR4b9zQ

[image]

vodyanov, 2025-02-23:
Еще заметил закономерность, при очистке сопла перед печатью, то "рисуется" короткая жирная линия, то как в родной прошивке узкая длинная по передней части стола. При этом никакие настройки не трогаются, тем более связанные с очисткой.

ghzserg, 2025-02-23:
1. Это не проблема мода.
2. У вас не включен контроль MD5
3. Родной экран ожидает картинку 140x110 в первой позиции, у вас в первой позиции 32x32, потом 300x300.

Я провел эксперимент и удалил в редакторе 32x32 и 300x300.

В результате картинка появилась на родном экране.

ghzserg, 2025-02-23:
> Еще заметил закономерность, при очистке сопла перед печатью, то "рисуется" короткая жирная линия, то как в родной прошивке узкая длинная по передней части стола. При этом никакие настройки не трогаются, тем более связанные с очисткой.

Создавайте  новый тикет. Одна проблема один тикет иначе будет путаница. 
PS: Смотрите  консоль - там все описывается

vodyanov, 2025-02-23:
Благодарю, раньше таких не было "неувязок", по этому и не предположил, что важна последовательность. 
Еще раз БОЛЬШОЕ СПАСИБО!!!

vodyanov, 2025-02-23:
К сожалению не помогло, все осталось так же. Сейчас последовательность такая "140x110/PNG, 32x32/PNG, 300x300/PNG"  Но эскиз так и не отображается.

vodyanov, 2025-02-23:
Если оставить только 140x110/PNG то эскиз есть, если через запятую добавить любые другие размеры то эскиза не будет.

ghzserg, 2025-02-24:
> Если оставить только 140x110/PNG то эскиз есть, если через запятую добавить любые другие размеры то эскиза не будет.

В любом случае картинку рисует родной экран, мод на это не влияет и влиять не может.
Единственное предположение, что могут повреждаться файлы.

Скрипт для мака приведен на гитхабе, чтобы добавить ему права надо сделать chmod +x имя_скрипта, находясь в папке рядом со скриптом

ghzserg, 2025-02-24:
Ошибку вашу нашел. У вас кривая карта стола. 
START_PRINT работает с картой стола MESH_DATA, которую загружает родной экран.
Карта  стола MESH_DATA снята у вас на высоте  2.2 мм.
После того как отрабатывает START_PRINT вы загружаете карту стола custom_textured_pei_105 из профиля филамента, которая снята у вас на высоте  -1.6.
В общем это не ошибка мода, мод делает все так как описано в документации.
Вам надо или START_PRINT переносить в профиль филамента и в него уже грузить нужный профиль через MESH