ghzserg/zmod #8: Проблема

mkbth, 2025-02-23:
Перед тем, как задавать вопрос прочитайте:
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) и gcode задания печати, на котором воспроизводится ошибка:

[config.tar+gcode](https://github.com/user-attachments/files/18931088/config.tar%2Bgcode.zip)

Я использую стартовый код:
```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
```

Завершающий код:
```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:
```
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
info Вы работаете с родным экраном
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
// SKEW коррекция включена. Профиль skew_profile // SAVE_ZMOD_DATA DISABLE_SKEW=0
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
// Алгоритм очистки: _CLEAR4 // SAVE_ZMOD_DATA CLEAR=_CLEAR4
info Параметры используемые при окончании и отмене печати [END_PRINT]
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
info Общесистемные параметры
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
// Яркость LED при включении: 20 // SAVE_ZMOD_DATA LED=20
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
// SWAP не используется // SAVE_ZMOD_DATA USE_SWAP=0
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
// Используeтся родной Klipper 11.
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
// Вес: 1500; Контроль настроен и не активен.
// zversion:
```

Подробное описание ошибки:

В начале печати, перед очисткой, сопло перестало "тыкаться" в стол. После нагрева стол поднимается где-то до 5мм от сопла, выполняет стабилизацию температуры, далее голова едет выполнять алгоритм очистки, при этом давит пластик в воздух. Потом уже стол поднимается на 0 и начинается нормальная печать.

Видео проблемы: https://t.me/FF_5M_5M_Pro/408369/588769

Проблема так же воспроизводится если задать алгоритм очистки _CLEAR3. Скорее всего и на других тоже.

ghzserg, 2025-02-23:
Вы не корректно сняли tar_config
После clear_emmc надо выключить принтер, а потом его включить. И сгенерировать проблему. В тар конфиг нет лог файлов

ghzserg, 2025-02-24:
Ошибку вашу нашел. У вас кривая карта стола.
START_PRINT работает с картой стола MESH_DATA, которую загружает родной экран.
Карта стола MESH_DATA снята у вас на высоте 2.2 мм.
После того как отрабатывает START_PRINT вы загружаете карту стола custom_textured_pei_105 из профиля филамента, которая снята у вас на высоте -1.6.
В общем это не ошибка мода, мод делает все так как описано в документации.
Вам надо или START_PRINT переносить в профиль филамента и в него уже грузить нужный профиль через MESH

ghzserg, 2025-02-24:
Отключите очистку сопла в START_PRINT и вызывайте ее в профиле прутка.
Документацию поправил.
Спасибо

closed 2025-02-24 by ghzserg