ghzserg/zmod #9: Остывание сопла при срабатывании датчика филамента

seligor, 2025-02-27:
Перед тем, как задавать вопрос прочитайте:
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [X] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я не загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```

```

Я написал завершающий код:

```

```

Вот мои параметры SAVE_ZMOD_DATA:

```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.
```

Подробное описание ошибки:

при срабатывании датчика окончания филамента принтер встаёт на паузу, при этом немного опускает стол и уводит голову в 110,110 (ну или около того)
При этом стол продолжает поддерживать температуру, а сопло начинает остывать, что на мой взгляд вряд ли защищает от подтёков филамента, но делает процесс снятия с паузы дольше. 
В связи с этим предлагаю вынести на обсуждение предложение о поддержании температуры сопла в случае сработки датчика окончания филамента

[image]

ghzserg, 2025-02-27:
Это не ошибка. 
Это повторение алгоритма паузы родного экрана.
Суть простая при постановке на паузу сопло отключается, т.к. пауза ограничена 12 часами и нет смысла держать все 12 часов сопло горячим.
При восстановлении печати температура восстанавливается.
При заправке филамента гуппи использует температуру указанную на экране.

seligor, 2025-02-27:
Я и не говорю, что это ошибка. Но да, логика вполне понятна. Пусть остаётся для истории - можно будет ссылаться на этот тикет. В данном контексте важнее, чтобы деталь не слетела со стола и соответственно нужен подогретый стоо. а сопло нагревается достаточно быстро