ghzserg/zmod #12: Проблема

koltinov, 2025-03-07:
Перед тем, как задавать вопрос прочитайте:
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ Х ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] FORCE_KAMP=True SKIP_LEVELING=True
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count] 
```

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
21:58:32 
info Вы работаете БЕЗ родного экрана
21:58:32 
// Используeтся родной Klipper 11.
21:58:32 
// zssh:
21:58:33 
// Установлено с флешки: 1.4. Обновление с Fluidd/Mainsaill: 1.4
21:58:33 
// Версия zmod: 1.4.0 Date:   Fri Mar 7 19:52:04 2025 +0500
21:58:33 
ok
21:58:53 
// zrestore:
21:58:53 
// Файл восстановления печати не найден
22:00:46 
$ TAR_CONFIG
22:00:46 
info Скачать архив можно в 'Конфигурация' -> 'mod_data' -> config.tar
22:01:08 
info Вы работаете БЕЗ родного экрана
22:01:08 
// Используeтся родной Klipper 11.
22:01:08 
// zssh:
22:01:09 
// Установлено с флешки: 1.4. Обновление с Fluidd/Mainsaill: 1.4
22:01:09 
// Версия zmod: 1.4.0 Date:   Fri Mar 7 19:52:04 2025 +0500
22:02:16 
$ GET_ZMOD_DATA
22:02:16 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
22:02:16 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
22:02:16 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
22:02:16 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
22:02:16 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
22:02:16 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
22:02:16 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
22:02:16 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
22:02:16 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
22:02:16 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
22:02:16 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
22:02:16 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
22:02:16 
info Параметры используемые при окончании и отмене печати [END_PRINT]
22:02:16 
// Играть MIDI в конце печати: KinoQuietNight.mid // SAVE_ZMOD_DATA MIDI_END=KinoQuietNight.mid
22:02:16 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
22:02:16 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
22:02:16 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
22:02:16 
// Выключать FF5M Pro через (мин): 30 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=30
22:02:16 
info Общесистемные параметры
22:02:16 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
22:02:16 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
22:02:16 
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
22:02:16 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
22:02:16 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
22:02:16 
// Яркость LED при включении: 100 // SAVE_ZMOD_DATA LED=100
22:02:17 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
22:02:17 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
22:02:17 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
22:02:17 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
22:02:17 
// Выключать родной экран через (сек): 10 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=10
22:02:17 
// Используeтся родной Klipper 11.
22:02:17 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
22:02:17 
// Вес: 800; Контроль настроен и не активен.
22:02:17 
// zversion:
22:02:17 
// Установлено с флешки: 1.4. Обновление с Fluidd/Mainsaill: 1.4

[config.zip](https://github.com/user-attachments/files/19130481/config.zip)

```

Подробное описание ошибки:
Максимально подробно опишите, что Вы делали, что Вы планировали  получить и что Вы получили.

Кнопка на передней панели в PRO не выключает принтер, вообще никак не реагирует во включенном состоянии принтера. в выключеном состоянии включает корректно

ghzserg, 2025-03-08:
Нужно сделать следующее.
1. Очистить логи, через клеам_еммц
2. Переключиться на родной экран
3. Нажать на родном экране кнопку выключения 
4. Включить принтер.
5. Снять логи через тар_конфиг
6. Приложить полученный файл сюда

koltinov, 2025-03-08:
сделал
[config.zip](https://github.com/user-attachments/files/19142898/config.zip)

ghzserg, 2025-03-08:
Попробуйте прописать в `mod_data/user.cfg`
```
[gcode_button btn_power]
pin: !PC15
press_gcode:
    SHUTDOWN
```
Если начнет выключать, то реализовать окно с вопросом не сложно

koltinov, 2025-03-09:
да, выключает. 

> то реализовать окно с вопросом не сложно
был бы благодарен))

ghzserg, 2025-03-09:
1. Удалите то что внесли в `user.cfg`
2. Обновите  мод до 1.4.0-80
3. Протестируйте

koltinov, 2025-03-09:
нажимаю кнопку, выскакивает "Выключить принтер?", нажимаю ДА - корректно выключает! всё отлично 👌 

нажимаю кнопку, выскакивает "Выключить принтер?", нажимаю ОТМЕНА - отменяет, но потом хоть обнажимайся эту кнопку, больше до перезагрузки не реагирует, "Выключить принтер?" не появляется))

ghzserg, 2025-03-09:
Проверю.
Установите с флешки 1.4.1 возможно проблема уйдет

koltinov, 2025-03-09:
обновил, нет не ушла, только еще обнаружил побочный эффект: теперь любой процесс выключения сопровождается диалогом "выключить или отмена". когда через вэбинтерфейс в правом верхнем углу выключаешь и, главное, вместо автовыключения на ПРО теперь выскакивает этот диалог.

ghzserg, 2025-03-09:
Установите последний апдейт. И проверьте  ушла ли проблема и не появилились ли новые.

closed 2025-03-10 by koltinov