ghzserg/zmod #13: Печать в воздухе после KAMP

mkbth, 2025-03-07:
- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen

Конфигурация: [config.tar.zip](https://github.com/user-attachments/files/19130822/config.tar.zip)
G-code: [Cube20x20_PETG_8m52s.gcode.zip](https://github.com/user-attachments/files/19130854/Cube20x20_PETG_8m52s.gcode.zip)

Стартовый G-code:
```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```
Завершающий G-code:
```
END_PRINT
```

Параметры SAVE_ZMOD_DATA:
```
 20:16:22  info Параметры zmod. Задаются через SAVE_ZMOD_DATA
20:16:22  info Вы работаете с родным экраном // DISPLAY_ON
20:16:22  info Параметры используемые при начале печати, построении карты стола [START_PRINT]
20:16:22  // При каждой печати строить карту стола средствами родного экрана. // SAVE_ZMOD_DATA PRINT_LEVELING=1
20:16:22  // Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
20:16:22  // Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
20:16:22  // Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
20:16:23  // SKEW коррекция включена. Профиль skew_profile // SAVE_ZMOD_DATA DISABLE_SKEW=0
20:16:23  // Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
20:16:23  // Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
20:16:23  // Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
20:16:23  // Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
20:16:23  info Параметры используемые при окончании и отмене печати [END_PRINT]
20:16:23  // Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
20:16:23  // Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
20:16:23  // Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
20:16:23  // Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
20:16:23  // Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
20:16:23  info Общесистемные параметры
20:16:23  // Сохранение управляется родным экраном. // SAVE_ZMOD_DATA SAVE_RESTORE=1
20:16:23  // Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
20:16:23  // Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
20:16:23  // Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
20:16:23  // Не исправлять ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=0
20:16:23  // Яркость LED при включении: 20 // SAVE_ZMOD_DATA LED=20
20:16:23  // Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
20:16:23  // Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
20:16:23  // SWAP не используется // SAVE_ZMOD_DATA USE_SWAP=0
20:16:23  // Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
20:16:23  // Используeтся родной Klipper 11.
20:16:23  // Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
20:16:24  // Вес: 1500; Контроль настроен и не активен.
20:16:24  // zversion:
20:16:24  // Установлено с флешки: 1.4. Обновление с Fluidd/Mainsaill: 1.4
```

Подробное описание ошибки:
Сегодня (07.03) с утра печать стартовала нормально при тех же параметрах. Где-то в 16:00, после завершения очередной печати, обновил zmod стандартно через fluidd. После этого печать стартует в воздухе ([видео](https://t.me/FF_5M_5M_Pro/408369/615703)).

ghzserg, 2025-03-08:
Установите последнее обновление от 08.03.2025

mkbth, 2025-03-09:
Установил последнее обновление - печать стартовала нормально.

closed 2025-03-09 by ghzserg