ghzserg/zmod #15: Проблема

Romka44, 2025-03-08:
Перед тем, как задавать вопрос прочитайте:
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [Х] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG]   (https://t.me/c/2496730150/418) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.
https://t.me/c/2496730150/418

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера

```END_PRINT

Вот мои параметры SAVE_ZMOD_DATA:
$ SAVE_ZMOD_DATA
15:33:07 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
15:33:07 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
15:33:07 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
15:33:07 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
15:33:07 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
15:33:07 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
15:33:07 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
15:33:08 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
15:33:08 
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
15:33:08 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
15:33:08 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
15:33:08 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
15:33:08 
info Параметры используемые при окончании и отмене печати [END_PRINT]
15:33:08 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
15:33:08 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
15:33:08 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
15:33:08 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
15:33:08 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
15:33:09 
info Общесистемные параметры
15:33:09 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
15:33:09 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
15:33:09 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
15:33:09 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
15:33:09 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
15:33:09 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
15:33:09 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
15:33:09 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
15:33:09 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
15:33:09 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
15:33:09 
// Выключать родной экран через (сек): 180 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=180
15:33:09 
// Используeтся родной Klipper 11.
15:33:09 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
15:33:09 
// Контроль веса не настроен. // Для настройки: NOZZLE_CONTROL WEIGHT=1500
15:33:10 
// zversion:
15:33:10 
// Установлено с флешки: 1.4. Обновление с Fluidd/Mainsaill: 1.4

​

Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.

$ GET_ZMOD_DATA
15:32:17 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
15:32:17 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
15:32:17 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
15:32:17 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
15:32:17 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
15:32:17 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
15:32:18 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
15:32:18 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
15:32:18 
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
15:32:18 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
15:32:18 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
15:32:18 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
15:32:18 
info Параметры используемые при окончании и отмене печати [END_PRINT]
15:32:18 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
15:32:18 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
15:32:18 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
15:32:18 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
15:32:18 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
15:32:18 
info Общесистемные параметры
15:32:18 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
15:32:18 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
15:32:18 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
15:32:18 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
15:32:18 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
15:32:18 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
15:32:18 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
15:32:18 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
15:32:18 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
15:32:18 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
15:32:18 
// Выключать родной экран через (сек): 180 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=180
15:32:18 
// Используeтся родной Klipper 11.
15:32:18 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
15:32:19 
// Контроль веса не настроен. // Для настройки: NOZZLE_CONTROL WEIGHT=1500
15:32:19 
// zversion:
15:32:19 
// Установлено с флешки: 1.4. Обновление с Fluidd/Mainsaill: 1.4

​

Подробное описание ошибки:
После установки мода версии 1.4  перестала работать камера (по морганию похоже на режим ожидания).Так вот скачал мод с гитхаба и установил, камера работала.Но после последнего обновления она не показывает.Просто мигает индикатор.Ваш мод очень нравится т.к привык уже к веб интерфейсу на другом принтере.

ghzserg, 2025-03-08:
Tar_config недоступен. Выложите в другое место

Romka44, 2025-03-08:
https://cloud.mail.ru/public/yVT3/kDoxcSBig

ghzserg, 2025-03-08:
Вы отправили пустые логи.
Выполните  по пунктам, все что указано [тут](https://github.com/ghzserg/zmod/wiki/FAQ#help).

Romka44, 2025-03-09:
[config.tar.zip](https://github.com/user-attachments/files/19148591/config.tar.zip)

Romka44, 2025-03-09:
Прошу простить за невнимательность!

ghzserg, 2025-03-09:
Судя по логам
1. У вас не последняя версия мода
2. Камера не  поддерживает MJPG

Обновите  мод/

[image] 
пришлите  скрин

Romka44, 2025-03-09:
[image]

Romka44, 2025-03-09:
До обновы на новую версию камера работала хорошо.

[image]

Romka44, 2025-03-09:
1.4.0-72-gf0fddfae  весия которая установлена

ghzserg, 2025-03-09:
Пришлите лог с обновленной версии

Romka44, 2025-03-09:
[camera.log](https://github.com/user-attachments/files/19149883/camera.log)
[console.1.log](https://github.com/user-attachments/files/19149872/console.1.log)
[console.2.log](https://github.com/user-attachments/files/19149882/console.2.log)
[console.log](https://github.com/user-attachments/files/19149876/console.log)
[fix_config.1.log](https://github.com/user-attachments/files/19149877/fix_config.1.log)
[fix_config.2.log](https://github.com/user-attachments/files/19149881/fix_config.2.log)
[fix_config.3.log](https://github.com/user-attachments/files/19149879/fix_config.3.log)
[fix_config.log](https://github.com/user-attachments/files/19149873/fix_config.log)
[guppyscreen.1.log](https://github.com/user-attachments/files/19149880/guppyscreen.1.log)
[guppyscreen.2.log](https://github.com/user-attachments/files/19149884/guppyscreen.2.log)
[guppyscreen.log](https://github.com/user-attachments/files/19149875/guppyscreen.log)
[ssh.log](https://github.com/user-attachments/files/19149886/ssh.log)
[zmod.1.log](https://github.com/user-attachments/files/19149878/zmod.1.log)
[zmod.2.log](https://github.com/user-attachments/files/19149885/zmod.2.log)
[zmod.log](https://github.com/user-attachments/files/19149874/zmod.log)

Romka44, 2025-03-09:
если правильно понимаю то их папки логи отправил файлы

ghzserg, 2025-03-09:
Да с логами ок.
Зайдите на http://IP:8080/
И пройдитесь по ссылкам там. 
Приложите  скрины.

Romka44, 2025-03-09:
[image]

Romka44, 2025-03-09:
[image]
[image]
[image]
[image]
[image]
[image]

ghzserg, 2025-03-09:
Пока для теста установите версию 1.3.0 и обновления не ставьте

Romka44, 2025-03-09:
Чудеса,заработала!

Romka44, 2025-03-09:
Прошивку не менял,стоит последняя.

closed 2025-03-10 by ghzserg