ghzserg/zmod #16: Проблема

loss-and-quick, 2025-03-08:
- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [X] Я работаю с Guppyscreen

[TAR_CONFIG](https://upload.disroot.org/r/0aTF6jX7#Z6XJLcNzJZIdeoL/H6K8JJuSmjLjdAXCVLGgKSv7/QY=)

```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
19:05:19 
info Вы работаете с родным экраном // DISPLAY_ON
19:05:19 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
19:05:19 
// При каждой печати строить карту стола средствами родного экрана. // SAVE_ZMOD_DATA PRINT_LEVELING=1
19:05:19 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
19:05:19 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
19:05:19 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
19:05:19 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
19:05:19 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
19:05:19 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
19:05:19 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
19:05:19 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
19:05:19 
info Параметры используемые при окончании и отмене печати [END_PRINT]
19:05:19 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
19:05:19 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
19:05:19 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
19:05:19 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
19:05:19 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
19:05:19 
info Общесистемные параметры
19:05:20 
// Сохранение управляется родным экраном. // SAVE_ZMOD_DATA SAVE_RESTORE=1
19:05:20 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
19:05:20 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
19:05:20 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
19:05:20 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
19:05:20 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
[... 694 lines cut ...]
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x1380  3x 896 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       7
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      2 Video Streaming
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x1400  3x 1024 bytes
        bInterval               1
```

ghzserg, 2025-03-08:
[image]
Нужно приложить скриншот вот из этого места. 
Судя по логам /dev/video0 отсутсвует, возможно другое имя устройства

loss-and-quick, 2025-03-08:
Вы понимаете, что раз дискриптор не может быть определен, то и никакого монтирования в систему не будет происходить

loss-and-quick, 2025-03-08:
Я попробую перезаказать этот же модуль, возможно в usb части модуля проблема

ghzserg, 2025-03-08:
Если нет устройства /dev/video0 или любого другого видео устройства, то это не проблема мода.
Это физическая проблема или нет нужных драйверов ядре

closed 2025-03-09 by ghzserg