ghzserg/zmod #17: Проблема

securny, 2025-03-08:
- [] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [X] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я использую стартовый код:

```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
21:10:27 
$ GET_ZMOD_DATA
21:10:27 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
21:10:28 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
21:10:28 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
21:10:28 
// При каждой печати строить карту стола. // SAVE_ZMOD_DATA PRINT_LEVELING=1
21:10:28 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
21:10:28 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
21:10:28 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
21:10:28 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
21:10:28 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
21:10:28 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
21:10:28 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
21:10:28 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
21:10:28 
info Параметры используемые при окончании и отмене печати [END_PRINT]
21:10:28 
// Играть MIDI в конце печати: BattleCity.mid // SAVE_ZMOD_DATA MIDI_END=BattleCity.mid
21:10:28 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
21:10:28 
// Не выключать моторы автоматически, после печати/отмены печати. // SAVE_ZMOD_DATA STOP_MOTOR=0
21:10:28 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
21:10:28 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
21:10:28 
info Общесистемные параметры
21:10:28 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
21:10:28 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
21:10:28 
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
21:10:28 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
21:10:29 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
21:10:29 
// Яркость LED при включении: 0 // SAVE_ZMOD_DATA LED=0
21:10:29 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
21:10:29 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
21:10:29 
// SWAP используется на USB FLASH// SAVE_ZMOD_DATA USE_SWAP=2
21:10:29 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
21:10:29 
// Выключать родной экран через (сек): 180 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=180
21:10:29 
// Используeтся родной Klipper 11.
21:10:29 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
21:10:30 
// Вес: 1500; Контроль настроен и активен.
21:10:30 
// zversion:
21:10:30 
// Установлено с флешки: 1.4. Обновление с Fluidd/Mainsaill: 1.4
```

Подробное описание ошибки:
- Обновился до 1.4 с флешки
- Обновил весь софт через веб
- Перезагрузился
- Через 180 сек родной экран зависает, а гуппи не появляется

[config.tar.zip](https://github.com/user-attachments/files/19145164/config.tar.zip)

ghzserg, 2025-03-08:
Я посмотрел логи и вижу, что гуппи должен был запуститься, но этого не произошло.
Поэтому скачайте  еще раз версию 1.4.0 - она была обновлена на гите.
Сравните  md5 скаченного файла.
Установите все обновления через fluidd - я внес логирование запуска гуппи.
И снова выложите  TAR_CONFIG

ghzserg, 2025-03-10:
Установите  1.4.1 и проведите  CHECK_SYSTEM

securny, 2025-03-10:
Проблема решена:
- скачал, проверил md5 и установил 1.4.1
- обновился через веб до 1.4.1-9-g29c86921
- все заработало, гуппи включился

Спасибо!

closed 2025-03-10 by securny