ghzserg/zmod #18: Не проходит установка

grizzart, 2025-03-09:
версия zmod 1.4.0
версия по 3.1.4

+ install_prepare
+ cat /data/update/mod/version.txt
cat: can't open '/data/update/mod/version.txt': No such file or directory

grizzart, 2025-03-09:
Расширенные логи последнего мода

+ install_prepare
+ mount
/dev/root on / type ext4 (rw,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,relatime,mode=755)
sysfs on /sys type sysfs (rw,relatime)
devtmpfs on /dev type devtmpfs (rw,nosuid,relatime,size=45748k,nr_inodes=11437,mode=755)
none on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
none on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p7 on /data type ext4 (rw,relatime)
/dev/sda on /mnt type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=936,iocharset=utf8,shortname=mixed,utf8,errors=remount-ro)
+ df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.2G    782.0M    441.8M  64% /
df: /dev/shm: No such file or directory
tmpfs                    53.9M     28.0K     53.9M   0% /tmp
tmpfs                    53.9M     28.0K     53.9M   0% /run
devtmpfs                 44.7M         0     44.7M   0% /dev
/dev/mmcblk0p7            5.8G    134.6M      5.3G   2% /data
/dev/sda                  3.7G    122.1M      3.6G   3% /mnt

ghzserg, 2025-03-09:
Приложите весь файл установки с последнего мода

grizzart, 2025-03-09:
[zmod_install.log](https://github.com/user-attachments/files/19148854/zmod_install.log)

grizzart, 2025-03-09:
[zmod_install.log](https://github.com/user-attachments/files/19148896/zmod_install.log)

провел еще раз туже опурацию.
При загрузке показывает установку пару секунд и выдает картинку красную с надписью вытащите флешку и выключите принтер. я подождал и вытащил. но файл такойже

nazbav, 2025-03-09:
> Приложите весь файл установки с последнего мода

FF5M
Попытка просто обновиться с 0.2.4 до последней выдала ошибку.
Снес до завода старый Z-MOD (0.2.4), снес KlipperMod, накатил дефолтную прошивку с папки.
Актуалка 3.1.5-2.2.3

Флешка FAT32, кластер 8192. Перед извлечением -- программное извлечение с винды.

На флешке архив:  Adventurer5M-zmod-1.4.0.tgz

Папок: 48
Файлов: 299
Размер: 127801233 байтов (121 MiB)
Сумма CRC32 для данных: F1738396-0000009A
Сумма CRC32 для данных и имен: 1A7A9D4F-000000AD
Сумма CRC64 для данных: 531D2720119B031C-00000098
Сумма CRC64 для данных и имен: E228BF85BDC856E0-000000B1
Сумма SHA256 для данных: 0a191b8544e9469c73102b0364860856424903db34eb8505d51932cbf649d652-00000094
Сумма SHA256 для данных и имен: 4c0c59a333aafa5c6a816a44fef8218b50671f6580133b7f66b6576b04d17fcd-000000A9
Сумма SHA1 для данных: 0ade9f8564a38b214d2e695569b03795fa48b513-00000097
Сумма SHA1 для данных и имен: f31ce6b37de61c0c46f15fb14e902d344a73b4d3-000000AD
Сумма BLAKE2sp для данных: e3263d21d83bfd345a3109c91741d38e31f616a34f1433b5fdd86df43a293c62-00000093
Сумма BLAKE2sp для данных и имен: 9a57e045c13c7429110d20d5d2a53b7841f3d6850900ecdd12deb8c345948ba7-000000A9

Лог файл после (НЕ) установки:
+ install_prepare
+ mount
/dev/root on / type ext4 (rw,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,relatime,mode=755)
sysfs on /sys type sysfs (rw,relatime)
devtmpfs on /dev type devtmpfs (rw,nosuid,relatime,size=45748k,nr_inodes=11437,mode=755)
none on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
none on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p7 on /data type ext4 (rw,relatime)
/dev/sda1 on /mnt type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=936,iocharset=utf8,shortname=mixed,utf8,errors=remount-ro)
+ df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.2G    782.7M    441.1M  64% /
df: /dev/shm: No such file or directory
tmpfs                    53.9M     28.0K     53.9M   0% /tmp
tmpfs                    53.9M     28.0K     53.9M   0% /run
devtmpfs                 44.7M         0     44.7M   0% /dev
/dev/mmcblk0p7            5.8G    739.4M      4.7G  13% /data
/dev/sda1                29.0G    122.2M     28.9G   0% /mnt

[image]

https://disk.yandex.ru/d/D-vMGvUL2S2AaA

ghzserg, 2025-03-09:
Скачайте еще раз 1.4.0
Чексума должна быть 4ca52c1297386b379f8b5ad90ef16e06

noyhay1, 2025-03-09:
у меня то же самое, версия ориг прошивки 3.1.5, пробовал разные флешки и разный размер кластера.
Файлы, также похоже не работают, потомучто ничего не  меняется
Adventurer5MPro-3.1.3-2.2.3-20250107-Factory-PRO25M

кэш суммы соответствуют

ghzserg, 2025-03-09:
> у меня то же самое, версия ориг прошивки 3.1.5, пробовал разные флешки и разный размер кластера. Файлы Adventurer5MPro-3.1.3-2.2.3-20250107-Factory-PRO25M, также похоже не работают, потомучто ничего не меняется
> 
> кэш суммы соответствуют
> 
> [zmod_install.log](https://github.com/user-attachments/files/19149194/zmod_install.log)

Чек сумма должна быть такой `4ca52c1297386b379f8b5ad90ef16e06`

nazbav, 2025-03-09:
> Скачайте еще раз 1.4.0 Чексума должна быть 4ca52c1297386b379f8b5ad90ef16e06

PS F:\> certutil -hashfile Adventurer5M-zmod-1.4.0.tgz md5
Хэш MD5 Adventurer5M-zmod-1.4.0.tgz:
4ca52c1297386b379f8b5ad90ef16e06
CertUtil: -hashfile — команда успешно выполнена.
PS F:\>

Уже лучше, на первых секундах не упал. ждем.

grizzart, 2025-03-09:
[zmod_install.log](https://github.com/user-attachments/files/19149207/zmod_install.log)

да с новым файлом намного дольше установка была

ghzserg, 2025-03-09:
xz: corrupted data
Побился файл

noyhay1, 2025-03-09:
> xz: corrupted data Побился файл
так вы зачем выложили кеш суммы тогда?) для этого и выкладывают)
Ну главное разобрались, спасибо!
Все установилось!

ghzserg, 2025-03-09:
> > xz: corrupted data Побился файл
> > так вы зачем выложили кеш суммы тогда?) для этого и выкладывают)
> > Ну главное разобрались, спасибо!
> > Все установилось!

Так у вас файл побился когда с флешки записывался на принтер.

noyhay1, 2025-03-09:
> > > xz: corrupted data Побился файл
> > > так вы зачем выложили кеш суммы тогда?) для этого и выкладывают)
> > > Ну главное разобрались, спасибо!
> > > Все установилось!
> 
> Так у вас файл побился когда с флешки записывался на принтер.

а у меня вопрос, не много не по теме
Управление светом чтобы с экрана было, нужно переменовать файл устновки с приставкой Pro на обычную версию?

ghzserg, 2025-03-09:
> [zmod_install.log](https://github.com/user-attachments/files/19149225/zmod_install.log)
> 
> > Уже лучше, на первых секундах не упал. ждем.
> 
> Все встало (по крайней мере, Fluidd открылся)

У вас не встал entware

ghzserg, 2025-03-09:
> > > > xz: corrupted data Побился файл
> > > > так вы зачем выложили кеш суммы тогда?) для этого и выкладывают)
> > > > Ну главное разобрались, спасибо!
> > > > Все установилось!
> > 
> > 
> > Так у вас файл побился когда с флешки записывался на принтер.
> 
> а у меня вопрос, не много не по теме Управление светом чтобы с экрана было, нужно переменовать файл установки с приставкой Pro?

[Перешивка в Pro](https://github.com/ghzserg/zmod/tree/main/%D0%A0%D0%BE%D0%B4%D0%BD%D0%B0%D1%8F_%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0/5m2Pro)

noyhay1, 2025-03-09:
> [Перешивка в Pro](https://github.com/ghzserg/zmod/tree/main/%D0%A0%D0%BE%D0%B4%D0%BD%D0%B0%D1%8F_%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0/5m2Pro)

понял, спасибо, уже не нужно, на алитернативном экране есть

grizzart, 2025-03-09:
> xz: corrupted data Побился файл Пробуйте еще раз

C:\Users\gd>certutil -hashfile Adventurer5M-zmod-1.4.0.tgz md5
Хэш MD5 Adventurer5M-zmod-1.4.0.tgz:
4ca52c1297386b379f8b5ad90ef16e06
CertUtil: -hashfile — команда успешно выполнена.

результат тотже 

[zmod_install.log](https://github.com/user-attachments/files/19149301/zmod_install.log)

ghzserg, 2025-03-09:
Выложил версию инсталятора, которая проверяет чек сумму критичных файлов, после того как родной установщик  копирует их с флешки в память принтера.
Нужно снова скачать 1.4.0 и чек сумма должна быть `f4929ff9ac18cf01cc9966974a647ca8`

grizzart, 2025-03-09:
> Выложил версию инсталятора, которая проверяет чек сумму критичных файлов, после того как родной установщик копирует их с флешки в память принтера. Нужно снова скачать 1.4.0 и чек сумма должна быть `f4929ff9ac18cf01cc9966974a647ca8`
> 
> В логе это отображается так
> 
> ```
> + install_prepare
> + pushd /data/update/mod/
> /data/update/mod /
> + md5sum -c md5sum.list
> entware.tar.xz: OK
> rootfs.tar.xz: OK
> version.txt: OK
> ```

[zmod_install.log](https://github.com/user-attachments/files/19149411/zmod_install.log)

ghzserg, 2025-03-09:
**entware.tar.xz: FAILED**
rootfs.tar.xz: OK
version.txt: OK
md5sum: WARNING: 1 of 3 computed checksums did NOT match
У вас при копировании/распаковки с флешки побились данные. 
Копирование и распаковку проводит родная прошивка, не мод. 
Мод только проверяет корректность

noyhay1, 2025-03-09:
> C:\Users\gd>certutil -hashfile Adventurer5M-zmod-1.4.0.tgz md5 Хэш MD5 Adventurer5M-zmod-1.4.0.tgz: 4ca52c1297386b379f8b5ad90ef16e06 CertUtil: -hashfile — команда успешно выполнена.
>
> результат [...]

там еще правильно надо качать файл

[image]

noyhay1, 2025-03-09:
> **entware.tar.xz: FAILED** rootfs.tar.xz: OK version.txt: OK md5sum: WARNING: 1 of 3 computed checksums did NOT match У вас при копировании/распаковки с флешки побились данные. Копирование и распаковку проводит родная прошивка, не мод. Мод только проверяет корректность

у вас не работает ссылка на [Не могу камеру включить. нажимаю CAMERA_ON а мне в ответ "отключите камеру на экране принтера". там я в настройках её отключил, принтер перезагрузил/](https://github.com/ghzserg/zmod/wiki/FAQ#%D0%BD%D0%B5-%D0%BC%D0%BE%D0%B3%D1%83-%D0%BA%D0%B0%D0%BC%D0%B5%D1%80%D1%83-%D0%B2%D0%BA%D0%BB%D1%8E%D1%87%D0%B8%D1%82%D1%8C-%D0%BD%D0%B0%D0%B6%D0%B8%D0%BC%D0%B0%D1%8E-camera_on-%D0%B0-%D0%BC%D0%BD%D0%B5-%D0%B2-%D0%BE%D1%82%D0%B2%D0%B5%D1%82-%D0%BE%D1%82%D0%BA%D0%BB%D1%8E%D1%87%D0%B8%D1%82%D0%B5-%D0%BA%D0%B0%D0%BC%D0%B5%D1%80%D1%83-%D0%BD%D0%B0-%D1%8D%D0%BA%D1%80%D0%B0%D0%BD%D0%B5-%D0%BF%D1%80%D0%B8%D0%BD%D1%82%D0%B5%D1%80%D0%B0-%D1%82%D0%B0%D0%BC-%D1%8F-%D0%B2-%D0%BD%D0%B0%D1%81%D1%82%D1%80%D0%BE%D0%B9%D0%BA%D0%B0%D1%85-%D0%B5%D1%91-%D0%BE%D1%82%D0%BA%D0%BB%D1%8E%D1%87%D0%B8%D0%BB-%D0%BF%D1%80%D0%B8%D0%BD%D1%82%D0%B5%D1%80-%D0%BF%D0%B5%D1%80%D0%B5%D0%B7%D0%B0%D0%B3%D1%80%D1%83%D0%B7%D0%B8%D0%BB)

ghzserg, 2025-03-09:
> у вас не работает ссылка на [Не могу камеру включить. нажимаю CAMERA_ON а мне в ответ "отключите камеру на экране принтера". там я в настройках её отключил, принтер [...]

Давайте  отдельным тикетом
Замусоревается содержание

ghzserg, 2025-03-09:
> > **entware.tar.xz: FAILED** rootfs.tar.xz: OK version.txt: OK md5sum: WARNING: 1 of 3 computed checksums did NOT match У вас при копировании/распаковки с флешки побились данные. Копирование и распаковку проводит родная прошивка, не мод. Мод только проверяет корректность
> 
> а чего делать по итогу)

Использовать другую флешку, более быструю или более медленную.
У вас бьются файлы при копировании, еще до этапа работы мода.

vodyanov, 2025-03-09:
> Вышла версия 1.4.1. В ней вклчбюен контроль файлов при установке, а также контроль уже установленннлой системы

Какие-то проблемы с репозиторием. Ни одна  версия tar не качается, в raw находится только описание и чексумма.

ghzserg, 2025-03-10:
Гитхаб заблокировал LFS.
Качать отсюда https://github.com/ghzserg/zmod/releases/tag/1.4.1

ghzserg, 2025-03-10:
Тикет закрываю, т.к. вышла  1.4.1 с контролем установки. 
Ошибки установки аппаратные и проявляются еще до передачи управления моду

closed 2025-03-10 by ghzserg

grizzart, 2025-03-10:
пришел второй принтер сейчас на нем проверю на разных флешках