ghzserg/zmod #21: Новый функционал: рассмотреть возможность получения температуры драйверов

seligor, 2025-03-10:
- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [X] Я работаю с Guppyscreen

Предполагаемая реализация: 
- [ ] Вариант а) по какому то макросу разово либо при переходе на новый срой пусть в консоль сообщает
- [ ] Вариант б) добавить в график температур. но это чревато появлением минимум трёх графиков и по сути нужно не всегда, а периодически для контроля охлаждения

На мой взгляд вариант а более подходящий. его можно использовать просто добавлением макроса при смере слоя. и при желании убрать этот макрос чтобы датчики не опрашивались. не думаю, что это будет большой нагрузкой для системы, но может помочь для оценки достаточности охлаждения драйверов

ghzserg, 2025-03-10:
Тут основной вопрос, где  можно взять эти температуры и есть ли они вообще.

ghzserg, 2025-03-10:
Кто первый напишеит, как это можно сделать, то решит эту задачу. Сам не могу найти на это время.

ghzserg, 2025-03-10:
Какие у нас драйвера? 
Вроде клипер умеет получать температуру только с TMC2240

closed 2025-03-10 by ghzserg