ghzserg/zmod #22: Проблема номер текущего слоя в Guppy всегда +1

R9YAG, 2025-03-11:
Проблема номер отображаемого текущего слоя в Guppy всегда +1. Пофиксить по возможности.

ghzserg, 2025-03-12:
Guppy выводит текущий слой, именно тот который получает от Monnraker. 
Я сделал вывод слоя меньше  на 1
Тестируйте.
1.4.1-48

R9YAG, 2025-03-12:
в целом, работает во время печати верно. с точки зрения гуппи 0 слой стал -1 (показывает перед началом очистки), ну да и ладно.

R9YAG, 2025-03-12:
также последний слой не отображает. из 100 слоев последние два будут 99, а потом переходит в -1.

ghzserg, 2025-03-13:
Считаю, что городить огород нет смысла. А надо вводить те же слои, что отдает мунракер.

ghzserg, 2025-03-13:
Сделал апдейт - посмотрите как сейчас

ghzserg, 2025-03-14:
Отображение изменено

R9YAG, 2025-03-16:
Предлагаю вернуть изначальный вариант, как было с отклонением в 1 слой. Сейчас цифры хаотично прыгают, могут на 50 слоев вверх например, версия 1.4.1-111-g78610e2e

closed 2025-03-17 by ghzserg