ghzserg/zmod #23: Доработка NOZZLE_CONTROL

Kazak56, 2025-03-12:
Здравствуйте!

У меня возникла проблема с работой принтера: когда я печатал несколько небольших моделей, почти в конце печати сопло уперлось в небольшой наплыв от предыдущего слоя, и сработал алгоритм контроля отрыва детали. После этого принтер перешёл в аварийный режим.

В аварийном режиме принтер не подал никакого сигнала и выключил подогрев стола, что, в свою очередь, привело к отлипанию всех моделей от стола. Возобновление печати не помогло т.к. все модели уже отлипли от стола.

Я хочу предложить вам немного изменить алгоритм, так как он хорошо работает только при печати одной модели. Предлагаю изменить алгоритм следующим образом: при ударе стола или отрыве детали принтер не должен выдавать критическую ошибку, при которой выключается нагрев стола и сопла. Вместо этого принтер должен вставать на паузу, издавать звуковой сигнал о срабатывании алгоритма и плавно возвращать экструдер в позицию HOME. Температура экструдера должна падать до 120 градусов, чтобы пластик не вытекал из хотенда. Подогрев стола также должен оставаться включённым хотя бы некоторое время (для безопасности), чтобы у оператора было достаточно времени для удаления/исключения оторвавшейся модели и продолжения работы с теми, которые ещё можно допечатать.

Огромное спасибо что занимаетесь доработкой этого принтера!

Готов помочь с тестированием данного алгоритма, здоровья вам и вашим близким.

ghzserg, 2025-03-12:
Вы по сути хотите  ставить принтер на паузу если оторвало деталь.
Но постановка на паузу выполнится уже после  очереди команд. 
Возьмем такой вариант.
Принтер начинает очищать сопло и срабатывает датчик.
В очередь уже успели попасть команды очистки сопла и снятия карты стола.
В этом случае принтер встанет на паузу, только после  того как окончательно продерет стол.
В Klipper для прерывания процесса есть только команда  аварийного завершения, которая по сути и вызывается в этом случае.

Kazak56, 2025-03-12:
Вы абсолютно правы. А можно ли разделить понятия касания стола и отрыва детали? Тогда мы могли бы реализовать вашу идею для случая касания стола, а для отрыва детали использовать предложенный вариант?

seligor, 2025-03-12:
Здесь потребуется вебкамера и ИИ с анализом картинки, который будет обнаруживать движущиеся детали кроме печатающей головки, при этом отбрасывать движения zhop. Принтер с этой задачей не справится. Внешний хост - возможно.  но это уже не имеет отношения к моду

Kazak56, 2025-03-12:
Я думаю вы сильно усложняете, допустим пока принтер не начал печать, а выполняет подготовительные действия будет отслеживаться именно удар об стол с экстренной остановкой, а при печати допустим начиная с 3 слоя начинать отслеживать именно срыв детали (так же по весам) но уже в случае срабатывания просто ставить печать на пайзу с последующим оповещением, как я понимаю после начала печати вряд ли возможно касания экструдером стола.

ghzserg, 2025-03-12:
Это можно реализовать, но это усложнение исходного кода клипера.
Сейчас все реализовано максимально просто - в модуле температуры проверяется превышение и если оно есть, все останавливается.
В Вашем варианте  нужно будет вставлять код обращения к модуля с паузой и отслеживание высоты печати. Это все можно реализовать - но нужно время и желание.

ghzserg, 2025-03-12:
В общем реализовал. Просьба протестировать.
По умолчанию защита работает в режиме прерывания работы
Чтобы активировать режим паузы надо например в смену слоя прописать 
`ZCONTROL_PAUSE` - для работы паузы
`ZCONTROL_ABORT` - для прерывания работы

Kazak56, 2025-03-12:
В ходе тестирования выяснилось:
   1. после слоя на котором вставлен код для работы паузы, код действительно срабатывает и в консоли появляется команда - !! Удар сопла о стол или отрыв детали . PAUSE;
   2. экструдер замирает на одном месте, и на экране появляется надпись о срабатывания алгоритма, с выбором закрытия сообщения и FIRMWARE_RESTART;
   3. после этого не срабатывает команда RESUME, она вроде как прописывается но после этой команды не чего не происходит и принтер перестаёт реагировать на другие консольные команды;
   4. не какими средствами возобновить печать не получается;
   5. температура стола и экструдера остаётся в рабочем режиме;
   6. пауза с экрана срабатывает штатно.

ghzserg, 2025-03-13:
Установите обновление от 13 марта и протестируйте, должно работать лучше

Kazak56, 2025-03-13:
Спасибо, в ходе теста данный алгоритм показал стабильность, подскажите пожалуйста можно ли его заставить дать звуковое сопровождение по срабатыванию алгоритма?

ghzserg, 2025-03-13:
Только переопределив макрос паузы
Или как второй вариант можно ввести глобальный параметр играть звук при паузе, как это сделано для старта печати

ghzserg, 2025-03-13:
> Спасибо, в ходе теста данный алгоритм показал стабильность, подскажите пожалуйста можно ли его заставить дать звуковое сопровождение по срабатыванию алгоритма?

Новый глобальный параметр MIDI_PAUSE
Обновитесь

closed 2025-03-14 by ghzserg