ghzserg/zmod #24: Вызов макроса контроль сопла перезапускает принтер

Zarcuss, 2025-03-12:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)**
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG]

[image]

Я использую стартовый код:
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]

Я использую завершающий код:
END_PRINT

Вот мои параметры SAVE_ZMOD_DATA:

15:19:44 
$ FIRMWARE_RESTART
15:19:44 
// Klipper state: Disconnect
15:20:11 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
15:20:11 
info Вы работаете с родным экраном // DISPLAY_ON
15:20:11 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
15:20:11 
// При каждой печати строить карту стола средствами родного экрана. // SAVE_ZMOD_DATA PRINT_LEVELING=1
15:20:11 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
15:20:11 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
15:20:11 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
15:20:11 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
15:20:11 
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
15:20:11 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
15:20:11 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
15:20:11 
// Алгоритм очистки: _CLEAR4 // SAVE_ZMOD_DATA CLEAR=_CLEAR4
15:20:11 
info Параметры используемые при окончании и отмене печати [END_PRINT]
15:20:11 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
15:20:11 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
15:20:11 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
15:20:11 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
15:20:11 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
15:20:11 
info Общесистемные параметры
15:20:11 
// Сохранение управляется родным экраном. // SAVE_ZMOD_DATA SAVE_RESTORE=1
15:20:11 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
15:20:11 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
15:20:11 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
15:20:11 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
15:20:12 
// Яркость LED при включении: 0 // SAVE_ZMOD_DATA LED=0
15:20:12 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
15:20:12 
// Использовать альтернативный NEW_SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=1
15:20:12 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
15:20:12 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
15:20:12 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
15:20:12 
// Используeтся родной Klipper 11.
15:20:12 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
15:20:12 
// Вес: 1500; Контроль настроен и не активен.
15:20:12 
// zversion:
15:20:13 
// Установлено с флешки: 1.4.1. Обновление с Fluidd/Mainsaill: 1.4.1

[config (1)_2.zip](https://github.com/user-attachments/files/19208806/config.1._2.zip)

Подробное описание ошибки:
Вызов макроса контроль сопла  перезагружает принтер,  после перезагрузки в консоли ошибка
15:18:33 
// Can not update MCU 'eboard' config as it is shutdown
// Once the underlying issue is corrected, use the
// "FIRMWARE_RESTART" command to reset the firmware, reload the
// config, and restart the host software.
// Error configuring printer
15:18:33 
!! Can not update MCU 'eboard' config as it is shutdown
15:18:33 
ok

ghzserg, 2025-03-12:
У вас вылетел MCU Eboard, введите  FIRMWARE_RESTART

ghzserg, 2025-03-12:
[Документация по макросу](https://github.com/ghzserg/zmod/wiki/Macros#nozzle_control)

closed 2025-03-12 by Zarcuss