ghzserg/zmod #27: AD5X после установки мода - экран замерз, вэбинтерфейс недоступен

mortonlame, 2025-03-14:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)**
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ ] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Подробное описание ошибки:

[zmod_install (3).log](https://github.com/user-attachments/files/19247424/zmod_install.3.log)

# ps
  PID USER       VSZ STAT COMMAND
    1 root      3132 S    {linuxrc} init
    2 root         0 SW   [kthreadd]
    3 root         0 IW<  [rcu_gp]
    4 root         0 IW<  [rcu_par_gp]
    8 root         0 IW<  [mm_percpu_wq]
    9 root         0 SW   [rcu_tasks_kthre]
   10 root         0 SW   [ksoftirqd/0]
   11 root         0 IW   [rcu_preempt]
   12 root         0 SW   [migration/0]
   13 root         0 IW   [kworker/0:1-eve]
   14 root         0 SW   [cpuhp/0]
   15 root         0 SW   [cpuhp/1]
   16 root         0 SW   [migration/1]
   17 root         0 SW   [ksoftirqd/1]
   20 root         0 SW   [kdevtmpfs]
   29 root         0 IW   [kworker/1:1-eve]
  289 root         0 SW   [khungtaskd]
  290 root         0 SW   [oom_reaper]
  291 root         0 IW<  [writeback]
  321 root         0 IW<  [kintegrityd]
  322 root         0 IW<  [kblockd]
  423 root         0 SW   [watchdogd]
  503 root         0 IW   [kworker/u4:2-ev]
  510 root         0 IW<  [cfg80211]
  518 root         0 SW   [kswapd0]
  569 root         0 IW<  [felix-vdec]
  574 root         0 IW<  [kworker/u5:0]
  575 root         0 IW<  [sdhci]
  576 root         0 SW   [irq/45-mmc0]
  582 root         0 IW   [kworker/1:3-rcu]
  641 root         0 IW<  [mmc_complete]
  680 root      3548 S    /sbin/mdev -df
  731 root         0 IW<  [kworker/1:2H-kb]
  744 root         0 IW<  [kworker/0:2H-mm]
  746 root         0 SW   [jbd2/mmcblk0p6-]
  747 root         0 IW<  [ext4-rsv-conver]
  823 root         0 IW<  [fb_work]
  830 root         0 IW<  [aic_work]
  932 root         0 SW   [jbd2/mmcblk0p7-]
  933 root         0 IW<  [ext4-rsv-conver]
  950 dbus      3244 S    dbus-daemon --system
  983 root      2828 S    /usr/sbin/dropbear -R
 1137 root         0 SW   [irq/73-tsc2007]
 1179 root         0 SW   [RTW_CMD_THREAD]
 1196 root      444m S    /usr/prog/PROGRAM/software/firmwareExe -1 -D -qws
 1247 root     31308 S<   /usr/prog/Python-3.8.2/bin/python3 /usr/prog/klipper/klippy/klippy.py /usr/data/config/print
 1323 root      2572 S    /opt/config/mod/.shell/root/console_log --not-save
 1352 root      2992 S    httpd -p 80 -f -h /root/fluidd
 1390 root      3132 S    /sbin/getty -L console 0 vt100
 1414 root      3340 S    /sbin/udhcpc -i eth0 -p /var/run/udhcpc.pid
 1511 root      3320 S    /usr/sbin/dropbear -R
 1515 root      3132 S    -sh
 1522 root      3320 R    /usr/sbin/dropbear -R
 1523 root      3288 S    -sh
 1553 root         0 IW<  [kworker/1:0H-kb]
 1565 root         0 IW<  [kworker/0:0H]
 1567 root         0 IW   [kworker/0:0-eve]
 1578 root         0 IW   [kworker/u4:0-ev]
 1587 root         0 IW   [kworker/0:2-mm_]
 1588 root         0 IW<  [kworker/1:1H-kb]
 1589 root         0 IW<  [kworker/0:1H]
 1609 root      3288 R    ps

# mount
/dev/root on / type squashfs (ro,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=242404k,nr_inodes=55666,mode=755)
tmpfs on /dev type tmpfs (rw,relatime)
proc on /proc type proc (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=666)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,relatime,mode=755)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p6 on /usr/prog type ext4 (rw,sync,relatime)
/dev/mmcblk0p7 on /usr/data type ext4 (rw,sync,relatime)
none on /sys/kernel/config type configfs (rw,relatime)
adb on /dev/usb-ffs/adb type functionfs (rw,relatime)
/dev/mmcblk0p6 on /etc type ext4 (rw,sync,relatime)
/proc on /usr/data/.mod/.zmod/proc type proc (rw,relatime)
sysfs on /usr/data/.mod/.zmod/sys type sysfs (rw,relatime)
none on /usr/data/.mod/.zmod/sys/kernel/config type configfs (rw,relatime)
tmpfs on /usr/data/.mod/.zmod/dev type tmpfs (rw,relatime)
devpts on /usr/data/.mod/.zmod/dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=666)
tmpfs on /usr/data/.mod/.zmod/dev/shm type tmpfs (rw,relatime,mode=777)
adb on /usr/data/.mod/.zmod/dev/usb-ffs/adb type functionfs (rw,relatime)
tmpfs on /usr/data/.mod/.zmod/tmp type tmpfs (rw,relatime)
tmpfs on /usr/data/.mod/.zmod/run type tmpfs (rw,nosuid,nodev,relatime,mode=755)
/dev/mmcblk0p7 on /usr/data/.mod/.zmod/opt/config type ext4 (rw,sync,relatime)
/dev/mmcblk0p7 on /usr/data/.mod/.zmod/usr/data/config type ext4 (rw,sync,relatime)
/dev/mmcblk0p7 on /opt type ext4 (rw,sync,relatime)
/dev/mmcblk0p7 on /opt/config type ext4 (rw,sync,relatime)
/dev/mmcblk0p7 on /usr/data/.mod/.zmod/usr/data/logs type ext4 (rw,sync,relatime)
/dev/mmcblk0p6 on /usr/data/.mod/.zmod/usr/prog/klipper type ext4 (rw,sync,relatime)
/dev/mmcblk0p7 on /usr/data/.mod/.zmod/usr/data/gcodes type ext4 (rw,sync,relatime)

mortonlame, 2025-03-14:
[zmod_install (4).log](https://github.com/user-attachments/files/19247512/zmod_install.4.log)
в usr/data лежит вот такой лог

ghzserg, 2025-03-14:
Инсталлятор обновлен релиз перезалит
https://github.com/ghzserg/zmod/releases/tag/1.4.1

closed 2025-03-14 by ghzserg