ghzserg/zmod #28: Проблема

llcux43, 2025-03-14:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)**
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ x] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [X] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] MESH=PETG FORCE_KAMP=False SKIP_LEVELING=True
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
19:55:56 
$ GET_ZMOD_DATA
19:55:56 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
19:55:56 
info Вы работаете с родным экраном // DISPLAY_ON
19:55:56 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
19:55:56 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
19:55:56 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
19:55:57 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
19:55:57 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
19:55:57 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
19:55:57 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
19:55:57 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
19:55:57 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
19:55:57 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
19:55:57 
// Алгоритм очистки: _CLEAR4 // SAVE_ZMOD_DATA CLEAR=_CLEAR4
19:55:57 
info Параметры используемые при окончании и отмене печати [END_PRINT]
19:55:57 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
19:55:57 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
19:55:57 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
19:55:57 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
19:55:57 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
19:55:57 
info Общесистемные параметры
19:55:57 
// Сохранение управляется родным экраном. // SAVE_ZMOD_DATA SAVE_RESTORE=1
19:55:57 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
19:55:57 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
19:55:57 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
19:55:57 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
19:56:05 
// Яркость LED при включении: 80 // SAVE_ZMOD_DATA LED=80
19:56:05 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
19:56:05 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
19:56:05 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
19:56:05 
// Разрешить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=1
19:56:05 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
19:56:05 
// Используeтся родной Klipper
19:56:05 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 40
19:56:05 
// Контроль веса не настроен. // Для настройки: NOZZLE_CONTROL WEIGHT=1500
19:56:05 
// zversion:
19:56:05 
// Установлено с флешки: 1.4.1. Обновление с Fluidd/Mainsaill: 1.4.1
```

Подробное описание ошибки:
Запустил печать на 24 часа, через неопределенное выходит ошибка mcu, пробовал и с экраном и с гуппи. результат один, продолжить печать дает после перезагрузки, но деталь в уру т.к есть видимые смещения. Проц 256 свап был выключен, с включенным свапом тоже самое, хотя напечатало чуть больше деталь. Сам файл печати большой около 200мб

llcux43, 2025-03-14:
https://t.me/FF_5M_5M_Pro/408369/631728

llcux43, 2025-03-14:
запустил печать с приоритетом 20, e0017 вроде у меня стоит 1 , посмотрим что получится. Получил за 3 дня такие ошибки mcu 3 раза, притом модели разные (печатаю корпус и крышку к нему), но высота отпечатанного один в один по слоям  

[image]

ghzserg, 2025-03-15:
> запустил печать с приоритетом 20, e0017 вроде у меня стоит 1 , посмотрим что получится. Получил за 3 дня такие ошибки mcu 3 раза, притом модели разные (печатаю корпус и крышку к нему), но высота отпечатанного один в один по слоям
> 
> [image]

Я ошибся надо включить е0011

llcux43, 2025-03-15:
> > запустил печать с приоритетом 20, e0017 вроде у меня стоит 1 , посмотрим что получится. Получил за 3 дня такие ошибки mcu 3 раза, притом модели разные (печатаю корпус и крышку к нему), но высота отпечатанного один в один по слоям
> > [image]
> 
> Я ошибся надо включить е0011

он тоже был включен, пробовал и с экраном и без экрана и с гуппи всё равно встаёт. печатаю уже не смотря на остановки. Пробовал и со свопом и без,  и приоритет поменял итог тот же. Сейчас выключил e0011 и e0017 и вроде уже полтора часа продолжает печатать. Еще в гуппи неплохо бы название принтера вывести а то при отправке на печать путаюсь в 5 принтаках, в стоке хоть с верху показывает.

ghzserg, 2025-03-15:
В тар конфиге е0011 был отключен. 
Прочитайте по ссылке про е0011, там указаны основные причины. 

Учитывая, что проблема на одной и той же высоте, возможна аппаратная проблема.

По гуппи делайте отдельный тикет.
Один тикет, одна задача.

ghzserg, 2025-03-20:
Закрыта ввиду отсутствия активности

closed 2025-03-20 by ghzserg