ghzserg/zmod #29: Проблема

WatWall, 2025-03-14:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)**
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ ] У меня AD5M
- [X] У меня AD5M PRO
- [ ] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Подробное описание ошибки:
I followed all steps in the installation guide and made a clean install by removing klipper mod, installing native firmare then installing the zmod.
I see that it installed the zmod on the screen. But when the installation completed, the printer restarted itself to the stock firmare.

Here's the log file:
https://pastebin.com/JK6hZJVW

ghzserg, 2025-03-14:
judging by the log, the mod is installed.
the mod does not replace the native screen, but delivers a web interface.
go to http://printer_ip/

WatWall, 2025-03-14:
Yes, mod was installed🤦‍♂️

closed 2025-03-14 by WatWall