ghzserg/zmod #31: Сломался гуппискрин.

K-i-ra, 2025-03-18:
- [х] У меня AD5M
- [х] Я работаю с Guppyscreen

Я загрузил тарконфиг и видосы с проблемой на файлообменник.
https://transfiles.ru/fznoi

Я использую стартовый код:
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]

Я написал завершающий код:
END_PRINT

Вот мои параметры SAVE_ZMOD_DATA:
09:21:41 
$ GET_ZMOD_DATA 
09:21:41 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
09:21:41 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
09:21:41 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
09:21:41 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
09:21:41 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
09:21:41 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
09:21:41 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
09:21:41 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
09:21:41 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
09:21:41 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
09:21:41 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
09:21:42 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
09:21:42 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
09:21:42 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
09:21:42 
info Параметры используемые при окончании и отмене печати [END_PRINT]
09:21:42 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
09:21:42 
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
09:21:42 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
09:21:42 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
09:21:42 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
09:21:42 
info Общесистемные параметры
09:21:42 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
09:21:42 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
09:21:42 
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
09:21:43 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
09:21:43 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
09:21:43 
// Яркость LED при включении: 0 // SAVE_ZMOD_DATA LED=0
09:21:43 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
09:21:43 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
09:21:43 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
09:21:43 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
09:21:43 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
09:21:43 
// Выключать родной экран через (сек): 10 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=10
09:21:43 
// Используeтся родной Klipper
09:21:43 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
09:21:43 
// Вес: 1500; Контроль настроен и не активен.
09:21:43 
// При сработке отключается Klipper. // ZCONTROL_ABORT
09:21:43 
// zversion:
09:21:43 
// Установлено с флешки: 1.4.1. Обновление с Fluidd/Mainsaill: 1.4.1

Подробное описание ошибки:
После обновления на 1.4.1-118-g16457ae5 сломался гуппискрин. Нет реакции на большинство кнопок или она очень сильно запаздывает. Проверка системы ошибок не находит, перезагрузка не помогает. Переустановка с флешки 1.4.1 и последующее обновление до 1.4.1-118-g16457ae5 помогает, но ненадолго.

ghzserg, 2025-03-18:
Обновите  до 
1.4.1-125-g4d5fed8e

K-i-ra, 2025-03-18:
Обновился. Существенно ничего не изменилось.

ghzserg, 2025-03-18:
Опишите, что нужно сделать, чтобы получить такое поведение.

K-i-ra, 2025-03-18:
Если б я знал :) Оно само. Принтером пользуюсь с лета. Сначала был клипермод. Пару месяцев назад поставил zmod. Последовательно обновлялся через флюид, затем с флешки на 1.4.1. и дальше через флюид. Пару дней назад обновил до 1.4.1-118 и заметил это. Пропадает после переустановки с флешки, но через минуты начинается снова. А так, принтер печатает, других проблем нет. Вот и вся история.

ghzserg, 2025-03-18:
Проверьте версию 1.4.1-126-g836f527f

K-i-ra, 2025-03-18:
Обновился до 1.4.1-128. Гуппи не успел загрузиться, вылезла 011. Перезагрузился по питанию, всё отлично, проблем не наблюдаю. Гуппи работает, по ощущениям, отзывчивость интерфейса увеличилась. Спасибо.

closed 2025-03-19 by ghzserg