ghzserg/zmod #32: Проблема с проверкой MD5

virtes, 2025-03-18:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)**
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [X] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.
https://transfiles.ru/ilmy3

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] FORCE_KAMP=True
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
```

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
16:26:54 
$ GET_ZMOD_DATA
16:26:54 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
16:26:54 
info Вы работаете с родным экраном // DISPLAY_ON
16:26:54 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
16:26:54 
// При каждой печати строить карту стола средствами родного экрана. // SAVE_ZMOD_DATA PRINT_LEVELING=1
16:26:55 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
16:26:55 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
16:26:55 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
16:26:55 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
16:26:55 
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
16:26:55 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
16:26:55 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
16:26:55 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
16:26:55 
info Параметры используемые при окончании и отмене печати [END_PRINT]
16:26:55 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
16:26:55 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
16:26:55 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
16:26:55 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
16:26:55 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
16:26:55 
info Общесистемные параметры
16:26:55 
// Сохранение управляется родным экраном. // SAVE_ZMOD_DATA SAVE_RESTORE=1
16:26:55 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
16:26:55 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
16:26:55 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
16:26:55 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
16:26:55 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
16:26:55 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
16:26:55 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
16:26:55 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
16:26:55 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
16:26:55 
// Используeтся родной Klipper 11.
16:26:55 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
16:26:56 
// Контроль веса не настроен. // Для настройки: NOZZLE_CONTROL WEIGHT=1500
16:26:56 
// zversion:
16:26:56 
// Установлено с флешки: 1.4. Обновление с Fluidd/Mainsaill: 1.4
```

Подробное описание ошибки:
Максимально подробно опишите, что Вы делали, что Вы планировали  получить и что Вы получили.

При включении проверки MD5 файла, проверка не проходит успешно:
```
В файле '/data/Top Cover Handle.gcode' не совпала MD5 сумма. Файл удален. Печать отменена
```
При этом файл на принтере и файл из орки совпадают:
```
 ~/Downloads  md5sum "Top Cover Handle.gcode"                                                                                                                       ✔  16:13:52
4d9e02195f7158db054d3618212411c4  Top Cover Handle.gcode
 ~/Downloads  md5sum "Top Cover Handle_orca.gcode"                                                                                                                  ✔  16:33:06
4d9e02195f7158db054d3618212411c4  Top Cover Handle_orca.gcode
```

Файл с принтера: https://transfiles.ru/2j59s
Файл из орки: https://transfiles.ru/268k0

Скрипт постобработки в орке:
```
"/Users/virtes/orca/addMD5.sh";
```
addMD5.sh скрипт:
```
 ~/orca  cat addMD5.sh                                                                                                                                              ✔  16:34:42
#!/bin/bash

# Author: Alexander

if [ $# -ne 1 ]; then echo "Usage: ./$0 your_file.gcode"; exit 1; fi
if ! [ -f $1 ]; then echo "File $1 not found"; exit 1; fi

# Calculate the MD5 hash of the file
md5=$(/opt/homebrew/Cellar/md5sha1sum/0.9.5_1/bin/md5sum "$1" | awk '{print $1}')

# Define prefix
prefix="; MD5:"

# Construct the payload
payload="${prefix}${md5}"

# Create a temporary file and append the original file content
tempfile=$(mktemp)
echo "$payload" > "$tempfile"
cat "$1" >> "$tempfile"

# Move the temp file to the original file location
mv "$tempfile" "$1"
```

Орка стоит на macos

ghzserg, 2025-03-18:
У вас не тот файл для проверки

Используйте [этот](https://github.com/ghzserg/zmod_ff5m/blob/1.4/addMD5.sh)

Или скачайте с принтера

virtes, 2025-03-18:
данные скрипты не отличаются с точки зрения генерации md5 хэша и формата записи его в файл. По какой то причине орка отказывается у меня работать с скриптом, в котором есть which, поэтому указывал полный путь к бинарнику.

но поменяв `/opt/homebrew/Cellar/md5sha1sum/0.9.5_1/bin/md5sum` на `/opt/homebrew/bin/md5sum` все заработало. оказалось, они дают разный результат