ghzserg/zmod #33: AD5M . После снятия карты перед печатью не отключается вентилятор обдува модели М106

mortonlame, 2025-03-19:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)**
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ ] У меня AD5M
- [ X] У меня AD5M PRO
- [ ] У меня AD5X

- [] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ X] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код: END_PRINT

[config (2).tar.log](https://github.com/user-attachments/files/19342264/config.2.tar.log)

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.
```

Подробное описание ошибки:
Максимально подробно опишите, что Вы делали, что Вы планировали  получить и что Вы получили.

ghzserg, 2025-03-19:
Обновитесь - исправлено

mortonlame, 2025-03-19:
Работает, да. Теперь, останавливается вентилятор перед первым слоем

closed 2025-03-19 by mortonlame