ghzserg/zmod #34: Изменение работы датчика филамента

nazbav, 2025-03-21:
А там можно как-то таймаут поставить после срабатывания датчика филамента, чтоб на паузу уходил позже и не приходилось по полметра пластика выкидывать?

Как у Elegoo Centauri Carbon (https://youtu.be/cqqSgdjK7B4)

ghzserg, 2025-03-21:
Перешел по ссылке, но там видео на 40 минут, у меня нет столько времени. Укажите тайминг и что именно я должен там увидеть

ghzserg, 2025-03-22:
Вы так и не привели свои настройки.

Поэтому разберу все случаи:
1. Работа с родным экраном - в этом случае, за обработку отвечает экран и вмешаться нет возможности
2. Работа без родного экрана:
2.1 Работа с датчиком движения филамента - в случае засора сопла сработает датчик движения филамента и ждать еще 30 секунд или любое другое время нельзя.
2.2 Работа с датчиком наличия филамента. Да в этом случае можно сделать паузу на 30 секунд, и за это время филамент  дойдет

Обновитесь - поведение с датчиком наличия филамента  без экрана изменено

closed 2025-03-23 by ghzserg