ghzserg/zmod #35: Проблема

kxrorxn, 2025-03-21:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)**
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ x] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [x ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.

https://drive.google.com/file/d/1rK2IhQ2Pg-R57iSLyxXzxa3ZRFmtF54L/view?usp=drive_link

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
START_PRINT
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
14:16:18 
$ SAVE_ZMOD_DATA
14:16:18 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
14:16:18 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
14:16:18 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
14:16:18 
// При каждой печати строить карту стола. // SAVE_ZMOD_DATA PRINT_LEVELING=1
14:16:18 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
14:16:18 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
14:16:18 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
14:16:18 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
14:16:18 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
14:16:18 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
14:16:18 
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
14:16:18 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
14:16:18 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
14:16:18 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
14:16:18 
info Параметры используемые при окончании и отмене печати [END_PRINT]
14:16:18 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
14:16:18 
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
14:16:18 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
14:16:18 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
14:16:18 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
14:16:18 
info Общесистемные параметры
14:16:18 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
14:16:18 
// Используется датчик движения филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=1
14:16:18 
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
14:16:18 
// Не исправлять ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=0
14:16:19 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
14:16:19 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
14:16:19 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
14:16:19 
// Разрешить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=1
```

Подробное описание ошибки:
До этого проблем таких не было, файл на печати, доходит до определенного момента и с ошибкой !! Extruder not configured просто встает на месте. температуры при этом не сбрасываются, но кнопок паузы/остановки печати нету. попробовал отключить питание и восстановить печать, но проблема повторяется ровно так же. перед этим выполнял проверку системы, ошибок нет

ghzserg, 2025-03-21:
1. Не приложен файл, который печатаете
2. В консоли пишет, что кончился филамент

xFlqw, 2025-03-21:
> 1. Не приложен файл, который печатаете
> 
>     2. В консоли пишет, что кончился филамент
файлы
https://drive.google.com/file/d/10vp-pb-DXe_AwRmd7V8RNu1dvwUD4KhJ/view?usp=sharing

филамент есть, я не разобрался как адекватно включить сток датчик филамента. в предыдущих печатях я просто его отрубал с раздела карты стола и запускал печать, все было нормально. на данный момент он включен в стандартном экране и в конфиге save zmod data

ghzserg, 2025-03-21:
Настройка датчика филамента в глобальных параметрах
У вас в коде 
M104 S255 T1
А экструдер у вас 1

xFlqw, 2025-03-21:
разобрался, спасибо. модель была под 2 экструдера, я проглядел и случайно сохранил конфиг

closed 2025-03-21 by xFlqw