ghzserg/zmod #36: Проблема

Imerch, 2025-03-21:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)**
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [X] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ X] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я использую стартовый код:

```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] MESH=auto
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:
16:31:32 
// Установлено с флешки: 1.4.2. Обновление с Fluidd/Mainsaill: 1.4.2
16:31:32 
// zversion:
16:31:32 
// При сработке отключается Klipper. // ZCONTROL_ABORT
16:31:32 
// Вес: 1500; Контроль настроен и не активен.
16:31:32 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
16:31:32 
// Используeтся родной Klipper
16:31:32 
// Выключать родной экран через (сек): 180 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=180
16:31:32 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
16:31:32 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
16:31:32 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
16:31:32 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
16:31:32 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
16:31:32 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
16:31:31 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
16:31:31 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
16:31:31 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
16:31:31 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
16:31:31 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
16:31:31 
info Общесистемные параметры
16:31:31 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
16:31:31 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
16:31:31 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
16:31:31 
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
16:31:31 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
16:31:31 
info Параметры используемые при окончании и отмене печати [END_PRINT]
16:31:31 
// Алгоритм очистки: _CLEAR3 // SAVE_ZMOD_DATA CLEAR=_CLEAR3
16:31:31 
// Использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=1
16:31:31 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
16:31:31 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
16:31:31 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
16:31:31 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
16:31:31 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
16:31:31 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
16:31:31 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
16:31:31 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
16:31:31 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
16:31:31 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
16:31:31 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
16:31:31 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
16:31:31 
$ SAVE_ZMOD_DATA

```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.
16:31:10 
// Установлено с флешки: 1.4.2. Обновление с Fluidd/Mainsaill: 1.4.2
16:31:10 
// zversion:
16:31:10 
// При сработке отключается Klipper. // ZCONTROL_ABORT
16:31:10 
// Вес: 1500; Контроль настроен и не активен.
16:31:09 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
16:31:09 
// Используeтся родной Klipper
16:31:09 
// Выключать родной экран через (сек): 180 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=180
16:31:09 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
16:31:09 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
16:31:09 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
16:31:09 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
16:31:09 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
16:31:09 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
16:31:09 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
16:31:09 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
16:31:09 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
16:31:09 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
16:31:09 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
16:31:09 
info Общесистемные параметры
16:31:09 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
16:31:09 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
16:31:09 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
16:31:09 
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
16:31:09 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
16:31:09 
info Параметры используемые при окончании и отмене печати [END_PRINT]
16:31:09 
// Алгоритм очистки: _CLEAR3 // SAVE_ZMOD_DATA CLEAR=_CLEAR3
16:31:09 
// Использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=1
16:31:09 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
16:31:09 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
16:31:09 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
16:31:08 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
16:31:08 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
16:31:08 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
16:31:08 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
16:31:08 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
16:31:08 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
16:31:08 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
16:31:08 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
16:31:08 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
16:31:08 
$ GET_ZMOD_DATA
```

Подробное описание ошибки:
Через несколько минут работы, после загрузки GuppyScreen экран гаснет и не реагирует на нажатия. Из fluidd принтер без проблем управляется, печатает, меняются параметры и т.д.

ghzserg, 2025-03-21:
Не хватает [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)

Imerch, 2025-03-21:
[config.tar.zip](https://github.com/user-attachments/files/19390309/config.tar.zip)

ghzserg, 2025-03-21:
Нужно сделать следующее
1. Обновить мод
2. Зайти в папку `mod_data` и открыть файл `guppyconfig.json`
3. Найти строчку   ```"display_sleep_sec": 300,``` и заменить ее на ```"display_sleep_sec": 15,```, сохранить файл
4. Вызвать макрос `CLEAR_EMMC`
5. Выключить принтер
6. Включить принтер
7. Дождаться отключения экрана
8. В консоли Fluidd вызвать команду ```RESPOND TYPE=error MSG="Тестовая ошибка"```
9. Посмотреть включился экран или нет
10. Сделать `TAR_CONFIG`

Imerch, 2025-03-21:
В файле guppyconfig.json значение display_sleep_sec": 300 было не 300, а 600.
После 8 пункта экран не включился, после того как все сделал, попробовал его разбудить ткнув в экран, так же без результата.

[config-2.tar.zip](https://github.com/user-attachments/files/19391088/config-2.tar.zip)

ghzserg, 2025-03-21:
Снова обновление и снова повторить все пункты

Imerch, 2025-03-21:
После 8 пункта на экране вышло уведомление - Тестовая ошибка.
Экран начал работать после сна и просыпался от нажатий на экран.
[config.tar.zip](https://github.com/user-attachments/files/19392034/config.tar.zip)

ghzserg, 2025-03-21:
Проблема решена?

Imerch, 2025-03-21:
Да, спасибо за оперативное решение!

closed 2025-03-21 by Imerch