ghzserg/zmod #37: AD5X проблема с быстрым закрытием диалогов и неправильная работа Start print

mortonlame, 2025-03-22:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)** - без него тикет рассматриваться не будет
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ ] У меня AD5M
- [ ] У меня AD5M PRO
- [ X] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:
13:31:06 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
13:31:06 
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
13:31:06 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
13:31:06 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
13:31:06 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
13:31:06 
info Общесистемные параметры
13:31:06 
// Сохранение управляется родным экраном. // SAVE_ZMOD_DATA SAVE_RESTORE=1
13:31:06 
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
13:31:06 
// Не исправлять ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=0
13:31:06 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
13:31:06 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
13:31:06 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
13:31:06 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
13:31:06 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
13:31:06 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.
```

Подробное описание ошибки:
Не работает закрытие диалогов быстрое, при этом, в глобальных флагах выставил 2. При старте работы с START PRINT - 2 раза происходит хоуминг, и нагрев происходит в углу, а не над сбросом, как родной макрос. 

[config (5).tar.log](https://github.com/user-attachments/files/19402006/config.5.tar.log)

ghzserg, 2025-03-22:
Одна проблема  - один тикет.
Обновитесь

mortonlame, 2025-03-22:
все, закрылись диалоги. простите, что создал 2 тикета в одном, это не повторится. теперь, паркуется, где нужно

closed 2025-03-22 by mortonlame