ghzserg/zmod #39: Проблема

Sly-Lynx, 2025-03-22:
- [x] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [x] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG] на файлообменник. https://cloud.mail.ru/public/Y6am/LtC2LWoC8

Я использую стартовый код:

```
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=bed_temperature_initial_layer_single] SKIP_ZOFFSET=False Z_OFFSET=0.025
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
19:54:35 
$ GET_ZMOD_DATA
19:54:35 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
19:54:35 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
19:54:35 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
19:54:35 
// При каждой печати строить карту стола. // SAVE_ZMOD_DATA PRINT_LEVELING=1
19:54:35 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
19:54:36 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
19:54:36 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
19:54:36 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
19:54:36 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
19:54:36 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
19:54:36 
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
19:54:36 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
19:54:36 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
19:54:36 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
19:54:36 
info Параметры используемые при окончании и отмене печати [END_PRINT]
19:54:36 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
19:54:36 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
19:54:36 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
19:54:36 
// Автоматически перезапускать принтер через 1.5 минуты после печати. //SAVE_ZMOD_DATA AUTO_REBOOT=1
19:54:37 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
19:54:37 
info Общесистемные параметры
19:54:37 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
19:54:37 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
19:54:37 
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
19:54:37 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
19:54:37 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
19:54:37 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
19:54:37 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
19:54:38 
// Использовать альтернативный NEW_SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=1
19:54:38 
// SWAP не используется // SAVE_ZMOD_DATA USE_SWAP=0
19:54:38 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
19:54:38 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
19:54:38 
// Выключать родной экран через (сек): 30 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=30
19:54:38 
// Используeтся родной Klipper
19:54:38 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
19:54:38 
// Вес: 1500; Контроль настроен и не активен.
19:54:38 
// При сработке отключается Klipper. // ZCONTROL_ABORT
19:54:38 
// zversion:
19:54:38 
// Установлено с флешки: 1.4.1. Обновление с Fluidd/Mainsaill: 1.4.2
```

Пытался настроить Telegram бота на внешнем хосте. Прошел все этапы расписанные в гайде. Docker запущен, контейнер с ботом тоже. Но принтер не подключается по ssh к хосту. Для проверки добавил в файл authorized_keys ssh-ed25519 ключ сгенерированный на компьютере и компьютер спокойно подключается по ключу.

[image]
[image]
[image]

ghzserg, 2025-03-22:
dbclient: Connection to server@192.168.0.120:22 exited: 

ssh-ed25519 host key mismatch for 192.168.0.120 !
Fingerprint is SHA256:WMxMwWx/GCpa72ColNhQHRRQLe+KjK1MybauGFX7AcI
Expected SHA256:PGYogFse15Zd/k+

Sly-Lynx, 2025-03-22:
Не подскажете в чем может быть проблема?

ghzserg, 2025-03-22:
Зайдите по SSH на принтер и сделайте 
```
rm -f .ssh/known_hosts
```

Sly-Lynx, 2025-03-22:
Удаление файла known_hosts помогло, спасибо.

closed 2025-03-22 by Sly-Lynx