ghzserg/zmod #40: Охлаждение сопла после очистки

s-grechkin, 2025-03-23:
Добрый день. В файле `/opt/config/mod/ff5.cfg` в макросе `CLEAR_NOZZLE`  (361, 362 строки) прописано охлаждение сопла до 120 градусов. Это замедляет старт печати на несколько минут (ждём пока сопло охладится, потом ждём пока нагреется обратно).

Предлагаю убрать это охлаждение. Ну либо объясните, пожалуйста, почему оно необходимо.
Как альтернативное решение можно сделать это конфигурируемым.

ghzserg, 2025-03-23:
Это реализовано по двум причинам.
1. Так сделано в родной прошивке
2. Если снимать не охлажденным и не запечатанным соплом, то на столе останутся точки от вытекшего филамента.

s-grechkin, 2025-03-23:
https://github.com/ghzserg/zmod_ff5m/pull/1 я набросал pull request на возможность управления этим.
Если подскажешь, где написать предупреждение о последствиях, я добавлю.

Насчёт точек, то построение карты стола через макрос AUTO_FULL_BED_LEVEL вроде бы не остужает, или я ошибаюсь?

s-grechkin, 2025-03-23:
надеюсь, не против, что на "ты"? рабочая привычка

ghzserg, 2025-03-23:
Добавлять новый параметр в глобальные флаги я не буду.
Там и так достаточно параметров.
А таких мест, которые кто-то бы хотел подкрутить для себя достаточно.
Для таких изменений есть `mod_data/user.cfg`
Переопределите в нем 
```
[gcode_macro CLEAR_NOZZLE]
```

ghzserg, 2025-03-23:
> Насчёт точек, то построение карты стола через макрос AUTO_FULL_BED_LEVEL вроде бы не остужает, или я ошибаюсь?
Остужает

mortonlame, 2025-03-23:
Параметров и так много в глобальных флагах. И да, в родной прошивке китайцы здраво рассужили, что для очистки сопло нагревается до рабочей температуры, а при снятии стола - остывает до тех пор, пока не перестанут лезть пластиковые сопли. Меняйте в user cfg, если смльно нужно. 99 процентов пользователей этр устраивает. Станочники вообще не делают камп для ускорения процесса. А снимают карты заранее и используют их. 
Экономия времени в 20-30 секунд не ускорит печать.

s-grechkin, 2025-03-23:
3 минуты у меня получилось, а не 20-30 секунд.

ответ принял, спасибо за уделённое время

closed 2025-03-23 by s-grechkin