ghzserg/zmod #41: Проблема

koltinov, 2025-03-25:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)** - без него тикет рассматриваться не будет
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ ] У меня AD5M
- [ x ] У меня AD5M PRO
- [ ] У меня AD5X

- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ x ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:
08:53:10 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
08:53:10 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
08:53:10 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
08:53:10 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
08:53:10 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
08:53:10 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
08:53:10 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
08:53:10 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
08:53:10 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
08:53:10 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
08:53:10 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
08:53:10 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
08:53:10 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
08:53:10 
// Алгоритм очистки: _CLEAR5 // SAVE_ZMOD_DATA CLEAR=_CLEAR5
08:53:10 
info Параметры используемые при окончании и отмене печати [END_PRINT]
08:53:10 
// Играть MIDI в конце печати: KinoQuietNight.mid // SAVE_ZMOD_DATA MIDI_END=KinoQuietNight.mid
08:53:10 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
08:53:10 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
08:53:10 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
08:53:10 
// Выключать FF5M Pro через (мин): 30 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=30
08:53:10 
info Общесистемные параметры
08:53:10 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
08:53:10 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
08:53:10 
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
08:53:10 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
08:53:10 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
08:53:10 
// Яркость LED при включении: 100 // SAVE_ZMOD_DATA LED=100
08:53:10 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
08:53:11 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
08:53:11 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
08:53:11 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
08:53:11 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
08:53:11 
// Выключать родной экран через (сек): 10 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=10
08:53:11 
// Используeтся родной Klipper
08:53:11 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
08:53:11 
// Вес: 800; Контроль настроен и не активен.
08:53:11 
// При сработке отключается Klipper. // ZCONTROL_ABORT
08:53:11 
// zversion:
08:53:12 
// Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3
08:53:12 
info Родной экран отключен

```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.
```

Подробное описание ошибки:
загрузка филамента завершается ошибкой
!! Error evaluating 'gcode_macro _DISABLE_SENSOR:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'filament_switch_sensor'

перезагрузка хоста через fluidd в правом верхнем углу 

[config.zip](https://github.com/user-attachments/files/19441513/config.zip)

приводит к ошибке 
!! Can not update MCU 'eboard' config as it is shutdown
перезагрузка через $ FIRMWARE_RESTART проходит нормально

ghzserg, 2025-03-25:
Одна проблема один тикет,
https://github.com/ghzserg/zmod/wiki/FAQ#can-not-update-mcu

ghzserg, 2025-03-25:
Поставьте обновление 1.4.3-18-g2341afb7

koltinov, 2025-03-25:
проблема ушла

closed 2025-03-25 by koltinov