ghzserg/zmod #42: некоректное отображение z-offset при нулевом значении

koltinov, 2025-03-25:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)** - без него тикет рассматриваться не будет
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ ] У меня AD5M
- [ х ] У меня AD5M PRO
- [ ] У меня AD5X

- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ х ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:
13:44:47 
$ GET_ZMOD_DATA
13:44:47 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
13:44:47 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
13:44:47 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
13:44:47 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
13:44:47 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
13:44:47 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
13:44:47 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
13:44:47 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
13:44:47 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
13:44:47 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
13:44:47 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
13:44:47 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
13:44:47 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
13:44:47 
// Алгоритм очистки: _CLEAR5 // SAVE_ZMOD_DATA CLEAR=_CLEAR5
13:44:47 
info Параметры используемые при окончании и отмене печати [END_PRINT]
13:44:47 
// Играть MIDI в конце печати: KinoQuietNight.mid // SAVE_ZMOD_DATA MIDI_END=KinoQuietNight.mid
13:44:47 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
13:44:47 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
13:44:47 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
13:44:47 
// Выключать FF5M Pro через (мин): 30 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=30
13:44:47 
info Общесистемные параметры
13:44:47 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
13:44:47 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
13:44:47 
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
13:44:47 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
13:44:47 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
13:44:48 
// Яркость LED при включении: 100 // SAVE_ZMOD_DATA LED=100
13:44:48 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
13:44:48 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
13:44:48 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
13:44:48 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
13:44:48 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
13:44:48 
// Выключать родной экран через (сек): 10 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=10
13:44:48 
// Используeтся родной Klipper
13:44:48 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
13:44:48 
// Вес: 800; Контроль настроен и не активен.
13:44:48 
// При сработке отключается Klipper. // ZCONTROL_ABORT
13:44:48 
// zversion:
13:44:49 
// Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3
```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.
```

Подробное описание ошибки:
Максимально подробно опишите, что Вы делали, что Вы планировали  получить и что Вы получили.

при загрузке z-offset из глобальных параметров при нулевом значении в консоли отображается не корректное значение, а именно:
// Загружен Z-Offset из глобальных параметров: {'z': 1.3877787807814457e-17}
остальные значения, отличные от нуля, отображаются правильно, но с кучей нулей например:
// Загружен Z-Offset из глобальных параметров: {'z': 0.005000000000000014}

[config.zip](https://github.com/user-attachments/files/19444663/config.zip)

ghzserg, 2025-03-25:
Это не ошибка, особенности реализации.
Обновитесь до 1.4.3-28-ge9f42654

ghzserg, 2025-03-25:
Если где-то еще не красиво выводятся параметры  - маякните, это не долго поправить.

koltinov, 2025-03-26:
спасибо, стало хорошо
// Загружено смещение из глобальных параметров Z=0.1200
// Загружено смещение из глобальных параметров Z=0.0000

closed 2025-03-26 by koltinov