ghzserg/zmod #43: Проблема

Imerch, 2025-03-25:
- [X] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [X ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я использую стартовый код:

```
START_PRINT EXTRUDER={material_print_temperature_layer_0} BED={material_bed_temperature_layer_0} MESH=auto
M190 S{material_bed_temperature_layer_0}
M104 S{material_print_temperature_layer_0}
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:
19:02:32 
// Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3
19:02:31 
// zversion:
19:02:31 
// При сработке отключается Klipper. // ZCONTROL_ABORT
19:02:31 
// Вес: 1500; Контроль настроен и не активен.
19:02:31 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
19:02:31 
// Используeтся родной Klipper
19:02:31 
// Выключать родной экран через (сек): 180 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=180
19:02:31 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
19:02:31 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
19:02:31 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
19:02:31 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
19:02:31 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
19:02:31 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
19:02:31 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
19:02:31 
// Не исправлять ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=0
19:02:30 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
19:02:30 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=-1
19:02:30 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
19:02:30 
info Общесистемные параметры
19:02:30 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
19:02:30 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
19:02:30 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
19:02:30 
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
19:02:30 
// Играть MIDI в конце печати: Indiana_Jones9.mid // SAVE_ZMOD_DATA MIDI_END=Indiana_Jones9.mid
19:02:30 
info Параметры используемые при окончании и отмене печати [END_PRINT]
19:02:30 
// Алгоритм очистки: _CLEAR3 // SAVE_ZMOD_DATA CLEAR=_CLEAR3
19:02:30 
// Использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=1
19:02:30 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
19:02:30 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
19:02:30 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
19:02:30 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
19:02:30 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
19:02:30 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
19:02:30 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
19:02:30 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
19:02:30 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
19:02:30 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
19:02:30 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
19:02:30 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.
19:02:49 
// Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3
19:02:49 
// zversion:
19:02:49 
// При сработке отключается Klipper. // ZCONTROL_ABORT
19:02:48 
// Вес: 1500; Контроль настроен и не активен.
19:02:48 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
19:02:48 
// Используeтся родной Klipper
19:02:48 
// Выключать родной экран через (сек): 180 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=180
19:02:48 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
19:02:48 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
19:02:48 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
19:02:48 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
19:02:48 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
19:02:48 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
19:02:48 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
19:02:48 
// Не исправлять ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=0
19:02:47 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
19:02:47 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=-1
19:02:47 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
19:02:47 
info Общесистемные параметры
19:02:47 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
19:02:47 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
19:02:47 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
19:02:47 
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
19:02:47 
// Играть MIDI в конце печати: Indiana_Jones9.mid // SAVE_ZMOD_DATA MIDI_END=Indiana_Jones9.mid
19:02:47 
info Параметры используемые при окончании и отмене печати [END_PRINT]
19:02:47 
// Алгоритм очистки: _CLEAR3 // SAVE_ZMOD_DATA CLEAR=_CLEAR3
19:02:47 
// Использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=1
19:02:47 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
19:02:47 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
19:02:47 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
19:02:47 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
19:02:47 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
19:02:47 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
19:02:47 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
19:02:47 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
19:02:47 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
19:02:47 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
19:02:47 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
19:02:47 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
```
[config.tar.zip](https://github.com/user-attachments/files/19450355/config.tar.zip)

Я использую CURA 5.9.1. Установил плагин Flashforge-Finder https://github.com/eskeyaar/Flashforge-Finder- (как я понимаю он конвертирует стандартный gcode в удобоваримый для принтера формат .GX) . Так же установил плагин OCTOPRINT CONNECTION. После передачи через OCTOPRINT в fluidd формат файла становится .gcode. 

После отправки файла на печать, он берет температуру стола 80 градусов (у меня в gcode установлена 50 градусов). Проходит CLEAR_3, принтер останавливается и сбрасывает температуру до 50 градусов. После помогает только отмена печати.

[Test cura.gcode.zip](https://github.com/user-attachments/files/19450478/Test.cura.gcode.zip)

Imerch, 2025-03-25:
Изменил температуру стола на 80 градусов, увидел что и сопло было на температуре 245 градусов, вместо 270. После предочистики принтер остановился и набрал нужную температуру сопла и начал печатать. Но печатать начал с очень странным звуком, я подумал что царапает стол, но стол целый. Через пару секунд печать остановилась с ошибкой, не успел увидеть что за ошибка. Файл config tag прикладываю.

[config-2.tar.zip](https://github.com/user-attachments/files/19450921/config-2.tar.zip)
И по ощущениям была сильная переэкструзия на первом слое.

ghzserg, 2025-03-25:
У вас код указан как Marlin -скорее всего надо в Cura указать Klipper

Imerch, 2025-03-25:
Нет варианта Klipper 

[image]

ghzserg, 2025-03-25:
Файл mod_data/midi/Indiana_Jones9.mid не найден
Правильно файл называется IndianaJones.mid пропишите его 
SAVE_ZMOD_DATA MIDI_END=IndianaJones.mid

Imerch, 2025-03-25:
Да, это я умышлено исказил, т.к. не получилось вообще отключить звук.

ghzserg, 2025-03-25:
И обновите  мод
https://github.com/ghzserg/zmod/wiki/Macros#midi_end

Imerch, 2025-03-26:
Стартовый код
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] MESH=auto
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]

После запуска печати, температура доходит до нужной, после зависает печать на месте, не стартовав. Температура стола отключается, температура сопла остается на месте.

[config.tar.zip](https://github.com/user-attachments/files/19462972/config.tar.zip)

ghzserg, 2025-03-26:
В общем взял стартовый код из [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ#%D0%BD%D0%B0%D0%B4%D0%BE-%D0%BB%D0%B8-%D0%BC%D0%B5%D0%BD%D1%8F%D1%82%D1%8C-%D1%87%D1%82%D0%BE-%D1%82%D0%BE-%D0%B2-%D1%81%D1%82%D0%B0%D1%80%D1%82%D0%BE%D0%B2%D0%BE%D0%BC-%D0%BA%D0%BE%D0%B4%D0%B5)
Заменил на параметры из CURA
```
START_PRINT EXTRUDER_TEMP={material_print_temperature_layer_0} BED_TEMP={material_bed_temperature_layer_0} MESH=auto
M190 S{material_bed_temperature_layer_0}
M104 S{material_print_temperature_layer_0}
```
Конечный код 
```
END_PRINT
```

[image]

[image]

[image]
Больше  ничего не делал. 
Все отпечталась сразу

ghzserg, 2025-03-26:
И обновите клиента до 1.4.3-29-gdc35cb04

Imerch, 2025-03-26:
Поставил стартовый код, как Вы и написали
START_PRINT EXTRUDER_TEMP={material_print_temperature_layer_0} BED_TEMP={material_bed_temperature_layer_0} MESH=auto
M190 S{material_bed_temperature_layer_0}
M104 S{material_print_temperature_layer_0}
Удалил из CURA все платины, кроме основных и октопринта и принтер начал печатать!

closed 2025-03-26 by Imerch

Y13Lab, 2025-09-14:
Imerch, как ты настроил OCTOPRINT CONNECTION в Cura ?

ghzserg, 2025-09-14:
https://github.com/ghzserg/zmod/wiki/Recomendations_en#send-files-via-octoklipper-for-printing

Y13Lab, 2025-09-17:
Да, я это читал.
Слайсим в Cura, настроил отправку из Cura на принтер с помощью плагина MoonrakerConnection.
Но в нём нет панели управления как в Cura-OctoPrintPlugin.

[image]

Вроде в Клиппере есть минимальная поддержка OctoPrint

[octoprint_compat]
Enables partial support of OctoPrint API is implemented with the purpose of allowing uploading of sliced prints to a moonraker instance. Currently PrusaSlicer derivatives and Cura with Cura-OctoPrint.

Посмотрю, может где что подностроить надо.