ghzserg/zmod #44: Проблема с проставлением статуса "complete" по завершении печати в интеграции Moonraker в HomeAssistant

vvkorolev93, 2025-03-26:
- [ X ] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [ ] Я работаю с родным экраном
- [ X ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
[https://disk.yandex.ru/d/D9zbEws3IpunQA](url)

Я использую стартовый код:

```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
Принтер_1 (прошивка 1.3.0 проставление статуса complete работает)
$ GET_ZMOD_DATA
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
info Вы работаете БЕЗ экрана
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
info Параметры используемые при окончании и отмене печати [END_PRINT]
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
info Общесистемные параметры
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
// Не исправлять ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=0
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0 
// SWAP используется EMMC // SAVE_ZMOD_DATA USE_SWAP=1
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
// Используeтся родной Klipper 11.
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
// zversion:

Принтер_2 (прошивка 1.4.3 проставление статуса complete НЕ работает)
$ GET_ZMOD_DATA
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
info Параметры используемые при окончании и отмене печати [END_PRINT]
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
info Общесистемные параметры
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
// Не исправлять ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=0
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
// Выключать родной экран через (сек): 180 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=180
// Используeтся родной Klipper
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
// Вес: 800; Контроль настроен и активен.
// При сработке отключается Klipper. // ZCONTROL_ABORT
// zversion:
// Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3
```

Подробное описание ошибки:
Доброго дня! Ситуация такая:
На столе располагаются два принтера (Ad5m), на первом (Принтер_1) стоит zMod версии 1.3.0, на втором (Принтер_2) - 1.4.3.
В локальной сети есть сервер, на котором крутится система умного дома Home Assistant. Внутри HA установлена интеграция Moonraker, позволяющая отслеживать состояния принтеров. HA используется в реализации дополнительной логики, в том числе для включения и отключения принтеров. В интеграции Moonraker есть свойство "%Printer% Current Print State", которое отображает текущее состояние принтера. Поддерживает варианты : standby, printing, paused, complete, cancelled, error.

Когда оба принтера были на прошивке 1.3.0 - после успешного окончания печати, указанное выше свойство проставлялось значением "complete" и дальнейшие автоматизации в HA отрабатывали. После обновления принтера (Принтер_2) на версию 1.4.3 - свойство по успешному окончанию печати начало проставляться в "standby". 

Принтер_2 обновлял с флешки, далее из флюид через WEB. Проверку ошибок мода запускал. Данные манипуляции не помогли и кроме "standby" и "printing" никакие значения больше не видел.
На файлообменнике вместе с TarConfig'ами выложил скриншоты системы, чтобы было более наглядна сложившаяся ситуация.

Значение у указанного свойства "complete" было крайне удобно, так как оно означает однозначность успешного завершения печати.
Спасибо!

ghzserg, 2025-03-26:
У вас некорректно снят лог для принтера 1.3.0

ghzserg, 2025-03-26:
Проверьте обновление 1.4.3-31

vvkorolev93, 2025-03-26:
На версии 1.4.3-31 все отработало как нужно!
Спасибо Вам огромнейшее!