ghzserg/zmod #45: Проблема

koltinov, 2025-03-27:
- [ ] У меня AD5M
- [ x ] У меня AD5M PRO
- [ ] У меня AD5X

- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ x ] Я работаю с Guppyscreen

Вот мои параметры SAVE_ZMOD_DATA:

```
08:14:25 
$ GET_ZMOD_DATA
08:14:25 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
08:14:25 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
08:14:25 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
08:14:25 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
08:14:25 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
08:14:25 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
08:14:25 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
08:14:25 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
08:14:25 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
08:14:25 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
08:14:25 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
08:14:25 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
08:14:25 
// Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
08:14:25 
// Алгоритм очистки: _CLEAR5 // SAVE_ZMOD_DATA CLEAR=_CLEAR5
08:14:25 
info Параметры используемые при окончании и отмене печати [END_PRINT]
08:14:25 
// Играть MIDI в конце печати: KinoQuietNight.mid // SAVE_ZMOD_DATA MIDI_END=KinoQuietNight.mid
08:14:25 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро). // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
08:14:25 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
08:14:25 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
08:14:25 
// Выключать FF5M Pro через (мин): 30 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=30
08:14:25 
info Общесистемные параметры
08:14:25 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
08:14:25 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
08:14:25 
// Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
08:14:25 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
08:14:25 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
08:14:25 
// Яркость LED при включении: 100 // SAVE_ZMOD_DATA LED=100
08:14:25 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
08:14:25 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
08:14:25 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
08:14:25 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
08:14:25 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
08:14:25 
// Выключать родной экран через (сек): 10 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=10
08:14:25 
// Используeтся родной Klipper
08:14:25 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
08:14:26 
// Вес: 800; Контроль настроен и не активен.
08:14:26 
// При сработке отключается Klipper. // ZCONTROL_ABORT
08:14:26 
// zversion:
08:14:26 
// Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3
```

Подробное описание ошибки:
в каком-то из последних обновлений физическая кнопка на передней панели перестала вызывать диалог выключения на экране гуппи, во флюид выскакивает диалог выкл\отмена, а на гуппи никакой реакции.

[config.zip](https://github.com/user-attachments/files/19478809/config.zip)

ghzserg, 2025-03-27:
Описывайте  проблему в заголовке. Описание проблема не информативно
Обновитесь до 1.4.3-32-gdd6d94c5

koltinov, 2025-03-27:
> Описывайте проблему в заголовке. Описание проблема не информативно
принял

koltinov, 2025-03-27:
> Обновитесь до 1.4.3-32-gdd6d94c5

работает, спасибо

closed 2025-03-27 by koltinov