ghzserg/zmod #47: Проблема

Imerch, 2025-03-28:
- [ Х ] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [X] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

[config.tar.zip](https://github.com/user-attachments/files/19508035/config.tar.zip)

При запуске печати выходит ошибка !! Error evaluating 'gcode_macro _MESH_COMPARE:gcode': KeyError: "% if lang == 'en' %" 
И так 4 раза
После последнего обновления и запуска печати, выходят вот такие ошибки. До этого было все нормально. Печать не начинается, держит температуру стола и температуру сопла.

Я использую стартовый код:

```
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] MESH=auto
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
07:14:09 PM 
// Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3
07:14:08 PM 
// zversion:
07:14:08 PM 
// При сработке отключается Klipper. // ZCONTROL_ABORT
07:14:08 PM 
// Вес: 1500; Контроль настроен и активен.
07:14:08 PM 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
07:14:08 PM 
// Используeтся родной Klipper
07:14:08 PM 
// Выключать родной экран через (сек): 180 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=180
07:14:08 PM 
// Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
07:14:08 PM 
// Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
07:14:08 PM 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
07:14:07 PM 
// Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
07:14:07 PM 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
07:14:07 PM 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
07:14:07 PM 
// Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
07:14:07 PM 
// Исправить ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=1
07:14:07 PM 
// Использовать SCV = 5 при построении графиков шейперов.. // SAVE_ZMOD_DATA FIX_SCV=0
07:14:07 PM 
// Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
07:14:07 PM 
// Сохранять данные для восстановлеия печати. // SAVE_ZMOD_DATA SAVE_RESTORE=1
07:14:07 PM 
info Общесистемные параметры
07:14:07 PM 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
07:14:07 PM 
// Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
07:14:07 PM 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
07:14:07 PM 
// Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
07:14:07 PM 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
07:14:07 PM 
info Параметры используемые при окончании и отмене печати [END_PRINT]
07:14:07 PM 
// Алгоритм очистки: _CLEAR3 // SAVE_ZMOD_DATA CLEAR=_CLEAR3
07:14:07 PM 
// Использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=1
07:14:06 PM 
// Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
07:14:06 PM 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
07:14:06 PM 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
07:14:06 PM 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
07:14:06 PM 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
07:14:06 PM 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
07:14:06 PM 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
07:14:06 PM 
// Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
07:14:06 PM 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
07:14:06 PM 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
07:14:06 PM 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
07:14:06 PM 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
```

Imerch, 2025-03-28:
Обновился до последней версии, 1.4.3-56-ge53679da, ошибка все еще актуальна.

[config.tar.zip](https://github.com/user-attachments/files/19508440/config.tar.zip)

ghzserg, 2025-03-28:
Проверьте 1.4.3-61-g3be75496

Imerch, 2025-03-28:
Да, ошибки больше нет, спасибо!

closed 2025-03-28 by Imerch