ghzserg/zmod #48: ошибка при замене филамента

koltinov, 2025-03-29:
- [ ] У меня AD5M
- [ х ] У меня AD5M PRO
- [ ] У меня AD5X

- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ х ] Я работаю с Guppyscreen

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Вот мои параметры SAVE_ZMOD_DATA:

```
13:18:09 
$ GET_ZMOD_DATA
13:18:09 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
13:18:09 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
13:18:09 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
13:18:09 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=.format(level=zprint_leveling)
13:18:09 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=.format(kamp=zuse_kamp)
13:18:09 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=.format(test=zmesh_test)
13:18:09 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=.format(midi=zmidi_start)
13:18:09 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=.format(midi=zmidi_pause)
13:18:09 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=.format(md5=zforce_md5)
13:18:09 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=.format(skew=zdisable_skew)
13:18:09 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=.format(zoff=zload_zoffset)
13:18:09 
// Включить очистку сопла линией // SAVE_ZMOD_DATA DISABLE_PRIMING=.format(prime=zdisable_priming)
13:18:09 
// Не использовать предочистку сопла // SAVE_ZMOD_DATA PRECLEAR=.format(pre=zpreclear)
13:18:09 
// Алгоритм очистки:  // SAVE_ZMOD_DATA CLEAR=.format(clear=zclear)
13:18:09 
info Параметры используемые при окончании и отмене печати [END_PRINT]
13:18:09 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=.format(midi=zmidi_end)
13:18:09 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро) // SAVE_ZMOD_DATA CLOSE_DIALOGS=.format(dialogs=zclose_dialogs)
13:18:09 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд // SAVE_ZMOD_DATA STOP_MOTOR=.format(motor=zstop_motor)
13:18:09 
// Не перезапускать принтер автоматически // SAVE_ZMOD_DATA AUTO_REBOOT=.format(reboot=zauto_reboot)
13:18:09 
// Выключать FF5M Pro через (мин):  // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=.format(timeout=pro_poweroff_timeout)
13:18:09 
info Общесистемные параметры
13:18:09 
// Сохранять данные для восстановления печати // SAVE_ZMOD_DATA SAVE_RESTORE=.format(save=save_restore)
13:18:09 
// Используется датчик наличия филамента // SAVE_ZMOD_DATA MOTION_SENSOR=.format(sensor=motion_sensor)
13:18:09 
// Использовать SCV из конфигурации при построении графиков шейперов // SAVE_ZMOD_DATA FIX_SCV=.format(scv=fix_scv)
13:18:09 
// Исправить ошибку E0011 // SAVE_ZMOD_DATA FIX_E0011=.format(fix=fix_e0011)
13:18:10 
// Исправить ошибку E0017 // SAVE_ZMOD_DATA FIX_E0017=.format(fix=fix_e0017)
13:18:10 
// Яркость LED при включении:  // SAVE_ZMOD_DATA LED=.format(led=zled)
13:18:10 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=.format(midi=zmidi_on)
13:18:10 
// Использовать стандартный SAVE_CONFIG // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=.format(config=znew_save_config)
13:18:10 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=.format(swap=zuse_swap)
13:18:10 
// Запретить китайские облака // SAVE_ZMOD_DATA CHINA_CLOUD=.format(cloud=zchina_cloud)
13:18:10 
// Использовать настройки кнопок Fluidd/Mainsail из ZMOD // SAVE_ZMOD_DATA SAVE_MOONRAKER=.format(mr=save_moonraker)
13:18:10 
// Выключать родной экран через (сек):  // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=.format(timeout=display_off_timeout)
13:18:10 
// Используется родной Klipper
13:18:10 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 
13:18:11 
// Вес: 800; Контроль настроен и не активен.
13:18:11 
// При сработке отключается Klipper. // ZCONTROL_ABORT
13:18:11 
// zversion:
13:18:11 
// Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3
```

Подробное описание ошибки:
при загрузке или выгрузке филамента появляется ошибка: 
!! RESPOND TYPE 'info' is invalid. Must be one of 'echo', 'command', or 'error'

[config.zip](https://github.com/user-attachments/files/19516323/config.zip)

ghzserg, 2025-03-29:
Обновитесь до 1.4.3-80-g2951648b

ghzserg, 2025-03-29:
Обновитесь до 1.4.3-86-g09cd4a47

koltinov, 2025-03-29:
ошибка всё равно осталась. 

я понял как она появляется: я нажимаю "загрузить филамент", пока греется сопло откусываю старый, вытягиваю из трубки его, вставляю новый, происходит загрузка, выскакивает ошибка.

Позволю себе такую наглость предположить причина в том, что команда на загрузку филамента происходит когда в трубке нет филамента.
перевод ошибки: !! Неверный ТИП ОТВЕТА "info". Должен быть один из "echo", "command" или "error"
RESPOND TYPE=info нашел  только в одном месте.
 ошибка кажется в строке:
RESPOND TYPE=info MSG={% if lang == "en" %}"Filament runout. Pausing in 30 seconds"{% else %}"Кончился филамент. Через 30 секунд будет пауза"{% endif %}
файл: /opt/config/mod/switch_sensor_display_off.cfg

поменял info на echo - ошибка вроде не выскакивает, только не знаю правильно ли сделал?

max-hasky, 2025-03-29:
Огромное вам спасибо. Ошибка действительно именно там. 
И она там с 25 марта.
Я поправил ее, если найдете  еще ошибки - сообщите

koltinov, 2025-03-29:
да, конечно, спасибо!

closed 2025-03-29 by koltinov