ghzserg/zmod #50: Проблема после обновления

vlad-yudin96, 2025-03-29:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)** - без него тикет рассматриваться не будет
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [х ] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
info Вы работаете с родным экраном // DISPLAY_ON
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
// При каждой печати строить карту стола средствами родного экрана. // SAVE_ZMOD_DATA PRINT_LEVELING=1
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
// Включить очистку сопла линией // SAVE_ZMOD_DATA DISABLE_PRIMING=0
// Не использовать предочистку сопла // SAVE_ZMOD_DATA PRECLEAR=0
// Алгоритм очистки: _CLEAR1 // SAVE_ZMOD_DATA CLEAR=_CLEAR1

info Параметры используемые при окончании и отмене печати [END_PRINT]
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро) // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
// Автоматически выключать моторы после печати/отмены печати через 25 секунд // SAVE_ZMOD_DATA STOP_MOTOR=1
 // Не перезапускать принтер автоматически // SAVE_ZMOD_DATA AUTO_REBOOT=0
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
info Общесистемные параметры
// Сохранение управляется родным экраном // SAVE_ZMOD_DATA SAVE_RESTORE=1
// Используется датчик наличия филамента // SAVE_ZMOD_DATA MOTION_SENSOR=0
// Использовать SCV из конфигурации при построении графиков шейперов // SAVE_ZMOD_DATA FIX_SCV=1
// Не исправлять ошибку E0011 // SAVE_ZMOD_DATA FIX_E0011=0
// Исправить ошибку E0017 // SAVE_ZMOD_DATA FIX_E0017=1
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
// Использовать стандартный SAVE_CONFIG // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
// Запретить китайские облака // SAVE_ZMOD_DATA CHINA_CLOUD=0
// Использовать настройки кнопок Fluidd/Mainsail из ZMOD // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
// Используется родной Klipper
// Приоритет процесса Klipper (1-минимальный, 40-максимальный):  
// Вес: 1500; Контроль настроен и активен.
// При сработке отключается Klipper. // ZCONTROL_ABORT
// zversion:
// Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3

Подробное описание ошибки:
Максимально подробно опишите, что Вы делали, что Вы планировали  получить и что Вы получили.
После обновления появляется вот такая строчка и печать не начинается:
15:28:05 info Получаю данные для KAMP.
15:28:06 // zprint:
15:28:06 // /opt/config/mod/.shell/zprint.sh: line 33: syntax error near unexpected token `fi'
// /opt/config/mod/.shell/zprint.sh: line 33: `if [ "$ip" == "" ]; thenip=$(ip addr | grep inet | grep eth0 | awk -F" " '{print $2}'| sed -e 's/\/.*$//'); fi'
15:28:06 ok
15:28:06 ok

Если запускать печать с родного экрана то всё запускается. Пробовал на разных Gкодах (в том числе и которые уже печатал). Обновление стоит последнее.

max-hasky, 2025-03-29:
Обновитесь до  1.4.3-90

ghzserg, 2025-03-29:
Обновитесь до  1.4.3-90

closed 2025-03-30 by ghzserg