ghzserg/zmod #51: Проблема

SantaNord, 2025-03-29:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)** - без него тикет рассматриваться не будет
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [X] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

[config-20250329-184434.zip](https://github.com/user-attachments/files/19521892/config-20250329-184434.zip)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

CHECK_MD5
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
G90
M83
G1 Z5 F6000
G1 E-0.2 F800
G1 X110 Y-110 F6000
G1 E2 F800
G1 Y-110 X55 Z0.25 F4800
G1 X-55 E8 F2400
G1 Y-109.6 F2400
G1 X55 E5 F2400
G1 Y-110 X55 Z0.45 F4800
G1 X-55 E8 F2400
G1 Y-109.6 F2400
G1 X55 E5 F2400
G92 E0
Впишите сюда Ваш стартовый G-код принтера
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера

END_PRINT
G1 E-3 F3600
G0 X50 Y50 Z2 F30000
M104 S0 ; turn off temperature

Вот мои параметры SAVE_ZMOD_DATA:

```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.

18:55
Установлено с флешки: 1.4.3. Обновление с Fluidd/Mainsaill: 1.4.3
18:55
zversion:
18:55
При сработке отключается Klipper. // ZCONTROL_ABORT
18:55
Вес: 800; Контроль настроен и не активен.
18:55
Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
18:55
Используeтся родной Klipper
18:55
Использовать настроки кнопок Fluidd/Mainsail из ZMOD. // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
18:55
Запретить китайские облака. // SAVE_ZMOD_DATA CHINA_CLOUD=0
18:55
SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
18:55
Использовать стандартный SAVE_CONFIG. // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
18:55
Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
18:55
Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
18:55
Исправить ошибку E0017. // SAVE_ZMOD_DATA FIX_E0017=1
18:55
Не исправлять ошибку E0011. // SAVE_ZMOD_DATA FIX_E0011=0
18:55
Использовать SCV из конфигурации при построении графиков шейперов. // SAVE_ZMOD_DATA FIX_SCV=1
18:55
Используется датчик наличия филамента. // SAVE_ZMOD_DATA MOTION_SENSOR=0
18:55
Сохранение управляется родным экраном. // SAVE_ZMOD_DATA SAVE_RESTORE=1
18:55
info Общесистемные параметры
18:55
Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
18:55
Не перезапускать принтер автоматически. // SAVE_ZMOD_DATA AUTO_REBOOT=0
18:55
Автоматически выключать моторы после печати/отмены печати через 25 секунд. // SAVE_ZMOD_DATA STOP_MOTOR=1
18:55
Не закрывать диалоги на родном экране. // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
18:55
Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
18:55
info Параметры используемые при окончании и отмене печати [END_PRINT]
18:55
Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
18:55
Не использовать предочистку сопла// SAVE_ZMOD_DATA PRECLEAR=0
18:55
Включить очистку сопла лининей // SAVE_ZMOD_DATA DISABLE_PRIMING=0
18:55
Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
18:55
SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
18:55
Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
18:55
Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
18:55
Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
18:55
Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
18:55
Не использоваить KAMP вместо BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
18:55
Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
18:55
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
18:55
info Вы работаете с родным экраном // DISPLAY_ON
18:55
info Параметры zmod. Задаются через SAVE_ZMOD_DATA

[Testkvadrat v0_PETG_1m47s.zip](https://github.com/user-attachments/files/19521965/Testkvadrat.v0_PETG_1m47s.zip)

Подробное описание ошибки:
Максимально подробно опишите, что Вы делали, что Вы планировали  получить и что Вы получили.
Моторы не отключались после завершения печати. Шум от мотора Z стола. Принудительно выключал посредством M84. Ввел SAVE_ZMOD_DATA STOP_MOTOR=1 .Перезагрузил проверил Моторы отключились не через 25 сек. после окончания печати, а через 11 мин. 
Возможно прописать, чтобы моторы отключались после опускания стола сразу после печати?
Заранее спасибо.

ghzserg, 2025-03-29:
В конечном коде надо оставить только END_PRINT

ghzserg, 2025-03-29:
У вас не последняя версия мода

ghzserg, 2025-03-29:
Не вижу тестового файла

ghzserg, 2025-03-29:
Конечный код, который вы привели в тикете, не соответствует конечному коду из файл
```
M106 S0
M106 P2 S0
;TYPE:Custom
; filament end gcode 
M106 P3 S0
; filament end gcode 
M106 P3 S0
G1 E-3 F3600
G0 X50 Y50 Z2 F30000
M104 S0 ; turn off temperature

M73 P100 R0
```
В Вашем конечном коде  не `END_PRINT`, а значит глобальные параметры `END_PRINT`, такие как `STOP_MOTOR` не работают.

SantaNord, 2025-03-29:
Да Спасибо сделал только END_PRINT к конечном G-коде, не понравилось, что после окончания печати сопло оставляет след на модели. На строчку выше поставил G0  Z3 F3000 перед END_PRINT след пропал, моторы отключаются.
Еще раз Спасибо за обратную связь.

closed 2025-03-29 by ghzserg