ghzserg/zmod #52: Problem

TMTYD, 2025-04-02:
Before asking a question, read:
- **[TAR_CONFIG formatting algorithm](https://github.com/ghzserg/zmod/wiki/FAQ_en#help)** - without it, the ticket will not be considered
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ_en)
- [Recommendations](https://github.com/ghzserg/zmod/wiki/Recomendations_en)
- [Macro documentation](https://github.com/ghzserg/zmod/wiki/Macros_en)

- [ ] I have AD5M
- [ ] I have AD5M PRO
- [ x] I have AD5X

- [X] I work with the native screen
- [ ] I work without a screen
- [ ] I work with Guppyscreen
(replace the space with x inside the brackets)

Detailed error description:

hi

Unfortunately, the Zmod for the AD5X is not installed.

I already wrote something about it here 

[Diskussion](https://github.com/ghzserg/zmod/discussions/46#discussioncomment-12695645)

I checked the hash before installing, it matches.

I then went through the steps as I mentioned in the discussion, but unfortunately without success.

No Zmod Log File is created on the USB stick. 

What I also noticed is that the ad5x factory file is displayed as corrupt if you want to open it with WinRAR.

Please describe in as much detail as possible what you did, what you planned to get, and what you got.

ghzserg, 2025-04-02:
question number 1. did you manage to install factory from flashforge? I posted 2 factors for 5x. one is the one you posted, the second is the one that was given out in ff.

ghzserg, 2025-04-02:
[image]

ghzserg, 2025-04-02:
Use AD5X Firmwaree 1.0.2 only

TMTYD, 2025-04-02:
No, I couldn't install the factory tgz for the AD5X because the file is corrupted when opened with WinRAR.

What factors do you mean?

So, don't use the 1.0.4 firmware?

TMTYD, 2025-04-02:
So, I've tried it twice now, using the new Factor.tgz file and the current mod file, but unfortunately, to no avail.

No log file is created either.

ghzserg, 2025-04-03:
You are downloading the Factory archive incorrectly. You need to click on the archive, go to the [page](https://github.com/ghzserg/zmod/blob/main/%D0%A0%D0%BE%D0%B4%D0%BD%D0%B0%D1%8F_%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0/AD5X-1.0.2-1.0.2-20250120-Factory.tgz). Click 3 dots and download the archive. Or download the archive from the link that you dropped in the discussion. You need to install version 1.0.2.
https://files.catbox.moe/3jusxc.tgz
Rename 3jusxc.tgz to AD5X-1.0.2-1.0.2-20250120-decrypted.tgz

mortonlame, 2025-04-03:
> So, I've tried it twice now, using the new Factor.tgz file and the current mod file, but unfortunately, to no avail.
> 
> No log file is created either.

I've downloaded and installed two times factory and mod just. 
Try another browser.  Instructions for download archives was given to you by zedserg.
You are doing something wrong when you try to download files. Its a reason of errors.

TMTYD, 2025-04-03:
I did it as described here. I downloaded the factory file via the three dots. I also downloaded the mod file for ad5x from the release page.

After that, I followed the instructions step by step, unfortunately, twice without success.

TMTYD, 2025-04-03:
I don't think that's the problem, but point 8 says you should configure the WiFi. It doesn't matter whether it's WiFi or LAN, does it?

[Restoring Printer](https://github.com/ghzserg/zmod/wiki/Setup_en#restoring-printer-to-factory-settings-required-for-mod-installation)

ghzserg, 2025-04-03:
You do not need to proceed to the next steps if you have not completed the previous step.
You need to install the Factory firmware which you need to download from this link https://github.com/ghzserg/zmod/releases/tag/R
Installing the native firmware checks the following things:
1. USB on the printer works
2. Flash is formatted in Fat
3. The specific Flash is supported by the printer
4. You have downloaded the correct firmware update file
5. The printer architecture matches the architecture in the firmware

When installing the native firmware, inscriptions in Chinese should appear, if this is not the case, then somewhere a mistake was made or one of the points listed above was not completed.

ghzserg, 2025-04-03:
Please attach a screenshot showing the native firmware file placed in the root of the flash drive.

ghzserg, 2025-04-03:
> I don't think that's the problem, but point 8 says you should configure the WiFi. It doesn't matter whether it's WiFi or LAN, does it?
> 
> [Restoring Printer](https://github.com/ghzserg/zmod/wiki/Setup_en#restoring-printer-to-factory-settings-required-for-mod-installation)

Fixed ib manual. Thanks

TMTYD, 2025-04-03:
> You do not need to proceed to the next steps if you have not completed the previous step.
> You need to install the Factory firmware which you need to download from this link [...]

Ok I'll sit down again tonight, I think I might know the error, thought the USB stick is automatically formatted in FAT format, it may be that it has been formatted as NTFS. 

look tonight tonight

TMTYD, 2025-04-03:
So the USB stick is formatted correctly.

When I turn on the printer, using the USB stick with the factory file, nothing in Chinese appears.

So the gcode file is in the root directory, so I don't know what the problem is.

I followed all the steps one after the other.

[image]
[image]
[image]
[image]
[image]

ghzserg, 2025-04-03:
Try using another flash drive, a smaller one. I think you need to contact FlashForge support and ask them for instructions on updating the firmware from a flash drive, as well as a flash drive image. You won't be able to run any mod if even the standard firmware won't install.

ghzserg, 2025-04-03:
[flashforge_init.sh.txt](https://github.com/user-attachments/files/19591516/flashforge_init.sh.txt)
Rename file to flashforge_init.sh and move it to USB flash.
I have attached a file that may help. You need to rename it to flashforge_init.sh and place it on the flash drive. After loading the printer, you will be able to connect to the printer via ssh through putty. Login root, password root. A file should also appear on the flash drive. You need to paste it here

TMTYD, 2025-04-03:
I just watched a video on how to update/downgrade FlashForge via USB stick. The video shows how to unpack the tgz file and pack the files it contains onto the USB stick. See for yourself.

https://youtu.be/BWC4E7JAZZk?si=Qik1bWsB0b3hZOeo&t=71

ghzserg, 2025-04-04:
This video is for 5m, in 5x the algorithm has changed. On 5m there is no need to unpack the data, although it is possible.

Have you experimented with flashforge_init.sh?

TMTYD, 2025-04-04:
good morning,

Didn't make it last night, I'll do it later today

TMTYD, 2025-04-04:
Here is the log file

[root.log](https://github.com/user-attachments/files/19606052/root.log)

ghzserg, 2025-04-04:
> Here is the log file
> 
> [root.log](https://github.com/user-attachments/files/19606052/root.log)

Great. Try to login as root. To do this, download the putty program. Enter the printer IP address and port 22. Username root, password root. Try to log in. And type `uname -a`

TMTYD, 2025-04-04:
please, added the .txt extension

```
# uname -a
Linux flashforge 5.10.186+ #126 SMP PREEMPT Mon Oct 28 18:48:28 PDT 2024 mips GNU/Linux

```

[passwd.txt](https://github.com/user-attachments/files/19608849/passwd.txt)
[shadow.txt](https://github.com/user-attachments/files/19608850/shadow.txt)

ghzserg, 2025-04-05:
Excellent. So the architecture matches. The next thing to do. Connect via SSH. Insert the flash drive and run the command 
```
tar -cf /media/usr.tar /usr/
sync
```
И приложить полученный файл 
usr.tar

ghzserg, 2025-04-05:
I also ask you to do the following.
Place the Factory archive on a flash drive.
Run Windows -> Run -> cmd.exe
```
E:
dir
```
Insert screenshot here

TMTYD, 2025-04-05:
Ok I'll try both tonight

TMTYD, 2025-04-05:
As I see, Klipper is already installed on the printer's system, as is Mainsail.

Isn't it possible to simply start Mainsail and open the Mainsail port?

ghzserg, 2025-04-05:
> As I see, Klipper is already installed on the printer's system, as is Mainsail.
> 
> Isn't it possible to simply start Mainsail and open the Mainsail port?

Klipper has always been in 5m, 5mpro, ad5x.
Moonraker is in 5x, but it is compiled without some libraries and does not work with Fluidd.
If you run the native moonraker and mainsail, there will be many errors and the native screen will freeze

TMTYD, 2025-04-05:
Here are the things in question

```
# tar -cf /media/usr.tar /usr/
tar: can't open '/media/usr.tar': Read-only file system
```

[image]

ghzserg, 2025-04-06:
This error can only happen if you haven't inserted the flash drive.
```
mount
tar -cf /media/usr.tar /usr/
```

TMTYD, 2025-04-06:
So now it worked. I've uploaded the folder usr to Google Drive. Below is the link.
[usr](https://drive.google.com/file/d/1V5Q_A5XoUNkBSK4dnuRKg-dYsPI9_0Br/view?usp=drive_link)

[image]

ghzserg, 2025-04-06:
Unzip https://github.com/ghzserg/zmod/releases/download/1.4.3/AD5X-zmod-1.4.3.tgz
Into the root of the flash drive and reboot the printer

ghzserg, 2025-04-06:
You have a completely standard printer. I did not find any differences.
But for some reason it does not find a file like AD5X-*.tgz on the flash drive. But it does find the file flashforge_init.sh.
Therefore, you need to unzip the archive onto the flash drive and the installation should start. After installing the mod, I would recommend inserting the flash drive into the printer again and running the command `ZFLASH`
This command will download the necessary image for the printer to the flash drive, check its checksum and reinstall the mod.
Installation from the unpacked archive is possible, but I do not recommend it, because when copying from the flash drive to the printer memory, the files may break.
After the second installation is complete, I recommend going to Fluidd -> Settings -> Software Update and updating all modules

TMTYD, 2025-04-06:
And why doesn't klipperscreen work? Can't this be done with a Raspberry Pi?

ghzserg, 2025-04-06:
I don't understand your question. 
You have an installation problem. Have you solved it?
There is no klipperscreen in the mod.
If you want to run klipperscreen on Raspberry Pi, then run it there and configure the connection to the printer

TMTYD, 2025-04-06:
I'll answer that later, see screenshot. Can you change the language from Russian to English, perhaps to German, and add a progress bar?

I'd be happy to help with the German translation. How long should the USB stick remain plugged in the first time?

[image]

TMTYD, 2025-04-06:
> I don't understand your question. You have an installation problem. Have you solved it? There is no klipperscreen in the mod. If you want to run klipperscreen on Raspberry Pi, then run it there and configure the connection to the printer

Is it not possible to install Klipperscreen on the ad5x?

Why is it not possible? Can this be achieved with a Raspberry Pi? If so, how would it work?

TMTYD, 2025-04-06:
> I'll answer that later, see screenshot. Can you change the language from Russian to English, perhaps to German, and add a progress bar?
> 
> I'd be happy to help with the German translation. How long should the USB stick remain plugged in the first time?
> 
> [image]

The screen is unchanged

TMTYD, 2025-04-06:
I turned off the printer after about 20 minutes, pulled out the stick and turned it back on, unfortunately no zmod installed

ghzserg, 2025-04-06:
Well, the screen says do not disconnect the flash drive or remove it.
There should be a file zmod_install.log on the flash drive.
Send it.
How did you determine that the mod was not installed? The mod adds a web interface - the native screen does not change.

TMTYD, 2025-04-06:
The printer restarted normally after I turned it back on. And Mainsail isn't accessible via ip:port.

ghzserg, 2025-04-07:
I looked at the log. The installation was going fine, but you interrupted it at the stage of copying files to the printer. You need to repeat the installation and wait until the window changes to another one. It should be my green window - ok. Or a window with a cross and an error.

TMTYD, 2025-04-07:
Ok I'll try again and just wait, I've started it now

TMTYD, 2025-04-07:
So, it's been unchanged for an hour. I'm attaching the installation log.

What I notice is that, like in the previous attempt, it stops here according to the log.

```
+ MOD=/usr/data/.mod/.zmod
+ mkdir -p /usr/data/.mod/.zmod
+ xz -dc /mnt/mod/rootfs.tar.xz
+ tar -xf - -C /usr/data/.mod/.zmod
```

[zmod_install.log](https://github.com/user-attachments/files/19636804/zmod_install.log)

ghzserg, 2025-04-08:
Let's try to do without a flash drive. Perhaps the reason is in it, since the original firmware also does not install.

1. Copy the [AD5X-zmod-1.4.3.tgz](https://github.com/ghzserg/zmod/releases/download/1.4.3/AD5X-zmod-1.4.3.tgz) file to `/usr/data/ ` via winscp.
2. Log in as root via ssh
```
cd /usr/data/
tar -xvf AD5X-zmod-1.4.3.tgz -C /usr/data/update/
/usr/data/update/flashforge_init.sh AD5X 0026
```

TMTYD, 2025-04-08:
So I tried, but the update folder doesn't exist, which is why the extraction didn't work (see screenshot). I then created the update folder using mkdir -p.

Then I ran the script and specified it, and the first message regarding the log file (Read Only File System) appeared.

Then a red x appeared on the printer screen (see screenshot), and the printer restarted, and nothing was installed.

[image]
[image]
[image]

[image]

ghzserg, 2025-04-08:
Add ```mount /sda1 /mnt```
And insert flash into printer

ghzserg, 2025-04-08:
in the new version only the pictures have changed. you need to download via winscp. insert the flash drive mount it and run the installation by

TMTYD, 2025-04-09:
So I downloaded the new zmod file, checked the md5 sum.

Then I transferred it via winscp, created the update folder using mkdir -p, extracted the files, and then executed the command.

Then the installation started up to a certain point, and the printer restarted. After the restart, the network connection was no longer active. See screenshot.

I let the whole thing run for about an hour, unfortunately without success.

There's probably something interesting in the zmod install log.

[image]
[image]

that came after the reboot:

[image]

[zmod_install.log](https://github.com/user-attachments/files/19660717/zmod_install.log)

now it does nothing and stays at the initialization screen

TMTYD, 2025-04-09:
It doesn't boot up anymore, the initialize screen, so there's no network connection

mortonlame, 2025-04-09:
> It doesn't boot up anymore, the initialize screen, so there's no network connection

You don't need a network at this stage. Use the file that the developer sent you. Upload it to a flash drive, preferably another one. You seem to have problems with this one. And run the printer with this flash drive in the USB slot.

TMTYD, 2025-04-09:
@mortonlame 

I've now tried a different USB stick. The one I was using before didn't create a root.log file and just froze on the initialization screen and didn't do anything (no root.log file was created).

With the current USB stick, a Sandisk 128GB, it created a root.log file and only booted up to the Flashforge script.

@ghzserg 

[root.log](https://github.com/user-attachments/files/19671684/root.log)

Here's the root log

ghzserg, 2025-04-09:
The fact that a root file has been created is very good news. It means that it will be quite easy to restore the printer.
1. Use an 8 GB flash drive and format it to FAT.

2. Write the native firmware to it https://github.com/ghzserg/zmod/releases/download/R/AD5X-1.0.2-1.0.2-20250120-Factory.tgz
3. Install the native firmware
4. Install the mod via flash drive

TMTYD, 2025-04-09:
Now this happened, I installed the firmware from the link

[image]
[image]

TMTYD, 2025-04-09:
and this, I haven't installed the zmod yet

[image]
[image]
[image]

TMTYD, 2025-04-09:
The host error keeps coming back, even if I wait 20 seconds before switching on, it still comes back

ghzserg, 2025-04-09:
Install zmod from flash drive.
After install zmod. Run check_system macros

TMTYD, 2025-04-09:
Okay, despite the error, I copied the zmod file to the USB stick. Plugged in the USB stick and turned on the printer. Now it says

[image]

TMTYD, 2025-04-09:
After a short time the printer restarted and the host error occurred again, I did not turn off the printer or remove the USB stick, it rebooted on its own

TMTYD, 2025-04-09:
I restarted the printer again, then skipped the introduction and was immediately in the printer's main menu.

I entered the printer's IP address: port (7125) and then this appeared.

[image]

ghzserg, 2025-04-10:
Excellent. You were able to install the mod. The problem was a bad flash. Now it's just a small matter, to restore the printer.
Go to the configuration section and make an archive of all the files that are there. Post this archive here, I'll see what's missing.

[43 more replies]

closed 2025-04-12 by ghzserg