ghzserg/zmod #61: New firmware from Flashforge

TMTYD, 2025-04-10:
Flashforge has released new versions of the original firmware.

See here:

new firmware out for the ad5x 

[Firmware Download](https://flashforge.com/blogs/download-document/ad5x)

[Firmware Release Notes](https://wiki.flashforge.com/en/ad5x/manual/firmware_release_history)

ghzserg, 2025-04-10:
The mod works on version AD5X-1.0.2-1.0.2-20250120
Version AD5X-1.0.4-1.0.3-20250317 was unsuccessful, there is incorrect nozzle cleaning and the inability to save configuration files.
I did not test version AD5X-1.0.5-1.0.3-20250402, because I do not have this printer, and the person who has it is completely satisfied with version AD5X-1.0.2-1.0.2-20250120.
If you want to test and fix errors for AD5X-1.0.5-1.0.3-20250402, then I will help.
But first you need to start the printer, read the documentation and learn how to work with the mod.

TMTYD, 2025-04-10:
> The mod works on version AD5X-1.0.2-1.0.2-20250120
> Version AD5X-1.0.4-1.0.3-20250317 was unsuccessful, there is incorrect nozzle cleaning and the inability to save configuration files.
> I did not [...]

Okay, all right, I'll do that. If everything works as expected, I'll get back to you.

ghzserg, 2025-04-22:
Firmware 1.0.8 is currently being tested.

TMTYD, 2025-04-23:
And does it work?

TMTYD, 2025-04-24:
> https://github.com/ghzserg/zmod/wiki/Changelog_en#version-151

When I'm done printing, I'll join the thing again

closed 2025-04-25 by ghzserg