ghzserg/zmod #62: The following errors occur repeatedly

TMTYD, 2025-04-11:
Before asking a question, read:
- **[TAR_CONFIG formatting algorithm](https://github.com/ghzserg/zmod/wiki/FAQ_en#help)** - without it, the ticket will not be considered
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ_en)
- [Recommendations](https://github.com/ghzserg/zmod/wiki/Recomendations_en)
- [Macro documentation](https://github.com/ghzserg/zmod/wiki/Macros_en)

- [ ] I have AD5M
- [ ] I have AD5M PRO
- [x ] I have AD5X

- [X] I work with the native screen
- [ ] I work without a screen
- [ ] I work with Guppyscreen
(replace the space with x inside the brackets)

I uploaded [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ_en#help) to the zmod chat (https://t.me/FF_5M_5M_Pro/408369) or to a file sharing service.
Attach a link to the message with the file or attach the file to this text using the button with a paperclip at the bottom.

If the problem occurs on a specific file with gcode, then attach this file too.
If the problem occurs even when printing a standard cube, then attach the file with the gcode of this cube.

Detailed error description:

Temporary failure in name resolution

This error keeps appearing in the moonraker.log file. What is the problem?

[moonraker.log](https://github.com/user-attachments/files/19698840/moonraker.log)

And this in the FirmwareExe.log file

```
2025.04.11 04:14:47.419 <DEBUG> app/appClient.cpp checkNetworkConnection checkNetworkConnection:  false 
2025.04.11 04:14:47.426 <CRITICAL> dialog/waitMovieDialog.cpp WaitMovieDialog 
2025.04.11 04:14:47.541 <CRITICAL> app/AppNetworkManager.cpp get  curl error code :  7 
```

[firmwareExe.log](https://github.com/user-attachments/files/19698853/firmwareExe.log)

ghzserg, 2025-04-11:
This is disabled Chinese clouds
https://github.com/ghzserg/zmod/wiki/Macros_en#china_cloud

TMTYD, 2025-04-11:
> This is disabled Chinese clouds https://github.com/ghzserg/zmod/wiki/Macros_en#china_cloud

ok thx