ghzserg/zmod #66: Version number MCU, software version Moonraker and clipper

TMTYD, 2025-04-12:
The version number of the MCUs in the Mainsail interface, as well as the software versions of Klipper and moonraker are missing

[image]

[image]

ghzserg, 2025-04-12:
That's how it should be. Flashforge Did not provide version numbers. Moonraker from master.

closed 2025-04-12 by ghzserg

TMTYD, 2025-04-12:
Okay, I understand that with the Flashforge version number.

But on my other printer, which also runs Klipper, Moonraker, and mainsail, the version number is displayed, but not on zmod (Mainsail).

[image]

ghzserg, 2025-04-12:
Here we use a Klipper from FlasMoonraker from zmod.
Automatic update of clipper and moonraker is not supported.

Moonraker updates come with zmod.
That is why their versions are not written.
That is how it should behForge

TMTYD, 2025-04-13:
ah ok, that's the reason, thanks