ghzserg/zmod #71: Download Klipper or Moonraker log file

TMTYD, 2025-04-13:
Before asking a question, read:
- **[TAR_CONFIG formatting algorithm](https://github.com/ghzserg/zmod/wiki/FAQ_en#help)** - without it, the ticket will not be considered
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ_en)
- [Recommendations](https://github.com/ghzserg/zmod/wiki/Recomendations_en)
- [Macro documentation](https://github.com/ghzserg/zmod/wiki/Macros_en)

- [ ] I have AD5M
- [ ] I have AD5M PRO
- [x ] I have AD5X

- [X] I work with the native screen
- [ ] I work without a screen
- [ ] I work with Guppyscreen
(replace the space with x inside the brackets)

When I try to download one of the two log files, I always get the following message

`{"error":{"code":404,"message":"Not Found","traceback":"Traceback (most recent call last):\n\n  File \"/root/moonraker-env/lib/python3.12/site-packages/tornado/web.py\", line 1790, in _execute\n    result = await result\n             ^^^^^^^^^^^^\n\n  File \"/root/moonraker-env/moonraker/components/application.py\", line 797, in get\n    self.absolute_path = self.validate_absolute_path(\n                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n\n  File \"/root/moonraker-env/lib/python3.12/site-packages/tornado/web.py\", line 2900, in validate_absolute_path\n    raise HTTPError(404)\n\ntornado.web.HTTPError: HTTP 404: Not Found\n"}}`

ghzserg, 2025-04-13:
Need tar_config

TMTYD, 2025-04-13:
No, I don't, because it's there. As soon as I copy the logs from the menu to the USB stick and unpack them, it's there. 

I can't download a config_tar because I no longer have access to it via PuTTY.

ghzserg, 2025-04-13:
You don't need ssh access to download the file. It is available from Flyuidd/Mainsaill bpo menby config

TMTYD, 2025-04-13:
So everything I had access to Mainsail, I tried to download the klippy.log and moonraker.log via the web interface, both got this error, I didn't delete anything

ghzserg, 2025-04-13:
[image]

[image]

TMTYD, 2025-04-13:
> [image]
> 
> [image]

OK, I think this option wasn't available before, but I'm not sure anymore. But it's no use to me if I no longer have access to Mainsail/Fluidd, nor do I have SSH access, so I can't access the config.tar to post it.

ghzserg, 2025-04-13:
Set up the network via the native screen. The mod does not interact with the network in any way. The network is controlled by the native firmware. Check if you have it set up correctly, if there is a signal, etc.

TMTYD, 2025-04-13:
> Set up the network via the native screen. The mod does not interact with the network in any way. The network is controlled by the native firmware. Check if you have it set up correctly, if there is a signal, etc. This option has always been there since the first versions of the mod.

Yes, the network is set up correctly, a ping from the laptop to the printer's IP was also successful

ghzserg, 2025-04-13:
From Mainsail
[image]
[image]
I have nothing to add except that the mod does not work with the network in any way.
And does not manage SSH.

TMTYD, 2025-04-13:
I closed it because I did a new installation

https://github.com/ghzserg/zmod/discussions/76#discussion-8196564

closed 2025-04-13 by TMTYD