ghzserg/zmod #73: Some errors

TMTYD, 2025-04-13:
- [ ] I have AD5M
- [ ] I have AD5M PRO
- [x] I have AD5X

- [X] I work with the native screen
- [ ] I work without a screen
- [ ] I work with Guppyscreen

```
Call GET_ZMOD_DATA and put the output here.
```

```
[01:00:52.096] info ZMOD parameters. Set via SAVE_ZMOD_DATA
[01:00:52.119] info You are using native screen // DISPLAY_ON
[01:00:52.123] info Parameters used at print start and bed leveling [START_PRINT]
[01:00:52.126] // Don't build bed mesh with native screen // SAVE_ZMOD_DATA PRINT_LEVELING=0
[01:00:52.148] // Don't use KAMP instead of BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
[01:00:52.169] // Test bed mesh before printing // SAVE_ZMOD_DATA MESH_TEST=1
[01:00:52.190] // Play MIDI on print start:  // SAVE_ZMOD_DATA MIDI_START=
[01:00:52.216] // Play MIDI on print pause:  // SAVE_ZMOD_DATA MIDI_PAUSE=
[01:00:52.238] // Check file MD5 checksum // SAVE_ZMOD_DATA FORCE_MD5=1
[01:00:52.260] // SKEW correction disabled // SAVE_ZMOD_DATA DISABLE_SKEW=1
[01:00:52.280] // Don't load Z-OFFSET from global settings // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
[01:00:52.301] // Enable nozzle line purge // SAVE_ZMOD_DATA DISABLE_PRIMING=0
[01:00:52.322] // Purge algorithm: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
[01:00:52.346] info Parameters used at print end and cancellation [END_PRINT]
[01:00:52.349] // Play MIDI at print end:  // SAVE_ZMOD_DATA MIDI_END=
[01:00:52.387] // Don't close dialogs on native screen // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
[01:00:52.391] // Auto disable motors after print/cancel in 25 sec // SAVE_ZMOD_DATA STOP_MOTOR=1
[01:00:52.411] // Don't reboot printer automatically // SAVE_ZMOD_DATA AUTO_REBOOT=0
[01:00:52.432] info System-wide parameters
[01:00:52.436] // Saving controlled by native screen // SAVE_ZMOD_DATA SAVE_RESTORE=1
[01:00:52.459] // Use SCV from config for shaper graphs // SAVE_ZMOD_DATA FIX_SCV=1
[01:00:52.481] // Don't fix E0011 error // SAVE_ZMOD_DATA FIX_E0011=0
[01:00:52.564] // Fix E0017 error // SAVE_ZMOD_DATA FIX_E0017=1
[01:00:52.645] // LED brightness at power on: 50 // SAVE_ZMOD_DATA LED=50
[01:00:52.671] // Play MIDI at power on:  // SAVE_ZMOD_DATA MIDI_ON=
[01:00:52.711] // Block Chinese clouds // SAVE_ZMOD_DATA CHINA_CLOUD=0
[01:00:52.733] // Use Fluidd/Mainsail button settings from ZMOD // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
```

Detailed error description:

I haven't been able to connect via SSH since yesterday morning. I keep getting the message "remote side unexpectedly closed network connection," which means I can't download config.tar via WinSCP.

I also noticed a few errors in the **zmod.log**:

This one:

grep: /opt/config/mod_data/variables.cfg: No such file or directory

ntpd: bad address 'ntp1.vniiftri.ru'

I've attached the zmod.log.

Then in the **guppyscreen_start.log**

mv: cannot stat '/opt/config/mod_data/log/guppyscreen.4.log': No such file or directory

Then in the **printer.log**, the one with ?

Starting Clippy...
Args: ['/usr/prog/klipper/klippy/klippy.py', '/usr/data/config/printer.cfg', '-l', '/usr/data/logs/printer.log', '-a', '/tmp/uds']
Git version: '?'
Branch: ?
Remote: ?
Tracked URL: ?
CPU: 2 cores?

[zmod.log](https://github.com/user-attachments/files/19723692/zmod.log)

[guppyscreen_start.log](https://github.com/user-attachments/files/19723693/guppyscreen_start.log)

[printer.log](https://github.com/user-attachments/files/19723694/printer.log)

ghzserg, 2025-04-13:
config.tar download from fluidd -> config -> mod_data

ghzserg, 2025-04-13:
opt/config/mod_data/log/guppyscreen.4.log its ok. It's not error

ghzserg, 2025-04-13:
Then in the printer.log, the one with ? - ist ok.
This is flashforge version klipper.

ghzserg, 2025-04-13:
ntp1.vniiftri.ru - send me ntp server open in your country.

ghzserg, 2025-04-13:
opt/config/mod_data/variables.cfg - later I see at error

ghzserg, 2025-04-13:
Ssh used from native firmware. It's not zmod error.

TMTYD, 2025-04-13:
> Ssh used from native firmware. It's not zmod error.

OK, how do I get SSH access again? Flash the firmware with the factory firmware?

TMTYD, 2025-04-13:
> config.tar download from fluidd -> config -> mod_data

I would like to, but I no longer have ssh access

TMTYD, 2025-04-13:
> opt/config/mod_data/log/guppyscreen.4.log its ok. It's not error 

MV means that he wants to move the file but can't find it?

TMTYD, 2025-04-13:
> ntp1.vniiftri.ru - send me ntp server open in your country.

here please de.pool.ntp.org

ghzserg, 2025-04-13:
> > config.tar скачать с fluidd -> config -> mod_data
> 
> Я бы хотел, но у меня больше нет доступа по ssh

Yo not need ssh access download from Fluidd/Mainsail

ghzserg, 2025-04-13:
> > opt/config/mod_data/log/guppyscreen.4.log все в порядке. Это не ошибка
> 
> МВ означает, что он хочет переместить файл, но не может его найти?

Это нормальное поведение. Это не ошибка

TMTYD, 2025-04-13:
> > > config.tar скачать с fluidd -> config -> mod_data
> > 
> > Я бы хотел, но у меня больше нет доступа по ssh
> 
> Yo not need ssh access download from Fluidd/Mainsail

I really think it's best to do a new clean installation before I start printing

I no longer have access to Mainsail/Fluidd or ssh access

TMTYD, 2025-04-13:
I closed it because I did a new installation

https://github.com/ghzserg/zmod/discussions/76#discussion-8196564

closed 2025-04-13 by TMTYD