ghzserg/zmod #74: Mainsail/Fluidd not started

TMTYD, 2025-04-13:
- [ ] I have AD5M
- [ ] I have AD5M PRO
- [X] I have AD5X

- [ ] I work with the native screen
- [ ] I work without a screen
- [x] I work with Guppyscreen

Here are my SAVE_ZMOD_DATA parameters:

```
Call GET_ZMOD_DATA and put the output here.

info ZMOD parameters. Set via SAVE_ZMOD_DATA
[01:00:52.119] info You are using native screen // DISPLAY_ON
[01:00:52.123] info Parameters used at print start and bed leveling [START_PRINT]
[01:00:52.126] // Don't build bed mesh with native screen // SAVE_ZMOD_DATA PRINT_LEVELING=0
[01:00:52.148] // Don't use KAMP instead of BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
[01:00:52.169] // Test bed mesh before printing // SAVE_ZMOD_DATA MESH_TEST=1
[01:00:52.190] // Play MIDI on print start:  // SAVE_ZMOD_DATA MIDI_START=
[01:00:52.216] // Play MIDI on print pause:  // SAVE_ZMOD_DATA MIDI_PAUSE=
[01:00:52.238] // Check file MD5 checksum // SAVE_ZMOD_DATA FORCE_MD5=1
[01:00:52.260] // SKEW correction disabled // SAVE_ZMOD_DATA DISABLE_SKEW=1
[01:00:52.280] // Don't load Z-OFFSET from global settings // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
[01:00:52.301] // Enable nozzle line purge // SAVE_ZMOD_DATA DISABLE_PRIMING=0
[01:00:52.322] // Purge algorithm: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
[01:00:52.346] info Parameters used at print end and cancellation [END_PRINT]
[01:00:52.349] // Play MIDI at print end:  // SAVE_ZMOD_DATA MIDI_END=
[01:00:52.387] // Don't close dialogs on native screen // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
[01:00:52.391] // Auto disable motors after print/cancel in 25 sec // SAVE_ZMOD_DATA STOP_MOTOR=1
[01:00:52.411] // Don't reboot printer automatically // SAVE_ZMOD_DATA AUTO_REBOOT=0
[01:00:52.432] info System-wide parameters
[01:00:52.436] // Saving controlled by native screen // SAVE_ZMOD_DATA SAVE_RESTORE=1
[01:00:52.459] // Use SCV from config for shaper graphs // SAVE_ZMOD_DATA FIX_SCV=1
[01:00:52.481] // Don't fix E0011 error // SAVE_ZMOD_DATA FIX_E0011=0
[01:00:52.564] // Fix E0017 error // SAVE_ZMOD_DATA FIX_E0017=1
[01:00:52.645] // LED brightness at power on: 50 // SAVE_ZMOD_DATA LED=50
[01:00:52.671] // Play MIDI at power on:  // SAVE_ZMOD_DATA MIDI_ON=
[01:00:52.711] // Block Chinese clouds // SAVE_ZMOD_DATA CHINA_CLOUD=0
[01:00:52.733] // Use Fluidd/Mainsail button settings from ZMOD // SAVE_ZMOD_DATA SAVE_MOONRAKER=0

```

Detailed error description:

Yesterday, I activated the Guppy screen to see what it looked like, and then I clicked the Web button in the settings.

I couldn't access mainsail anymore, and the screen went out and then came back on, and the native screen was visible again.

But since then, I can't access either mainsail or fluidd.

If you need the **config.tar**, unfortunately, this isn't possible, because I haven't been able to connect via **putty** or winscp since yesterday evening.

I keep getting the message:
```
remote side unexpectedly closed network connection
```

TMTYD, 2025-04-13:
Why closed?

ghzserg, 2025-04-13:
closed this as a duplicateof The following errors occur repeatedly #62 2 hours ago
https://github.com/ghzserg/zmod/issues/62

TMTYD, 2025-04-13:
This has nothing to do with the China Cloud if I tap on web on the Guppyscreen and then it switches to Fluidd and I no longer have access to Mainsail or Fluidd

ghzserg, 2025-04-13:
closed this as a duplicateof Some errors #73 now

closed 2025-04-13 by ghzserg

TMTYD, 2025-04-13:
> closed this as a duplicateof Some errors #73 now
> 

ah ok now the link is correct, but what does it have to do with the fact that I have tapped on the web button in the Guppyscreen and now neither Mainsail/Fluidd can be called,

ghzserg, 2025-04-13:
Without TAR_CONFIG I can't tell you anything. The WEB macro simply changes fluidd to mainsail

TMTYD, 2025-04-13:
> Without TAR_CONFIG I can't tell you anything. The WEB macro simply changes fluidd to mainsail

Yes, exactly, that's why I want to make the Tar config available. But tapping the web button switched me to fluidd, and I can no longer access fluidd via a web browser.