ghzserg/zmod #78: Problem

TMTYD, 2025-04-13:
- [ ] I have AD5M
- [ ] I have AD5M PRO
- [x] I have AD5X

- [X] I work with the native screen
- [ ] I work without a screen
- [ ] I work with Guppyscreen
(replace the space with x inside the brackets)

I uploaded [TAR_CONFIG](https://we.tl/t-Eg2dIdR0DN) **My Tar Config**

Here are my SAVE_ZMOD_DATA parameters:

```zversion:
19:22:21
Klipper process priority (1-min, 40-max): 20
19:22:21
Using native Klipper
19:22:21
Use Fluidd/Mainsail button settings from ZMOD // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
19:22:21
Block Chinese clouds // SAVE_ZMOD_DATA CHINA_CLOUD=0
19:22:21
Play MIDI at power on:  // SAVE_ZMOD_DATA MIDI_ON=
19:22:21
LED brightness at power on: 50 // SAVE_ZMOD_DATA LED=50
19:22:21
Fix E0017 error // SAVE_ZMOD_DATA FIX_E0017=1
19:22:21
Don't fix E0011 error // SAVE_ZMOD_DATA FIX_E0011=0
19:22:21
Use SCV from config for shaper graphs // SAVE_ZMOD_DATA FIX_SCV=1
19:22:21
Saving controlled by native screen // SAVE_ZMOD_DATA SAVE_RESTORE=1
19:22:21
info System-wide parameters
19:22:21
Don't reboot printer automatically // SAVE_ZMOD_DATA AUTO_REBOOT=0
19:22:21
Auto disable motors after print/cancel in 25 sec // SAVE_ZMOD_DATA STOP_MOTOR=1
19:22:21
Don't close dialogs on native screen // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
19:22:21
Play MIDI at print end:  // SAVE_ZMOD_DATA MIDI_END=
19:22:21
info Parameters used at print end and cancellation [END_PRINT]
19:22:21
Purge algorithm: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
19:22:21
Enable nozzle line purge // SAVE_ZMOD_DATA DISABLE_PRIMING=0
19:22:21
Don't load Z-OFFSET from global settings // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
19:22:21
SKEW correction disabled // SAVE_ZMOD_DATA DISABLE_SKEW=1
19:22:21
Check file MD5 checksum // SAVE_ZMOD_DATA FORCE_MD5=1
19:22:21
Play MIDI on print pause:  // SAVE_ZMOD_DATA MIDI_PAUSE=
19:22:21
Play MIDI on print start:  // SAVE_ZMOD_DATA MIDI_START=
19:22:21
Test bed mesh before printing // SAVE_ZMOD_DATA MESH_TEST=1
19:22:21
Don't use KAMP instead of BED_MESH_CALIBRATE // SAVE_ZMOD_DATA USE_KAMP=0
19:22:21
Don't build bed mesh with native screen // SAVE_ZMOD_DATA PRINT_LEVELING=0
19:22:21
info Parameters used at print start and bed leveling [START_PRINT]
19:22:21
info You are using native screen // DISPLAY_ON
19:22:21
info ZMOD parameters. Set via SAVE_ZMOD_DATA
```

Detailed error description:

Unfortunately, I can't perform an update via the update manager; I get this message.

[image]

I'm also unable to connect via SSH; the new installation also gives this message:

[image]

ghzserg, 2025-04-14:
You have network problems. Possibly IP address conflict. In any case, the mod does not contain code for working with the network.

TMTYD, 2025-04-14:
> You have network problems. Possibly IP address conflict. In any case, the mod does not contain code for working with the network.

So my network works, it's just the firmware that has the problem. How can I restart the network from the firmware?

TMTYD, 2025-04-14:
> You have network problems. Possibly IP address conflict. In any case, the mod does not contain code for working with the network.

I ran zcheck on mainsail and, lo and behold, it works again. I can connect via putty again.

Perhaps this should be added to the FAQ before creating an issue.

[image]

ghzserg, 2025-04-14:
[image]
This is in the documentation
Recommendations for Improving Printer Stability
[Read if something isn’t working](https://github.com/ghzserg/zmod/wiki/Recomendations_en)

https://github.com/ghzserg/zmod/wiki/Recomendations_en#verify-stock-os-file-integrity

Before asking a question, read:
- **[TAR_CONFIG formatting algorithm](https://github.com/ghzserg/zmod/wiki/FAQ_en#help)** - without it, the ticket will not be considered
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ_en)
- [Recommendations](https://github.com/ghzserg/zmod/wiki/Recomendations_en)
- [Macro documentation](https://github.com/ghzserg/zmod/wiki/Macros_en)

closed 2025-04-14 by ghzserg

TMTYD, 2025-04-14:
There it is, at the bottom. :-)