ghzserg/zmod #89: If you find an error in the documentation (Если вы нашли ошибку в документации)

ghzserg, 2025-04-14:
If you find an error in the documentation.
Do the following:
1. Clone a copy of the documentation ```git clone https://github.com/ghzserg/zmod.wiki.git```
2. Edit the documentation
3. In the `zmod.wiki` folder, make ```git diff > fix.diff```
4. Attach this file to this topic

I don't have enough time to correct the documentation, and therefore the English version may have incorrect links or missing parts of the text. The documentation in Russian is supported by me whenever possible.

Если вы нашли ошибку в документации.
Сделайте следующее:
1. Клонируйте себе  копию документации ```git clone https://github.com/ghzserg/zmod.wiki.git```
2. Отредактируйте  документацию
3. В папке `zmod.wiki` сделайте ```git diff > fix.diff```
4. Приложите этот файл в эту тему

У меня не хватает времени исправлять документацию, и поэтому в английской версии могут быть некорректные ссылки или отсутствовать части текста. Документация на русском языке по возможности поддерживается мной.

TMTYD, 2025-04-14:
that's a good idea👍

TMTYD, 2025-04-16:
First fix.diff

Remove the txt extension, as the diff file type is not allowed here

[fix.diff.txt](https://github.com/user-attachments/files/19783996/fix.diff.txt)

TMTYD, 2025-04-16:
Changelog fix

[fix.diff.txt](https://github.com/user-attachments/files/19784080/fix.diff.txt)

ghzserg, 2025-04-17:
Thanks for the corrections.

TMTYD, 2025-04-17:
> Changelog fix
> 
> [Changelog_en.md.diff.txt](https://github.com/user-attachments/files/19784172/Changelog_en.md.diff.txt)

The note has been fixed
[Changelog_en.md.diff.txt](https://github.com/user-attachments/files/19801391/Changelog_en.md.diff.txt)

ghzserg, 2025-04-18:
> FAQ updated, I'm curious about the result
> 
> [FAQ_en.md.diff.txt](https://github.com/user-attachments/files/19813130/FAQ_en.md.diff.txt)

Thank you very much for your help. I have made corrections. Having corrected some links and TOC

TMTYD, 2025-04-18:
> > FAQ updated, I'm curious about the result
> > [FAQ_en.md.diff.txt](https://github.com/user-attachments/files/19813130/FAQ_en.md.diff.txt)
> 
> Thank you very much for your help. I have made corrections. Having corrected some links and TOC

[I fixed this](https://github.com/ghzserg/zmod/wiki/FAQ_en#whats-the-difference-between-using-the-screen-and-without-the-native-screen)

[FAQ_en.md.diff.txt](https://github.com/user-attachments/files/19815250/FAQ_en.md.diff.txt)

at the word elk

ghzserg, 2025-04-19:
> > > FAQ updated, I'm curious about the result
> > > [FAQ_en.md.diff.txt](https://github.com/user-attachments/files/19813130/FAQ_en.md.diff.txt)
> > 
> > 
> > Thank you very much for your help. I have made corrections. Having corrected some links and TOC
> 
> Are you satisfied with the result?

Thanks for the new patch, but it didn't install.
You need to get the changes via `git pull` or do `git clone` again
```
patching file FAQ_en.md
Hunk #1 FAILED at 2.
Hunk #2 FAILED at 12.
Hunk #3 succeeded at 38 with fuzz 2 (offset 1 line).
Hunk #4 FAILED at 59.
Hunk #5 FAILED at 75.
Hunk #6 FAILED at 86.
Hunk #7 succeeded at 420 with fuzz 2 (offset 318 lines).
Hunk #8 FAILED at 140.
Hunk #9 FAILED at 176.
Hunk #10 FAILED at 187.
Hunk #11 FAILED at 203.
Hunk #12 FAILED at 232.
Hunk #13 FAILED at 247.
Hunk #14 FAILED at 290.
Hunk #15 FAILED at 303.
Hunk #16 FAILED at 345.
Hunk #17 FAILED at 399.
Hunk #18 FAILED at 434.
Hunk #19 FAILED at 447.
17 out of 19 hunks FAILED -- saving rejects to file FAQ_en.md.rej
```

TMTYD, 2025-04-19:
> Thanks for the new patch, but it didn't install.
> You need to get the changes via `git pull` or do `git clone` again

ok all right I'll take a look at it

TMTYD, 2025-04-20:
Can you try it please

[FAQ_en.md.diff.txt](https://github.com/user-attachments/files/19824044/FAQ_en.md.diff.txt)

**Just corrected that**

[image]

TMTYD, 2025-04-21:
I've started editing the macro page, what do you think?

Of course it is not finished yet

[image]

ghzserg, 2025-04-25:
Thank you very much for the work done.
I have uploaded it here
https://github.com/ghzserg/zmod/wiki/Macros_Main_en
But as you can see, there is still a lot of work to do.
The idea of ​​structuring is very good, and thank you very much for spending your time on it.
I would suggest you test and write md files in one of the online editors.
For example, in this one
https://markdown-editor.github.io/

TMTYD, 2025-04-25:
Ok I'll take a look, yes there is still a lot to do, I wanted to make it available to see if you like it

TMTYD, 2025-04-25:
> But as you can see, there is still a lot of work to do.
> The idea of ​​structuring is very good, and thank you very much for spending your time on it.
> I would suggest you test and write md files in one of the online editors.
> For example, in this one

Very cool, thanks for the tip about the editor. I'm writing everything with Notepad++ and testing it here: https://markdownlivepreview.com/, but the other editor looks good too, thanks.

TMTYD, 2025-04-25:
> Thank you very much for the work done. I have uploaded it here https://github.com/ghzserg/zmod/wiki/Macros_Main_en But as you can see, there is still a lot of work to do. The idea of ​​structuring [...]

I found a problem with the linking.

This: https://github.com/ghzserg/zmod/wiki/Macros_Main_en#map-removal-algorithm

This should actually look like this:

https://github.com/ghzserg/zmod/wiki/Macros_Main_en#map_removal_algorithm

Can you please fix this?

TMTYD, 2025-04-25:
Updated

[Macros_Main_en.md.diff.txt](https://github.com/user-attachments/files/19915675/Macros_Main_en.md.diff.txt)

ghzserg, 2025-04-26:
I have uploaded your files, but github does not support all the features.
I think it is easier for you to create a fork of this repository.
https://github.com/TMTYD/zmod
And then use the wiki in your repository, changing the data there through the github interface
https://github.com/TMTYD/zmod.wiki.git
When you have something ready, I will clone your wiki and take the changes

https://github.com/ghzserg/zmod/wiki/Macros_Calibrations_en
https://github.com/ghzserg/zmod/wiki/Macros_ParaEnd_en

TMTYD, 2025-04-26:
Ok I see the style attributes he didn't adopt, although I tested it in your suggested editor.

Ok, I'll do it like this, I'll have to see how it works first

TMTYD, 2025-05-01:
Here is a preview of what it looks like, please don't clone it or anything like that

[Fork](https://github.com/TMTYD/zmod/wiki/Macros_en)

ghzserg, 2025-05-02:
Looks just great.

I would suggest adding to this same file a description of all the macros from https://github.com/ghzserg/zmod/wiki/Macros_en
And what you did, leave as a heading, because your version is convenient to use.
Also one macro is missing
https://github.com/ghzserg/zmod/wiki/Macros_en#restore_tar_config

TMTYD, 2025-05-02:
It's not finished yet, I'm in the process of doing one side after others. This should just be a preview for you

TMTYD, 2025-05-06:
So, I've edited everything. Please take a look to see if everything is OK.

If not, please make a list of what I need to change.

Here you go.

https://github.com/TMTYD/zmod/wiki

ghzserg, 2025-05-07:
Looks pretty good.
The only thing is that everywhere in the mod and documentation the links are to Macros_en, and there is no data in it now.
I would suggest collecting everything in one file or in Macros_en put links to all the functions that will link to Calibrations_en, etc.
And also rename Calibrations to Calibrations_en, etc.

TMTYD, 2025-05-07:
Ok I can do that, that's a little thing

ghzserg, 2025-05-09:
Hello.

I see a lot of work has been done.

But on the page:
1. https://github.com/TMTYD/zmod/wiki/Macros_en there are no links to macros.
In other parts of the documentation there are links like Macros_en#lang, which will stop working.
2. On the Telegram_en.md page there is also a problem with links - for example ZSSH_ON
3. In the description of System-Wide-Parameters_en, Start-Print-Parameters_en, End-Print-Parameters_en I did not find this text https://github.com/ghzserg/zmod/wiki/Macros#save_zmod_data. It describes what the global parameters SAVE_ZMOD_DATA are and how to use them
4. I did not find 2 main macros GET_ZMOD_DATA and SAVE_ZMOD_DATA
5. START PRINT in Macros_en is written incorrectly - it should be START_PRINT. The same goes for other macros.
6. No macros from the Pro version of the printer, AIR_CIRCULATION_INTERNAL, AIR_CIRCULATION_EXTERNAL, AIR_CIRCULATION_STOP
7. No macros CAMERA_ON, CAMERA_OFF, CAMERA_RESTART
8. No macro RESTORE_TAR_CONFIG

I think that all links to macros should remain on the Macros_en page, so as not to change the mod code and the code of all other documentation.
And from the Macros_en page redirect to https://github.com/TMTYD/zmod/wiki/Main_en#lang

TMTYD, 2025-05-09:
Ok I'll take a look, thanks for the feedback

TMTYD, 2025-05-12:
> But on the page:
> 
>     1. https://github.com/TMTYD/zmod/wiki/Macros_en there are no links to macros.
>        In other parts of the documentation there are links like Macros_en#lang, which will stop working.

The links are the headings, depending on the macro, and then on the corresponding page, there are links to the macros. This is better for the overview.

[image]

TMTYD, 2025-05-12:
> 2\. On the Telegram_en.md page there is also a problem with links - for example ZSSH_ON

link fixed

TMTYD, 2025-05-12:
> * In the description of System-Wide-Parameters_en, Start-Print-Parameters_en, End-Print-Parameters_en I did not find this text https://github.com/ghzserg/zmod/wiki/Macros#save_zmod_data. It describes what the global parameters SAVE_ZMOD_DATA are and how to use them
> 
> * I did not find 2 main macros GET_ZMOD_DATA and SAVE_ZMOD_DATA

Okay, I see, where should the text go, or in which category? I also noticed that the part about the camera is missing.

The part about the camera, can it go in the System category?

TMTYD, 2025-05-12:
> 5\. START PRINT in Macros_en is written incorrectly - it should be START_PRINT. The same goes for other macros.

This is just an overview table, where and which macros are available, correctly written they are then in the corresponding category

ghzserg, 2025-05-12:
> The links are the headings, depending on the macro, and then on the corresponding page, there are links to the macros. This is better for the overview.
>
> [image]

Hello.

It looks better for visual presentation of information.
But in the mod and in other places of the documentation there are many links to Macros_en with the name of a specific macro indicated after #.

And if left as is, these links lead to nowhere.

The user will not be able to find a link to the required macro, because he does not know which section this or that macro belongs to.

Remaking all the links to new pages is a lot of work. And in the versions of the mod that came out before the remake, the old links to Macros_enб will remain, which will lead to nowhere. The easiest option is to place a general list of all macros after the list of macros divided into categories, in the description of which there will be a link to the desired section.

TMTYD, 2025-05-12:
> 8\. No macro RESTORE_TAR_CONFIG

I added it

https://github.com/TMTYD/zmod/wiki/System_en#tar_config

ghzserg, 2025-05-12:
> > 5. START PRINT in Macros_en is written incorrectly - it should be START_PRINT. The same goes for other macros.
> 
> This is just an overview table, where and which macros are available, correctly written they are then in the corresponding category

The overview category will confuse people and 100% people will write START PRINT instead of START_PRINT.
Just like now people write USE_KAMP=1 instead of SAVE_ZMOD_DATA USE_KAMP=1
The fewer places that can be misunderstood, the fewer mistakes users will make.

TMTYD, 2025-05-13:
> Hello.
>
> It looks better for visual presentation of information. But in the mod and in other places of the documentation there are many links to Macros_en with the name of a specific macro indicated [...]

Hi

What exactly do you mean?

TMTYD, 2025-05-13:
I added the rest of the stuff with the zmod, please take a look

ghzserg, 2025-05-14:
Thanks for your work. I reworked your main page and posted it here.
I changed the structure of links, also added macros that were not in your version
https://github.com/ghzserg/zmod/wiki/Macros_en_new
As soon as I have time, I will rework and supplement the pages Telegram_en, Main_en, etc.

ghzserg, 2025-05-14:
> I added the rest of the stuff with the zmod, please take a look

https://github.com/ghzserg/zmod/wiki/Telegram_en

ghzserg, 2025-05-14:
https://github.com/ghzserg/zmod/wiki/Filament_en

ghzserg, 2025-05-14:
https://github.com/ghzserg/zmod/wiki/Zmod_en

ghzserg, 2025-05-14:
> I added the rest of the stuff with the zmod, please take a look

https://github.com/ghzserg/zmod/wiki/Global_en
All files translated
https://github.com/ghzserg/zmod/wiki/Macros_en
Thank you very much for doing this work.

TMTYD, 2025-05-16:
Glad to hear you like it. However, I noticed a mistake.

[Global_en](https://github.com/ghzserg/zmod/wiki/Global_en)

[image]

The forward button points to System_en, although Zmod would be next in the sequence

[image]

And the one at System_en points to Zmod, although Telegram would be next

[System_en](https://github.com/ghzserg/zmod/wiki/System_en)

aldiserg, 2025-07-05:
Telegram Bot Installation On Kubernetes Via Helm

[fix.diff.txt](https://github.com/user-attachments/files/21080481/fix.diff.txt)

ghzserg, 2025-08-25:
> Telegram Bot Installation On Kubernetes Via Helm
> 
> [fix.diff.txt](https://github.com/user-attachments/files/21080481/fix.diff.txt)

When you sent the fix I was on vacation and missed your message. I added your fix to the documentation. Thanks.

ghzserg, 2025-09-15:
Большое  спасибо.

john-clark, 2025-10-28:
This may be too much changes. I will not be offended if you do not use. 

This change will allow for more languages and easier expansion of the wiki.  Using a common standard top language selector on each page and folder for each language. This way pages can be named the same for each language, and only the folder prefix change in URLs. People can copy a language folder and edit all the files in it and contribute the full language. 

I hope you find this useful.
[fix.diff.txt](https://github.com/user-attachments/files/23177080/fix.diff.txt)

You may find this more useful due to the large number of changes.
[zmod.wiki.zip](https://github.com/user-attachments/files/23177135/zmod.wiki.zip)

Thank you for your work on ZMOD.

ghzserg, 2025-11-30:
Я  не знал, что так возможно. Прочитаю и попробую использовать

closed 2025-12-01 by ghzserg