ghzserg/zmod #91: Problem

TMTYD, 2025-04-14:
- [ ] I have AD5M
- [ ] I have AD5M PRO
- [x] I have AD5X

- [X] I work with the native screen
- [ ] I work without a screen
- [ ] I work with Guppyscreen
(replace the space with x inside the brackets)

[My tar config](https://we.tl/t-VMWs2tT4Vj)

I use the start code:

```
CHECK_MD5
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]  
M190 S[bed_temperature_initial_layer_single]  
M104 S[nozzle_temperature_initial_layer] 
FORCE_KAMP=True
```

I wrote the end code:

```
END_PRINT
; total layers count = [total_layer_count]
```

Here are my SAVE_ZMOD_DATA parameters:

```
Installed from USB: 1.4.3. Update from Fluidd/Mainsail: 1.4.3
19:09:59
zversion:
19:09:59
Klipper process priority (1-min, 40-max): 20
19:09:59
Using native Klipper
19:09:59
Save Fluidd/Mainsail button changes // SAVE_ZMOD_DATA SAVE_MOONRAKER=1
19:09:59
Allow Chinese clouds // SAVE_ZMOD_DATA CHINA_CLOUD=1
19:09:59
Play MIDI at power on:  // SAVE_ZMOD_DATA MIDI_ON=
19:09:59
LED brightness at power on: 50 // SAVE_ZMOD_DATA LED=50
19:09:59
Fix E0017 error // SAVE_ZMOD_DATA FIX_E0017=1
19:09:59
Don't fix E0011 error // SAVE_ZMOD_DATA FIX_E0011=0
19:09:59
Use SCV from config for shaper graphs // SAVE_ZMOD_DATA FIX_SCV=1
19:09:59
Saving controlled by native screen // SAVE_ZMOD_DATA SAVE_RESTORE=1
19:09:58
info System-wide parameters
19:09:58
Don't reboot printer automatically // SAVE_ZMOD_DATA AUTO_REBOOT=0
19:09:58
Auto disable motors after print/cancel in 25 sec // SAVE_ZMOD_DATA STOP_MOTOR=1
19:09:58
Auto close dialogs on native screen after 20 sec (fast) // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
19:09:58
Play MIDI at print end:  // SAVE_ZMOD_DATA MIDI_END=
19:09:58
info Parameters used at print end and cancellation [END_PRINT]
19:09:58
Purge algorithm: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
19:09:58
Enable nozzle line purge // SAVE_ZMOD_DATA DISABLE_PRIMING=0
19:09:58
Don't load Z-OFFSET from global settings // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
19:09:58
SKEW correction disabled // SAVE_ZMOD_DATA DISABLE_SKEW=1
19:09:58
Check file MD5 checksum // SAVE_ZMOD_DATA FORCE_MD5=1
19:09:58
Play MIDI on print pause:  // SAVE_ZMOD_DATA MIDI_PAUSE=
19:09:58
Play MIDI on print start:  // SAVE_ZMOD_DATA MIDI_START=
19:09:58
Test bed mesh before printing // SAVE_ZMOD_DATA MESH_TEST=1
19:09:58
Use adaptive mesh (KAMP) where possible instead of full bed mesh // SAVE_ZMOD_DATA USE_KAMP=1
19:09:58
Build bed mesh with native screen on each print. // SAVE_ZMOD_DATA PRINT_LEVELING=1
19:09:58
info Parameters used at print start and bed leveling [START_PRINT]
19:09:58
info You are using native screen // DISPLAY_ON
19:09:58
info ZMOD parameters. Set via SAVE_ZMOD_DATA
```

Detailed error description:

I started my first print and wanted to print an xyz cube. When I started printing, this error appeared in the console:

`Error on 'G92 E': unable to parse`

I let it continue. When it started printing, the print didn't stick to the print bed, so I aborted the print.

I then enabled **firmware retraction** in the orca slicer as described [here](https://github.com/ghzserg/zmod/wiki/FAQ_en#what-is-firmware-retraction) and sliced ​​again. I uploaded the file and tried to print it, but it finished with a status of OK and did nothing else.

See here:

```
Status: OK
19:05:25
action:prompt_end
19:05:25
ok
19:05:25
ok
19:05:25
action:prompt_show
19:05:25
action:prompt_footer_button Cancel|RESPOND TYPE=command MSG=action:prompt_end
19:05:24
action:prompt_footer_button Start print|PRINT_ZCOLOR LEVELING=1 FILENAME="tsXYZCalibrationCube20mm.gcode" TOOL0=1 TOOL1=2 TOOL2=3 TOOL3=4|red
19:05:24
action:prompt_button_group_end
19:05:24
action:prompt_button In file 1 → in spool 2: PLA/black|CHANGE_TOOL_ZCOLOR TOOL=1 LEVELING=1 FILENAME="tsXYZCalibrationCube20mm.gcode" TOOL0=1 TOOL1=2 TOOL2=3 TOOL3=4|primary
19:05:23
action:prompt_button_group_start
19:05:23
action:prompt_text Print with bed leveling
19:05:23
action:prompt_text File to print: tsXYZCalibrationCube20mm.gcode
19:05:23
action:prompt_text Map file color to spool
19:05:23
action:prompt_begin Loaded material
19:05:22
action:prompt_end
19:05:22
info Getting data for KAMP.
19:05:22
info File /usr/data/gcodes/tsXYZCalibrationCube20mm.gcode verified successfully. MD5 checksum matches.
19:05:21
info Started MD5 check tsXYZCalibrationCube20mm.gcode
19:05:20
File selected
19:05:20
File opened:tsXYZCalibrationCube20mm.gcode Size:464159
```
And the printer log says this:

```
Traceback (most recent call last):
  File "/usr/prog/klipper/klippy/gcode.py", line 459, in _respond_raw
    os.write(self.fd, (msg+"\n").encode())
BlockingIOError: [Errno 11] Resource temporarily unavailable

```
[printer.log](https://github.com/user-attachments/files/19740450/printer.log)

max-hasky, 2025-04-14:
Please insert the file you are typing. Without it, the developer will not be able to understand what exactly went wrong.

TMTYD, 2025-04-14:
please

[gcodes-2025314-19330.zip](https://github.com/user-attachments/files/19740826/gcodes-2025314-19330.zip)

TMTYD, 2025-04-14:
> * [ ] 
>       [ ]  I have AD5M
>       [ ] 
>       [ ]  I have AD5M PRO
>       [x] 
>       [x]  I have AD5X
>       [x] 
>       [x]  I work with the native screen
>       [ ] 
>       [ ]  I work without a screen
>       [ ] 
>       [ ]  I work with Guppyscreen
>       (replace the space with x inside the brackets)
> 
> 
> [My tar config](https://we.tl/t-VMWs2tT4Vj)
> 
> I use the start code:
> 
> ```
> CHECK_MD5
> SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
> START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]  
> M190 S[bed_temperature_initial_layer_single]  
> M104 S[nozzle_temperature_initial_layer] 
> FORCE_KAMP=True
> ```
> 
> I wrote the end code:
> 
> ```
> END_PRINT
> ; total layers count = [total_layer_count]
> ```
> 
> Here are my SAVE_ZMOD_DATA parameters:
> 
> ```
> Installed from USB: 1.4.3. Update from Fluidd/Mainsail: 1.4.3
> 19:09:59
> zversion:
> 19:09:59
> Klipper process priority (1-min, 40-max): 20
> 19:09:59
> Using native Klipper
> 19:09:59
> Save Fluidd/Mainsail button changes // SAVE_ZMOD_DATA SAVE_MOONRAKER=1
> 19:09:59
> Allow Chinese clouds // SAVE_ZMOD_DATA CHINA_CLOUD=1
> 19:09:59
> Play MIDI at power on:  // SAVE_ZMOD_DATA MIDI_ON=
> 19:09:59
> LED brightness at power on: 50 // SAVE_ZMOD_DATA LED=50
> 19:09:59
> Fix E0017 error // SAVE_ZMOD_DATA FIX_E0017=1
> 19:09:59
> Don't fix E0011 error // SAVE_ZMOD_DATA FIX_E0011=0
> 19:09:59
> Use SCV from config for shaper graphs // SAVE_ZMOD_DATA FIX_SCV=1
> 19:09:59
> Saving controlled by native screen // SAVE_ZMOD_DATA SAVE_RESTORE=1
> 19:09:58
> info System-wide parameters
> 19:09:58
> Don't reboot printer automatically // SAVE_ZMOD_DATA AUTO_REBOOT=0
> 19:09:58
> Auto disable motors after print/cancel in 25 sec // SAVE_ZMOD_DATA STOP_MOTOR=1
> 19:09:58
> Auto close dialogs on native screen after 20 sec (fast) // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
> 19:09:58
> Play MIDI at print end:  // SAVE_ZMOD_DATA MIDI_END=
> 19:09:58
> info Parameters used at print end and cancellation [END_PRINT]
> 19:09:58
> Purge algorithm: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
> 19:09:58
> Enable nozzle line purge // SAVE_ZMOD_DATA DISABLE_PRIMING=0
> 19:09:58
> Don't load Z-OFFSET from global settings // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
> 19:09:58
> SKEW correction disabled // SAVE_ZMOD_DATA DISABLE_SKEW=1
> 19:09:58
> Check file MD5 checksum // SAVE_ZMOD_DATA FORCE_MD5=1
> 19:09:58
> Play MIDI on print pause:  // SAVE_ZMOD_DATA MIDI_PAUSE=
> 19:09:58
> Play MIDI on print start:  // SAVE_ZMOD_DATA MIDI_START=
> 19:09:58
> Test bed mesh before printing // SAVE_ZMOD_DATA MESH_TEST=1
> 19:09:58
> Use adaptive mesh (KAMP) where possible instead of full bed mesh // SAVE_ZMOD_DATA USE_KAMP=1
> 19:09:58
> Build bed mesh with native screen on each print. // SAVE_ZMOD_DATA PRINT_LEVELING=1
> 19:09:58
> info Parameters used at print start and bed leveling [START_PRINT]
> 19:09:58
> info You are using native screen // DISPLAY_ON
> 19:09:58
> info ZMOD parameters. Set via SAVE_ZMOD_DATA
> ```
> 
> Detailed error description:
> 
> I started my first print and wanted to print an xyz cube. When I started printing, this error appeared in the console:
> 
> `Error on 'G92 E': unable to parse`
> 
> I let it continue. When it started printing, the print didn't stick to the print bed, so I aborted the print.
> 
> I then enabled **firmware retraction** in the orca slicer as described [here](https://github.com/ghzserg/zmod/wiki/FAQ_en#what-is-firmware-retraction) and sliced ​​again. I uploaded the file and tried to print it, but it finished with a status of OK and did nothing else.
> 
> See here:
> 
> ```
> Status: OK
> 19:05:25
> action:prompt_end
> 19:05:25
> ok
> 19:05:25
> ok
> 19:05:25
> action:prompt_show
> 19:05:25
> action:prompt_footer_button Cancel|RESPOND TYPE=command MSG=action:prompt_end
> 19:05:24
> action:prompt_footer_button Start print|PRINT_ZCOLOR LEVELING=1 FILENAME="tsXYZCalibrationCube20mm.gcode" TOOL0=1 TOOL1=2 TOOL2=3 TOOL3=4|red
> 19:05:24
> action:prompt_button_group_end
> 19:05:24
> action:prompt_button In file 1 → in spool 2: PLA/black|CHANGE_TOOL_ZCOLOR TOOL=1 LEVELING=1 FILENAME="tsXYZCalibrationCube20mm.gcode" TOOL0=1 TOOL1=2 TOOL2=3 TOOL3=4|primary
> 19:05:23
> action:prompt_button_group_start
> 19:05:23
> action:prompt_text Print with bed leveling
> 19:05:23
> action:prompt_text File to print: tsXYZCalibrationCube20mm.gcode
> 19:05:23
> action:prompt_text Map file color to spool
> 19:05:23
> action:prompt_begin Loaded material
> 19:05:22
> action:prompt_end
> 19:05:22
> info Getting data for KAMP.
> 19:05:22
> info File /usr/data/gcodes/tsXYZCalibrationCube20mm.gcode verified successfully. MD5 checksum matches.
> 19:05:21
> info Started MD5 check tsXYZCalibrationCube20mm.gcode
> 19:05:20
> File selected
> 19:05:20
> File opened:tsXYZCalibrationCube20mm.gcode Size:464159
> ```
> 
> And the printer log says this:
> 
> ```
> Traceback (most recent call last):
>   File "/usr/prog/klipper/klippy/gcode.py", line 459, in _respond_raw
>     os.write(self.fd, (msg+"\n").encode())
> BlockingIOError: [Errno 11] Resource temporarily unavailable
> ```
> 
> [printer.log](https://github.com/user-attachments/files/19740450/printer.log)

In addition, the console also says this

[image]

ghzserg, 2025-04-14:
1. `FORCE_KAMP=True `
You wrote it in the wrong place - it is a parameter of the START_PRINT macro, not a separate macro.
2. `Error on 'G92 E': unable to parse` and `cool_fun`
This command is sent by the native screen. But it does not interrupt printing.
3. CANCEL_PRINT was sent by you

ghzserg, 2025-04-14:
CHECK_MD5 - лишнее, он уже включен по умолчанию в глобальных параметрах

TMTYD, 2025-04-14:
I just noticed that too, thanks

closed 2025-04-14 by TMTYD

TMTYD, 2025-04-14:
If I want to print an object again after canceling, I still get this message

```
Traceback (most recent call last):
  File "/usr/prog/klipper/klippy/gcode.py", line 459, in _respond_raw
    os.write(self.fd, (msg+"\n").encode())
BlockingIOError: [Errno 11] Resource temporarily unavailable
```

ghzserg, 2025-04-15:
Do it Tar_config Without it, it is impossible to understand the reason.

TMTYD, 2025-04-15:
> Do it Tar_config Without it, it is impossible to understand the reason.
> Off spoolman

here is the link to my current tar config
[tar config](https://we.tl/t-VMWs2tT4Vj)

ghzserg, 2025-04-15:
You sent the same file. Create a new ticket and generate a new file by following ALL the steps in sequence.
https://github.com/ghzserg/zmod/wiki/FAQ_en#how-to-contact-support

TMTYD, 2025-04-15:
> 1. `FORCE_KAMP=True `
>        You wrote it in the wrong place - it is a parameter of the START_PRINT macro, not a separate macro.
> 
Why would this be written in the wrong place? It's described as follows: [FAQ_en](https://github.com/ghzserg/zmod/wiki/FAQ_en#by-modifying-the-start-code-and-start_print-macro)

ghzserg, 2025-04-15:
The FAQ is written correctly.
```
1: START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] FORCE_LEVELING=True
```
You wrote it incorrectly
```
1: START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
2: FORCE_LEVELING=True
```
I numbered the lines so that the difference is visible.
In the FAQ there is 1 line and FORCE_LEVELING is a parameter of START_PRINT
You have 2 lines and FORCE_LEVELING is considered a separate macro - which is not there

TMTYD, 2025-04-15:
> The FAQ is written correctly.
>
> ```
> 1: START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] FORCE_LEVELING=True
> ```
>
> You wrote it incorrectly [...]

It should be like the first picture of me:

[image]

So it is wrong?

[image]

ghzserg, 2025-04-15:
[image]
I'm not looking at your settings in the slicer, I'm looking at your settings in the final gcode.