ghzserg/zmod #97: Проблема

gipermoth, 2025-04-16:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)** - без него тикет рассматриваться не будет
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ X] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [X ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.
https://www.dropbox.com/scl/fi/c1smzsbeahg7nu1856cvu/config.tar?rlkey=rdzdq1zhoaq2iffynpavl7hav&st=cohblwzx&dl=0

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
CHECK_MD5 DELETE=true
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
```

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```

21:28:54 
$ GET_ZMOD_DATA
21:28:54 
info Параметры zmod. Задаются через SAVE_ZMOD_DATA
21:28:54 
info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
21:28:54 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
21:28:54 
// При каждой печати строить карту стола. // SAVE_ZMOD_DATA PRINT_LEVELING=1
21:28:55 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
21:28:55 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
21:28:55 
// Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
21:28:55 
// Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
21:28:55 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
21:28:55 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
21:28:55 
// Не загружать Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
21:28:55 
// Включить очистку сопла линией // SAVE_ZMOD_DATA DISABLE_PRIMING=0
21:28:55 
// Не использовать предочистку сопла // SAVE_ZMOD_DATA PRECLEAR=0
21:28:55 
// Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
21:28:55 
info Параметры используемые при окончании и отмене печати [END_PRINT]
21:28:55 
// Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
21:28:55 
// Не закрывать диалоги на родном экране // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
21:28:55 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд // SAVE_ZMOD_DATA STOP_MOTOR=1
21:28:55 
// Не перезапускать принтер автоматически // SAVE_ZMOD_DATA AUTO_REBOOT=0
21:28:55 
// Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
21:28:55 
info Общесистемные параметры
21:28:55 
// Сохранять данные для восстановления печати // SAVE_ZMOD_DATA SAVE_RESTORE=1
21:28:55 
// Используется датчик наличия филамента // SAVE_ZMOD_DATA MOTION_SENSOR=0
21:28:55 
// Использовать SCV из конфигурации при построении графиков шейперов // SAVE_ZMOD_DATA FIX_SCV=1
21:28:55 
// Исправить ошибку E0011 // SAVE_ZMOD_DATA FIX_E0011=1
21:28:56 
// Исправить ошибку E0017 // SAVE_ZMOD_DATA FIX_E0017=1
21:28:56 
// Яркость LED при включении: 50 // SAVE_ZMOD_DATA LED=50
21:28:56 
// Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
21:28:56 
// Использовать стандартный SAVE_CONFIG // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
21:28:56 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
```

Подробное описание ошибки:
Максимально подробно опишите, что Вы делали, что Вы планировали  получить и что Вы получили.

Был перерыв в печати неделю. Была разборка экструдера (забился ПЛА). До этого печатал нормально. Сейчас после проверки стола ничего не происходит. Стол и хотэнд на рабочей темперутуре. Флюд  пишет что идет печать, но ни чего не происходит.

ghzserg, 2025-04-16:
Без Gcode я не смогу посмотреть в чем у вас проблема

gipermoth, 2025-04-16:
Добавил ссылку на Gcode

ghzserg, 2025-04-16:
Судя по логам у вас скачет PID на экструдере. 
Я видел калибровку экструдера  под 245 градусов, но печатаете вы на 220 перекалибруйте экструдер под 220

gipermoth, 2025-04-16:
Перекалибровал на 220. Печать началась, но со странной паузой в пару минут после достижения температуры. Стоит соплемод. Очень быстро идет нагрев сопла и его охлаждение. Не сравнить с родным. Может в этом проблема?

ghzserg, 2025-04-16:
Да с этим проблема.

closed 2025-04-17 by ghzserg