ghzserg/zmod #98: Проблема AD5X

mortonlame, 2025-04-19:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)** - без него тикет рассматриваться не будет
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ ] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика. Стало земетно, если область печати на весь стол. на небольших печатях некритично. Но смещение всегда непостоянно, но вправо и вверх, то - только вправо. орку переустанавливал и менял профили

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.
```

Подробное описание ошибки:
Я думаю, это ошибка камп. После замера печать начинается со смещением вправо и вверх или вверх, в зависимости от расположения линии очистки. это вызывает удары в крайнем верхнем или левом положении. И соответственно - смещение слоев печати.

[]     (https://disk.yandex.ru/d/j_RtmApdRfadqQ)

ghzserg, 2025-04-20:
Проблема в том что реально используемый стол по конфигу 215 на 215

mortonlame, 2025-04-21:
На самом деле, нет. я в стоке напечатал со стандартным профилем из орки 220х220. Как вариант, хочу попробовать удалить мод, сбросить на заводские и накатить фактори, если нет мыслей по решению этой проблемы.

TMTYD, 2025-04-22:
Yes, I think this will be related to Macro Smartpark

ghzserg, 2025-04-22:
Indeed it is related to parking. I fixed it, but I can't test it. If you haven't uninstalled the mod yet. Your test will help

ghzserg, 2025-04-23:
https://github.com/ghzserg/zmod/wiki/Changelog_en#version-151

TMTYD, 2025-04-24:
> Indeed it is related to parking. I fixed it, but I can't test it. If you haven't uninstalled the mod yet. Your test will help

Unfortunately, as mentioned here, I've uninstalled zmod for now. I'll take another look at it when I'm done printing.

[thx](https://github.com/ghzserg/zmod/discussions/95#discussion-8206791)

mortonlame, 2025-04-25:
Спасибо, теперь всё работает!

closed 2025-04-25 by mortonlame