ghzserg/zmod #108: Проблема Зависает принтер с ошибкой E0011 при печати с нечётно оболочкой

soblod, 2025-05-11:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)** - без него тикет рассматриваться не будет
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [X] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
Впишите сюда Ваш стартовый G-код принтера
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
```

Я написал завершающий код:

```
Впишите сюда Ваш завершающий G-код принтера
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
Вызовите  GET_ZMOD_DATA и впишите  вывод сюда.

15:18:59 
// Play MIDI on print pause:  // SAVE_ZMOD_DATA MIDI_PAUSE=
15:19:00 
// Check file MD5 checksum // SAVE_ZMOD_DATA FORCE_MD5=1
15:19:00 
// SKEW correction disabled // SAVE_ZMOD_DATA DISABLE_SKEW=1
15:19:00 
// Don't load Z-OFFSET from global settings // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
15:19:00 
// Enable nozzle line purge // SAVE_ZMOD_DATA DISABLE_PRIMING=0
15:19:00 
// Don't use nozzle pre-purge // SAVE_ZMOD_DATA PRECLEAR=0
15:19:00 
// Purge algorithm: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
15:19:00 
info Parameters used at print end and cancellation [END_PRINT]
15:19:00 
// Play MIDI at print end:  // SAVE_ZMOD_DATA MIDI_END=
15:19:00 
// Don't close dialogs on native screen // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
15:19:00 
// Auto disable motors after print/cancel in 25 sec // SAVE_ZMOD_DATA STOP_MOTOR=1
15:19:00 
// Don't reboot printer automatically // SAVE_ZMOD_DATA AUTO_REBOOT=0
15:19:00 
// Power off FF5M Pro after (min):  // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
15:19:00 
info System-wide parameters
15:19:00 
// Saving controlled by native screen // SAVE_ZMOD_DATA SAVE_RESTORE=1
15:19:00 
// Using filament presence sensor // SAVE_ZMOD_DATA MOTION_SENSOR=0
15:19:00 
// Use SCV from config for shaper graphs // SAVE_ZMOD_DATA FIX_SCV=1
15:19:00 
// Fix E0011 error // SAVE_ZMOD_DATA FIX_E0011=1
15:19:00 
// Fix E0017 error // SAVE_ZMOD_DATA FIX_E0017=1
15:19:00 
// LED brightness at power on: 50 // SAVE_ZMOD_DATA LED=50
15:19:00 
// Play MIDI at power on:  // SAVE_ZMOD_DATA MIDI_ON=
15:19:00 
// Use standard SAVE_CONFIG // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
15:19:01 
// SWAP used on EMMC // SAVE_ZMOD_DATA USE_SWAP=1
15:19:01 
// Block Chinese clouds // SAVE_ZMOD_DATA CHINA_CLOUD=0
15:19:01 
// Use Fluidd/Mainsail button settings from ZMOD // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
15:19:01 
// Using native Klipper
15:19:01 
// Klipper process priority (1-min, 40-max): 20
15:19:01 
// Weight: 800; Control is configured but inactive.
15:19:01 
// Klipper is disabled when triggered. // ZCONTROL_ABORT
15:19:01 
// Available languages: en, ru, de, fr, it, es, ja, ko, zh // LANG LANG=en
15:19:01 
// zversion:
15:19:02 
// Installed from USB: 1.5.1. Update from Fluidd/Mainsail: 1.5.1
```

Подробное описание ошибки:
Максимально подробно опишите, что Вы делали, что Вы планировали  получить и что Вы получили.

При включение в Orca слайсере последней версии режим нечётной оболочки принтер зависает во время печати с ошибкой E0011, в рандомные моменты, на некоторых принтерах может несколько печатей сделать потом зависнуть, на других каждую печать, при удаление ZMOD печать идет стабильно. Пробовал сделать все рекомендации по устранению ошибки E0011 включая UPDATE_MCU, все рекомендации из раздела вики по стабильной печати.

[config.zip](https://github.com/user-attachments/files/20149110/config.zip)

ghzserg, 2025-05-11:
1. Включите контроль md5
2. Приложите файл, который не печатается
3. Проведите проверку системы через чек систем
4. Отключите родной экран и напечатайте этот же файл без родного экрана

soblod, 2025-05-11:
> 1. Включите контроль md5
> 2. Приложите файл, который не печатается
> 3. Провести проверку системы через чек-системы
> 4. Отключите родной экран и напечатайте этот же файл без родного экрана.

1. как включить ? в описание написано

1. На машине с Windows, где установлен слайсер.
  а) Копируем из папки "mod" с принтера файл "addMD5.bat" в удобной место
  б) Добавляем в слайсер скрипт из пункта 1.а, 
     заменяя "disk:\patch\to\file\" на свой путь к данному скрипту: 

вот скрипт из файла addMD5.bat я не вижу параметр "disk:\patch\to\file\"

@echo off
set prefecs=; MD5:
FOR /F %%i IN ('certutil -hashfile %1 md5^|find /v "MD5"^|find /v "CertUtil"') DO set md5=%%i
set payload=%prefecs%%md5%
REM echo %payload%>>%1 
SET filename=%1
echo %payload% > %TEMP%\temp.md5 & type %filename:/=\% >> %TEMP%\temp.md5 & move /y %TEMP%\temp.md5 %filename:/=\% >nul

ghzserg, 2025-05-12:
> > 1. Включите контроль md5
> > 2. Приложите файл, который не печатается
> > 3. Провести проверку системы через чек-системы
> > 4. Отключите родной экран и напечатайте этот же файл без родного экрана.
> 
> 1. как включить ? в описание написано
> 2. На машине с Windows, где установлен слайсер.
>    а) Копируем из папки "mod" с принтера файл "addMD5.bat" в удобной место
>    б) Добавляем в слайсер скрипт из пункта 1.а,
>    заменяя "disk:\patch\to\file" на свой путь к данному скрипту:
> 
> вот скрипт из файла addMD5.bat я не вижу параметр "disk:\patch\to\file"
> 
> [@echo](https://github.com/echo) off set prefecs=; MD5: FOR /F %%i IN ('certutil -hashfile %1 md5^|find /v "MD5"^|find /v "CertUtil"') DO set md5=%%i set payload=%prefecs%%md5% REM echo %payload%>>%1 SET filename=%1 echo %payload% > %TEMP%\temp.md5 & type %filename:/=% >> %TEMP%\temp.md5 & move /y %TEMP%\temp.md5 %filename:/=% >nul

[image]
Замените  ```I:\3D\FF5M\addMD5.bat;``` на путь к вашему файлу.
Это и есть ```disk:\patch\to\file```
В примере выше:
- `disk` = `I`
- `\patch\to\file` = `\3D\FF5M\addMD5.bat`

soblod, 2025-05-12:
> [image] Замените `I:\3D\FF5M\addMD5.bat;` на путь к вашему файлу. Это и есть `disk:\patch\to\file` В примере выше:
>
> * `disk` = `I`
> * `\patch\to\file` = `\3D\FF5M\addMD5.bat`

1. Сделал
2. В архиве
3. Сделал
4. Пока не получается, не могу разобраться со стартом печати

Прикладываю архив с проектом и с TAR_CONFIG последней попытки

[Files.zip](https://github.com/user-attachments/files/20160609/Files.zip)

ghzserg, 2025-05-12:
Так же у вас стоит не последняя версия `Версия zmod: 1.5.1 Date:   Thu May 8 19:15:44 2025 +0500`
И не включен контроль md5

soblod, 2025-05-12:
> Так же у вас стоит не последняя версия `Версия zmod: 1.5.1 Date: Thu May 8 19:15:44 2025 +0500` И не включен контроль md5

Приложил gcode проблемной печати, сейчас MD5 Включен ? перепрошил по новой принтер последним релизом 1ю5ю1 который сегодня скачал

[File.zip](https://github.com/user-attachments/files/20165269/File.zip)

ghzserg, 2025-05-12:
Вы не правильно сняли логи. Снимите по алгоритму указанному в вашем первом сообщении.

ghzserg, 2025-05-14:
Закрыто, т.к. запрос заброшен

soblod, 2025-05-14:
[config (1)_2.zip](https://github.com/user-attachments/files/20214464/config.1._2.zip)

Сейчас правильно логи сняты ?

ghzserg, 2025-05-15:
1. [У вас не обновлен мод.](https://github.com/ghzserg/zmod/wiki/FAQ#%D1%8F-%D1%83%D1%81%D1%82%D0%B0%D0%BD%D0%BE%D0%B2%D0%B8%D0%BB-%D0%BF%D0%BE%D1%81%D0%BB%D0%B5%D0%B4%D0%BD%D1%8E%D1%8E-%D0%B2%D0%B5%D1%80%D1%81%D0%B8%D1%8E-%D0%B0-%D1%80%D0%B0%D0%B7%D1%80%D0%B0%D0%B1%D0%BE%D1%82%D1%87%D0%B8%D0%BA-%D0%B3%D0%BE%D0%B2%D0%BE%D1%80%D0%B8%D1%82-%D1%87%D1%82%D0%BE-%D0%BD%D0%B0%D0%B4%D0%BE-%D0%BE%D0%B1%D0%BD%D0%BE%D0%B2%D0%B8%D1%82%D1%8C%D1%81%D1%8F)
2. У вас не включен контроль MD5 в файле. Во всяком случае в логах проверки MD5 не видно. Скажите каким именно образом вы запускаете файл на печать.
3. Вы не приложили файл Красный 2.gcode
4. В логах не вижу  результатов проверки системы `CHECK_SYSTEM`

soblod, 2025-05-15:
> 1. [У вас не обновлен [...]

[Desktop.zip](https://github.com/user-attachments/files/20225601/Desktop.zip)

1. Теперь понятно насчет обновления, сделал
2.  В орка в Прочие прописан путь в скрипты постобработки адрес C:\ORCA\addMD5.bat; там лежит файл скрипта, на принтере сделана команда SAVE_ZMOD_DATA FORCE_MD5=1, нарезаю стол, нажимаю отправить на печать, принтер подключен по по протоколу klipper IP:7125, далее появляется и закрывается окно командной строки, принтер где то минуту думает, после у него появляется задание и он начинает печать.
3. приложил еще раз в архиве
4. Сделал CHECK_SYSTEM перед печатью

ghzserg, 2025-05-15:
Логи сняты правильно. Попробуйте  отключить fix_e0011. Также  попробуйте  отключить родной экран и печатать без экрана - очень похоже, что не хватает оперативной памяти

soblod, 2025-05-18:
Закрывайте пока тему, удалил мод с части принтеров, как разберусь про печать без экрана поэкспериментирую

closed 2025-05-19 by ghzserg

soblod, 2025-06-04:
Проблема стала проявлять и на стоковой прошивке, получилось найти файл который 100% не печатался на всех принтерах на стоке, был отправлен запрос в саппорт, они сказали изменить параметр - Расстояние между точками нечётной оболочки до 0,8 собственно на этом значении принтер с модом смог напечатать файл, на стоке принтер смог напечатать при значении 0,5 по умолчанию в Орке стоит 0,3 Я так понимаю ресурсов не хватает принтеру, потому, что на принтерах других производителей таких проблем не было, и печаталось на параметрах по умолчанию

ghzserg, 2025-06-04:
> Проблема стала проявлять и на стоковой прошивке, получилось найти файл который 100% не печатался на всех принтерах на стоке, был отправлен запрос в саппорт, они сказали изменить параметр - [...]

Пришлите этот файл, я проверю на своем принтере